openwrt/openwrt #9779: bug: WS-AP3825i dtb is not actually padded by the build process, preventing reliable boot

Hurricos, 2022-04-26:
Despite our work in [this thread](https://github.com/openwrt/openwrt/pull/4911#discussion_r799812574) during initial porting of the AP3825i, the image build process doesn't actually pad the device tree blob; see the linked thread for the implications of that. It appears that in the thread, I accepted "the image happens to boot this time [because the device tree blob is spaced in such a way that `fdt resize` adds a lot to get to the next 4k-byte page]" in place of "the image boots this time because the fdt is actually greatly padded]".

Anyways, in practice, what happens is this:

```txt
Boot (PRI)-> interrupts off;
Boot (PRI)-> bootm start 0x2000000;
## Booting kernel from FIT Image at 02000000 ...
   Using 'config-1' configuration
   Trying 'kernel-1' kernel subimage
     Description:  POWERPC OpenWrt Linux-5.10.111
:
## Flattened Device Tree from FIT Image at 02000000
   Using 'config-1' configuration
   Trying 'fdt-1' FDT blob subimage
     Description:  POWERPC OpenWrt extreme-networks_ws-ap3825i device tree blob
:
     Data Size:    11524 Bytes = 11.3 KiB
:
   Booting using the fdt blob at 0x2681ffc
Boot (PRI)-> bootm loados;
   Uncompressing Kernel Image ... OK
Boot (PRI)-> fdt resize;
Boot (PRI)-> fdt boardsetup;
ft_fixup_l2cache: FDT_ERR_NOTFOUND
Boot (PRI)-> fdt chosen;
WARNING: could not set bootargs FDT_ERR_NOSPACE.
WARNING: could not set linux,stdout-path FDT_ERR_NOSPACE.
Boot (PRI)->
```

This was discovered while trying to boot OpenWrt 22.03-rc1 through the serial console.

A temporary user workaround is to double each `fdt` operation in the `bootcmd` and add an `fdt resize` between the two copies; again, see the linked thread. However, I need to look much more closely at how OpenWrt device image formulas to work, as clearly [this code does not pad the DTB](https://github.com/openwrt/openwrt/pull/4911#discussion_r819144403).

Hurricos, 2022-04-26:
The build script which is responsible for building a FIT image consumes `build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/image-ws-ap3825i.dtb`:

```txt
/devel/openwrt/AP3825i/openwrt/scripts/mkits.sh -D extreme-networks_ws-ap3825i -o /devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/extreme-networks_ws-ap3825i-kernel.bin.its -k /devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/extreme-networks_ws-ap3825i-kernel.bin -C lzma   -d /devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/image-ws-ap3825i.dtb   -a 0x00000000 -e 0x00000000    -c "config-1" -A powerpc -v 5.15.31
```

This file is produced much earlier:

```txt
powerpc-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp  -I/devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/linux-5.15.31/arch/powerpc/boot/dts -I/devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/linux-5.15.31/arch/powerpc/boot/dts/include -I/devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/linux-5.15.31/include/ -undef -D__DTS__  -o /devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/image-ws-ap3825i.dtb.tmp /devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/linux-5.15.31/arch/powerpc/boot/dts/ws-ap3825i.dts
:
/devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/linux-5.15.31/scripts/dtc/dtc -O dtb -i/devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/linux-5.15.31/arch/powerpc/boot/dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address   -o /devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/image-ws-ap3825i.dtb /devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/image-ws-ap3825i.dtb.tmp
```

The `Build/dtb` recipe does run later (though before `mkits`), but outputs elsewhere:

```txt
powerpc-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp  -I/devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/linux-5.15.31/arch/powerpc/boot/dts -I/devel/openwrt/AP3825i/openwrt/build_dir/target-
powerpc_8540_musl/linux-mpc85xx_p1020/linux-5.15.31/arch/powerpc/boot/dts/include -I/devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/linux-5.15.31/include/ -undef -D__DTS__  -o /devel/openwrt/AP3825i/o
penwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/extreme-networks_ws-ap3825i-kernel.bin.dtb.tmp /devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/linux-5.15.31/arch/powerpc/boot/dts/ws-ap38
25i.dts
/devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/linux-5.15.31/scripts/dtc/dtc -O dtb -i/devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/linux-5.15.31/arch/powerpc/
boot/dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-un
ique_unit_address --space 20480  -o /devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-mpc85xx_p1020/extreme-networks_ws-ap3825i-kernel.bin.dtb /devel/openwrt/AP3825i/openwrt/build_dir/target-powerpc_8540_musl/linux-m
pc85xx_p1020/extreme-networks_ws-ap3825i-kernel.bin.dtb.tmp
``` 

So basically, I just need to find a way to make sure the output is redirected accordingly. Tough to do. Groan.

Hurricos, 2022-04-26:
OK, I've fixed it.

This diff:

```diff
diff --git a/target/linux/mpc85xx/image/Makefile b/target/linux/mpc85xx/image/Makefile
index e1d1bc6948..fb931760a7 100644
--- a/target/linux/mpc85xx/image/Makefile
+++ b/target/linux/mpc85xx/image/Makefile
@@ -6,7 +6,7 @@ include $(INCLUDE_DIR)/image.mk
 DEVICE_VARS += DTB_SIZE
 
 define Build/dtb
-       $(call Image/BuildDTB,$(DTS_DIR)/$(DEVICE_DTS).dts,$@.dtb,,--space $(DTB_SIZE))
+       $(call Image/BuildDTB,$(DTS_DIR)/$(DEVICE_DTS).dts,$(dir $@)/image-$(DEVICE_DTS).dtb,,--space $(DTB_SIZE))
 endef
 
 define Device/Default
```

... yields this build output:

```txt
FIT description: POWERPC OpenWrt FIT (Flattened Image Tree)
Created:         Fri Apr  1 19:58:03 2022
 Image 0 (kernel-1)
  Description:  POWERPC OpenWrt Linux-5.15.31
  Created:      Fri Apr  1 19:58:03 2022
  Type:         Kernel Image
  Compression:  lzma compressed
  Data Size:    6895562 Bytes = 6733.95 KiB = 6.58 MiB
  Architecture: PowerPC
  OS:           Linux
  Load Address: 0x00000000
  Entry Point:  0x00000000
  Hash algo:    crc32
  Hash value:   c3a78d66
  Hash algo:    sha1
  Hash value:   9b6ada90c4c55fabdd5bdcc814a2bb8938f6b111
 Image 1 (fdt-1)
  Description:  POWERPC OpenWrt extreme-networks_ws-ap3825i device tree blob
  Created:      Fri Apr  1 19:58:03 2022
  Type:         Flat Device Tree
  Compression:  uncompressed
  Data Size:    20480 Bytes = 20.00 KiB = 0.02 MiB
  Architecture: PowerPC
  Hash algo:    crc32
  Hash value:   366df45c
  Hash algo:    sha1
  Hash value:   93ce4ed0430a7f246c0077ba7999e3d8868eead6
 Default Configuration: 'config-1'
 Configuration 0 (config-1)
  Description:  OpenWrt extreme-networks_ws-ap3825i
  Kernel:       kernel-1
  FDT:          fdt-1
```

Hurricos, 2022-04-26:
Once concern I have about this diff:

```diff
diff --git a/target/linux/mpc85xx/image/Makefile b/target/linux/mpc85xx/image/Makefile
index e1d1bc6948..fb931760a7 100644
--- a/target/linux/mpc85xx/image/Makefile
+++ b/target/linux/mpc85xx/image/Makefile
@@ -6,7 +6,7 @@ include $(INCLUDE_DIR)/image.mk
 DEVICE_VARS += DTB_SIZE
 
 define Build/dtb
-       $(call Image/BuildDTB,$(DTS_DIR)/$(DEVICE_DTS).dts,$@.dtb,,--space $(DTB_SIZE))
+       $(call Image/BuildDTB,$(DTS_DIR)/$(DEVICE_DTS).dts,$(dir $@)/image-$(DEVICE_DTS).dtb,,--space $(DTB_SIZE))
 endef
 
 define Device/Default
 ```

... is that it looks like it impacts anywhere in the `mpc85xx` target where we construct an image using the `dtb` step. Rather than having our later redefinition overwrite the `image-$(DEVICE_DTS).dtb` file, I see two alternatives:

1) Have the ws-ap3825i use a different `dtb` file:

```diff
modified   target/linux/mpc85xx/image/Makefile
@@ -6,7 +6,7 @@ include $(INCLUDE_DIR)/image.mk
 DEVICE_VARS += DTB_SIZE

 define Build/dtb
-    $(call Image/BuildDTB,$(DTS_DIR)/$(DEVICE_DTS).dts,$(dir $@)/image-$(DEVICE_DTS).dtb,,--space $(DTB_SIZE))
+    $(call Image/BuildDTB,$(DTS_DIR)/$(DEVICE_DTS).dts,$(dir $@)/padded-image-$(DEVICE_DTS).dtb,,--space $(DTB_SIZE))
 endef

 define Device/Default
modified   target/linux/mpc85xx/image/p1020.mk
@@ -74,7 +74,7 @@ define Device/extreme-networks_ws-ap3825i
   DEVICE_PACKAGES := kmod-ath10k-ct ath10k-firmware-qca988x-ct
   BLOCKSIZE := 128k
   DTB_SIZE := 20480
-  KERNEL = kernel-bin | lzma | dtb | fit lzma $(KDIR)/image-$$(DEVICE_DTS).dtb
+  KERNEL = kernel-bin | lzma | dtb | fit lzma $(KDIR)/padded-image-$$(DEVICE_DTS).dtb
   IMAGES := sysupgrade.bin
   IMAGE/sysupgrade.bin := append-kernel | append-rootfs | pad-rootfs | append-metadata
 endef
 ```
 
 2) Explicitly change the build rule from `dtb` to `padded-dtb`:
 ```diff
 modified   target/linux/mpc85xx/image/Makefile
@@ -5,8 +5,8 @@ include $(INCLUDE_DIR)/image.mk

 DEVICE_VARS += DTB_SIZE

-define Build/dtb
-    $(call Image/BuildDTB,$(DTS_DIR)/$(DEVICE_DTS).dts,$(dir $@)/image-$(DEVICE_DTS).dtb,,--space $(DTB_SIZE))
+define Build/padded-dtb
+    $(call Image/BuildDTB,$(DTS_DIR)/$(DEVICE_DTS).dts,$(dir $@)/padded-image-$(DEVICE_DTS).dtb,,--space $(DTB_SIZE))
 endef

 define Device/Default
modified   target/linux/mpc85xx/image/p1020.mk
@@ -74,7 +74,7 @@ define Device/extreme-networks_ws-ap3825i
   DEVICE_PACKAGES := kmod-ath10k-ct ath10k-firmware-qca988x-ct
   BLOCKSIZE := 128k
   DTB_SIZE := 20480
-  KERNEL = kernel-bin | lzma | dtb | fit lzma $(KDIR)/image-$$(DEVICE_DTS).dtb
+  KERNEL = kernel-bin | lzma | padded-dtb | fit lzma $(KDIR)/padded-image-$$(DEVICE_DTS).dtb
   IMAGES := sysupgrade.bin
   IMAGE/sysupgrade.bin := append-kernel | append-rootfs | pad-rootfs | append-metadata
 endef
 ```
 
 I honestly prefer the latter as it seems safer, but I first need to test:

# Checklist
 - [ ] Is my assertion correct that the `Build/dtb` definition override will break other, non-`DTB_SIZE` boards?
 - [ ] Does the second patch actually *work*, or does it misunderstand the expected input / output of the `dtb` build definition?

hauke, 2022-04-30:
The problem was fixed in 1d06277407d3e294589ecde46328303a9f3803fd,

closed 2022-04-30 by hauke