openwrt/openwrt #9781: iptables: iptables command failed

henok400, 2022-04-27:
I tried currently openwrt 22.03 new released. Many packages use iptable get command fail to apply.
My tested package is wifidog the error show 
```
Wed Apr 27 09:31:25 2022 daemon.err wifidog[4127]: iptables command failed(127): iptables -t filter -A WiFiDog_br-lan_Unknown -d 0.0.0.0/0 -p udp --dport 67 -j ACCEPT
Wed Apr 27 09:31:25 2022 daemon.err wifidog[4127]: iptables command failed(127): iptables -t filter -A WiFiDog_br-lan_Unknown -d 0.0.0.0/0 -p tcp --dport 67 -j ACCEPT
Wed Apr 27 09:31:25 2022 daemon.err wifidog[4127]: iptables command failed(127): iptables -t filter -A WiFiDog_br-lan_Unknown -d 0.0.0.0/0 -p udp --dport 8000 -j REJECT
Wed Apr 27 09:31:25 2022 daemon.err wifidog[4127]: iptables command failed(127): iptables -t filter -A WiFiDog_br-lan_Unknown -j REJECT --reject-with icmp-port-unreachable
```
this Stable openwrt 19.xx.x 
iptables-mod-extra
iptables-mod-ipopt
iptables-mod-nat-extra
but now new openwrt this bugs affect using packages.

jow-, 2022-04-27:
Is there any error reported when you run `iptables -t filter -A WiFiDog_br-lan_Unknown -d 0.0.0.0/0 -p udp --dport 67 -j ACCEPT` on the command line?

henok400, 2022-04-27:
This the full details system log show 
i'm not familiar in development but good tester 
```
Wed Apr 27 10:00:53 2022 daemon.info wifidog[4127]: Handler for termination caught signal 15
Wed Apr 27 10:00:53 2022 daemon.info wifidog[4127]: Cleaning up and exiting
Wed Apr 27 10:00:53 2022 daemon.info wifidog[4127]: Flushing firewall rules...
Wed Apr 27 10:00:53 2022 daemon.info wifidog[4127]: Closing ICMP socket
Wed Apr 27 10:00:53 2022 daemon.info wifidog[4127]: Removing Firewall rules
Wed Apr 27 10:00:54 2022 daemon.info wifidog[4127]: Explicitly killing the fw_counter thread
Wed Apr 27 10:00:55 2022 daemon.info wifidog[4923]: Reading configuration file '/etc/wifidog.conf'
Wed Apr 27 10:00:55 2022 daemon.warn wifidog[4923]: PopularServers not set in config file, this will become fatal in a future version.
Wed Apr 27 10:00:55 2022 daemon.info wifidog[4923]: Setting started_time
Wed Apr 27 10:00:55 2022 daemon.notice wifidog[4923]: Creating web server on 192.168.1.1:2060
Wed Apr 27 10:00:55 2022 daemon.info wifidog[4923]: Closing ICMP socket
Wed Apr 27 10:00:55 2022 daemon.info wifidog[4923]: Removing Firewall rules
Wed Apr 27 10:00:56 2022 daemon.info wifidog[4923]: Creating ICMP socket
Wed Apr 27 10:00:56 2022 daemon.info wifidog[4923]: Initializing Firewall
Wed Apr 27 10:00:56 2022 daemon.info wifidog[4923]: get_ext_iface(): Detected wlan0 as the default interface after trying 1
Wed Apr 27 10:00:56 2022 daemon.err wifidog[4923]: iptables command failed(127): iptables -t mangle -N WiFiDog_br-lan_Trusted
Wed Apr 27 10:00:56 2022 daemon.err wifidog[4923]: iptables command failed(127): iptables -t mangle -N WiFiDog_br-lan_Outgoing
Wed Apr 27 10:00:56 2022 daemon.err wifidog[4923]: iptables command failed(127): iptables -t mangle -N WiFiDog_br-lan_Incoming
Wed Apr 27 10:00:56 2022 daemon.err wifidog[4923]: iptables command failed(127): iptables -t mangle -I PREROUTING 1 -i br-lan -j WiFiDog_br-lan_Outgoing
Wed Apr 27 10:00:56 2022 daemon.err wifidog[4923]: iptables command failed(127): iptables -t mangle -I PREROUTING 1 -i br-lan -j WiFiDog_br-lan_Trusted
Wed Apr 27 10:00:56 2022 daemon.err wifidog[4923]: iptables command failed(127): iptables -t mangle -I POSTROUTING 1 -o br-lan -j WiFiDog_br-lan_Incoming
Wed Apr 27 10:00:56 2022 daemon.err wifidog[4923]: iptables command failed(127): iptables -t nat -N WiFiDog_br-lan_Outgoing
Wed Apr 27 10:00:56 2022 daemon.err wifidog[4923]: iptables command failed(127): iptables -t nat -N WiFiDog_br-lan_Router
Wed Apr 27 10:00:56 2022 daemon.err wifidog[4923]: iptables command failed(127): iptables -t nat -N WiFiDog_br-lan_Internet
Wed Apr 27 10:00:56 2022 daemon.err wifidog[4923]: iptables command failed(127): iptables -t nat -N WiFiDog_br-lan_Global
Wed Apr 27 10:00:56 2022 daemon.err wifidog[4923]: iptables command failed(127): iptables -t nat -N WiFiDog_br-lan_Unknown
Wed Apr 27 10:00:56 2022 daemon.err wifidog[4923]: iptables command failed(127): iptables -t nat -N WiFiDog_br-lan_AuthServers
```

jow-, 2022-04-27:
Please run the command `iptables -t filter -A WiFiDog_br-lan_Unknown -d 0.0.0.0/0 -p udp --dport 67 -j ACCEPT` on the command line and see if it prints any error message.

hnyman, 2022-04-30:
> I tried currently openwrt 22.03 new released. Many packages use iptable get command fail to apply.
> ...
>  it say iptable not found

Have you installed iptables-nft ?
22.03 comes with nftables (nft) based firewall, and there is no iptables by default.
opkg install iptables-nft to have a conversion tool for applying old iptables commands.

henok400, 2022-04-30:
@hnyman please read fully what i'm saying
this Stable openwrt 19.xx.x
iptables-mod-extra
iptables-mod-ipopt
iptables-mod-nat-extra
If you csn go to solution.

hnyman, 2022-04-30:
@henok  please read fully what i'm saying

You are talking about iptables extensions (in 19.07?), but apparently have not even installed the basic iptables itself into your 22.03. (You have only update the packages lists with "opkg update" but not installed anything, apparently)

Like I said above, 22.03 does not have iptables by default.
You are running into that.
[image]
[image]

`opkg install iptables-nft` (after the "opkg update" that you already ran), and you will have the iptables command. 
Then you can look into needed specific submodules for the packages you are using.

Ps. 
I have no idea what "csn" means

henok400, 2022-04-30:
[image]
iptables-nft already up to date 
but still wifidog sending firewall command problem continued. This happen new openwrt.
csn i mean spelling issue  "can".
other if i use wifidog additional error show the firewall can't handle iptable
```
Sat Apr 30 08:21:34 2022 daemon.info wifidog[4657]: Captured 192.168.1.165 requesting [http%3A%2F%2Fdetectportal.firefox.com%2Fcanonical.html] and re-directing them to login page
Sat Apr 30 08:21:35 2022 daemon.info wifidog[4657]: Received connection from 192.168.1.165, spawning worker thread
Sat Apr 30 08:21:35 2022 daemon.info wifidog[4657]: Got client MAC address for ip 192.168.1.165: 1c:bf:ce:22:3e:56
Sat Apr 30 08:21:35 2022 daemon.info wifidog[4657]: Check host 192.168.199.1 is in whitelist or not 

```

Alex2357, 2023-03-18:
It worked for me

```
root@OpenWrt:~# iptables --list
# Warning: iptables-legacy tables present, use iptables-legacy to see them
Chain INPUT (policy ACCEPT)
target     prot opt source               destination         

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination         

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         
root@OpenWrt:~# 

```

```
root@OpenWrt:~# opkg install iptables-nft 
Installing iptables-nft (1.8.7-7) to root...
Downloading https://downloads.openwrt.org/releases/22.03.2/targets/ath79/generic/packages/iptables-nft_1.8.7-7_mips_24kc.ipk
Installing kmod-nf-ipt (5.10.146-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.2/targets/ath79/generic/packages/kmod-nf-ipt_5.10.146-1_mips_24kc.ipk
Installing kmod-ipt-core (5.10.146-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.2/targets/ath79/generic/packages/kmod-ipt-core_5.10.146-1_mips_24kc.ipk
Installing libxtables12 (1.8.7-7) to root...
Downloading https://downloads.openwrt.org/releases/22.03.2/targets/ath79/generic/packages/libxtables12_1.8.7-7_mips_24kc.ipk
Installing libiptext0 (1.8.7-7) to root...
Downloading https://downloads.openwrt.org/releases/22.03.2/targets/ath79/generic/packages/libiptext0_1.8.7-7_mips_24kc.ipk
Installing libiptext6-0 (1.8.7-7) to root...
Downloading https://downloads.openwrt.org/releases/22.03.2/targets/ath79/generic/packages/libiptext6-0_1.8.7-7_mips_24kc.ipk
Installing libiptext-nft0 (1.8.7-7) to root...
Downloading https://downloads.openwrt.org/releases/22.03.2/targets/ath79/generic/packages/libiptext-nft0_1.8.7-7_mips_24kc.ipk
Installing kmod-nft-compat (5.10.146-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.2/targets/ath79/generic/packages/kmod-nft-compat_5.10.146-1_mips_24kc.ipk
Installing xtables-nft (1.8.7-7) to root...
Downloading https://downloads.openwrt.org/releases/22.03.2/targets/ath79/generic/packages/xtables-nft_1.8.7-7_mips_24kc.ipk
Configuring kmod-nf-ipt.
Configuring kmod-nft-compat.
Configuring libxtables12.
Configuring libiptext0.
Configuring kmod-ipt-core.
Configuring libiptext6-0.
Configuring libiptext-nft0.
Configuring xtables-nft.
Configuring iptables-nft.
root@OpenWrt:~# 
```

AzhomeTech, 2024-08-14:
it work for me as well