openwrt/openwrt #9783: Troubles with ipset and dnsmasq after update to 21.02.3

mikhirev, 2022-04-27:
I've recently updated from OpenWrt 21.02.1 to 21.02.3. Before that I configured dnsmasq+ipset as described [here](https://openwrt.org/docs/guide-user/firewall/fw3_configurations/dns_ipset) and it worked perfectly. But after update it is working very unstable.

First, the ipset configured in `/etc/config/firewall` does not appear at startup or firewall restart. Second, if I create it manually with `ipset create <...>`, dnsmasq is not always able to add new addresses there. This behavior is unstable, so it did not work after startup, but at some moment (after several restarts, but not immediately after one of them) it started working. Several days later it stopped working again. I restarted dnsmasq that did not help, but after several minutes it was able to add IP addresses to ipset again.

I don't see any related error messages in system log or kernel log. Please help me to investigate this issue.

[List of installed packages.](https://github.com/openwrt/openwrt/files/8571337/packages.txt)

```
# dnsmasq --version
Dnsmasq version 2.85  Copyright (c) 2000-2021 Simon Kelley
Compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
# ipset --version
ipset v7.6, protocol version: 7
# uname -a
Linux OpenWrt 5.4.188 #0 Sat Apr 16 12:59:34 2022 mips GNU/Linux
```

wongsyrone, 2022-04-27:
ipset in /etc/config/firewall is converted to nftables set since firewall4. You have to use the latest dnsmasq testing release to use nftset.

jow-, 2022-04-27:
@mikhirev - ist an `fw3 restart` on the command line reporting any error wrt. ipset?

mikhirev, 2022-04-27:
> @mikhirev - is an `fw3 restart` on the command line reporting any error wrt. ipset?

Oops, thank you, there was a mistake in my config: i missed a `match` option. Now it looks as follows:

```
config ipset                  
        option name 'myipset'
        option match 'dst_ip' 
        option storage 'hash'
        option enabled '1'
```

And now the ipset is added correctly. But this does not solve the dnsmasq issue.

mikhirev, 2022-04-27:
> ipset in /etc/config/firewall is converted to nftables set since firewall4. You have to use the latest dnsmasq testing release to use nftset.

I'm on 21.02.3 and use firewall3.

closed 2022-05-16 by mikhirev