openwrt/openwrt #9788: Fw4 won't start after upgrade to 2022-04-21-fc83d462-1

pronichkin, 2022-04-27:
I have Linksys WRT1900AC v1. It was running OpenWRT 21.02.1 until recently. Last weekend I upgraded it to 22.03.0-rc1. It kept running well.

Yesterday I decided to update some packages. I know it's generally not recommended, but I was curious.

Here's the list of the packages I updated.

```
iwinfo - 2022-03-01-90bfbb9a-1 - 2022-04-24-a479b9b0-1
luci-app-statistics - git-22.002.35118-8572ae0 - git-22.115.68420-46794ea
libiwinfo-lua - 2022-03-01-90bfbb9a-1 - 2022-04-24-a479b9b0-1
ucode-mod-ubus - 2022-02-11-a29bad94-1 - 2022-04-13-e14b0993-1
ucode-mod-uci - 2022-02-11-a29bad94-1 - 2022-04-13-e14b0993-1
firewall4 - 2022-02-12-53caa1a7-1 - 2022-04-21-fc83d462-1
ucode-mod-fs - 2022-02-11-a29bad94-1 - 2022-04-13-e14b0993-1
libiwinfo-data - 2022-03-01-90bfbb9a-1 - 2022-04-24-a479b9b0-1
ucode - 2022-02-11-a29bad94-1 - 2022-04-13-e14b0993-1
libiwinfo20210430 - 2022-03-01-90bfbb9a-1 - 2022-04-24-a479b9b0-1
```

After that update, firewall4 refuses to start anymore. Originally, it showed the following error.

```
root@pronichkin-router:~# fw4 start
Error loading shared library libucode.so.20220322: No such file or directory (needed by /usr/bin/utpl)
Error relocating /usr/bin/utpl: uc_vm_init: symbol not found
Error relocating /usr/bin/utpl: uc_vm_invoke: symbol not found
Error relocating /usr/bin/utpl: ucv_from_json: symbol not found
Error relocating /usr/bin/utpl: ucv_object_delete: symbol not found
Error relocating /usr/bin/utpl: uc_vm_trace_set: symbol not found
Error relocating /usr/bin/utpl: uc_vm_scope_get: symbol not found
Error relocating /usr/bin/utpl: uc_source_new_buffer: symbol not found
Error relocating /usr/bin/utpl: ucv_object_add: symbol not found
Error relocating /usr/bin/utpl: ucv_string_new: symbol not found
Error relocating /usr/bin/utpl: ucv_get: symbol not found
Error relocating /usr/bin/utpl: ucv_object_get: symbol not found
Error relocating /usr/bin/utpl: uc_vm_free: symbol not found
Error relocating /usr/bin/utpl: ucv_int64_get: symbol not found
Error relocating /usr/bin/utpl: uc_compile: symbol not found
Error relocating /usr/bin/utpl: ucv_put: symbol not found
Error relocating /usr/bin/utpl: ucv_array_push: symbol not found
Error relocating /usr/bin/utpl: uc_program_write: symbol not found
Error relocating /usr/bin/utpl: uc_source_new_file: symbol not found
Error relocating /usr/bin/utpl: ucv_array_new: symbol not found
Error relocating /usr/bin/utpl: uc_stdlib_load: symbol not found
Error relocating /usr/bin/utpl: uc_vm_execute: symbol not found
root@pronichkin-router:~#
```

My system has `/usr/lib/libucode.so.20220206` but not `/usr/lib/libucode.so.20220322`. So, just for testing, I decided to copy the former into the later. Unfortunately, it did not solve the problem, but now fw4 gives an entirely different error error. (Which might mean I'm onto something.)

```
root@pronichkin-router:~# cp /usr/lib/libucode.so.20220206 /usr/lib/libucode.so.20220322
root@pronichkin-router:~# fw4 start
/proc/self/fd/0:1:8-8: Error: syntax error, unexpected '='

/proc/self/fd/0:2:9-9: Error: syntax error, unexpected '='

/proc/self/fd/0:3:10-10: Error: syntax error, unexpected '='

/proc/self/fd/0:5:15-15: Error: syntax error, unexpected '='

/proc/self/fd/0:7:18-18: Error: syntax error, unexpected '='

/proc/self/fd/0:8:18-18: Error: syntax error, unexpected '='

/proc/self/fd/0:9:18-18: Error: syntax error, unexpected '='

/proc/self/fd/0:10:18-18: Error: syntax error, unexpected '='

/proc/self/fd/0:11:18-18: Error: syntax error, unexpected '='

/proc/self/fd/0:12:18-18: Error: syntax error, unexpected '='

root@pronichkin-router:~#
```

so far, I'm out of ideas. Any suggestions, besides reflashing my router to the original 22.03.0-rc1 firmware and never updating packages again?

P.S. I also noticed that on my home network (behind said router) I now have Internet working, but only IPv6. IPv4 is not passing by. Not sure if that's the intended consequence of fw4 being down.

dave14305, 2022-04-27:
> I also noticed that on my home network (behind said router) I now have Internet working, but only IPv6. IPv4 is not passing by. (Ironically, this means that most of the websites work--except for GitHub.) Not sure if that's the intended consequence of fw4 being down.

Without the firewall performing NAT, your IPv4 traffic is unable to go out. That’s why only IPv6 works. But it’s a very vulnerable position with no firewall active. Hope you saw the advice on the forums to reset.

DocSniper, 2022-04-27:
The problem 
`Error loading shared library libucode.so.20220322: No such file or directory (needed by /usr/bin/utpl)`
can be solved with: 
`opkg update && opkg install libucode20220322`

pronichkin, 2022-04-27:
> can be solved with:
> `opkg update && opkg install libucode20220322`

whoa! This magically solved everything. Thanks a bunch!

```
root@pronichkin-router:~# opkg install libucode20220322
Installing libucode20220322 (2022-04-13-e14b0993-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc1/packages/arm_cortex-a9_vfpv3-d16/base/libucode20220322_2022-04-13-e14b0993-1_arm_cortex-a9_vfpv3-d16.ipk
Configuring libucode20220322.
root@pronichkin-router:~# fw4 start
root@pronichkin-router:~#
```

now everything seems back to normal.

DocSniper, 2022-04-27:
Great, glad I could help. 😄

However, there really seems to be a problem with some 22.03.0-rc1 images with (IPv4) NAT/masquerading, there, as described by you, only IPv6 works, but no IPv4. I have this problem e.g. on my Archer C7 v2.

commit 573ce80c (2022-04-28): ucode: reorder BuildPackage calls

Ensure that the libucode recipe is processed before the ucode one in
order to reliably encode the ABI version into ucode's libucode dependency.

Fixes: #9788
Ref: https://forum.openwrt.org/t/fw4-wont-start-after-upgrade/126308
commit 5a117042 (2022-04-28): ucode: reorder BuildPackage calls

Ensure that the libucode recipe is processed before the ucode one in
order to reliably encode the ABI version into ucode's libucode dependency.

Fixes: #9788
Ref: https://forum.openwrt.org/t/fw4-wont-start-after-upgrade/126308

(cherry picked from commit 573ce80ca612b0f642b7cbd5d9d33d89b7e96670)
closed 2022-04-28 by jow-