openwrt/openwrt #9796: openwrt-22.03.0-rc1-rockchip-armv8-friendlyarm_nanopi-r4s-ext4 Upstream speed limit（bug）

f4nff, 2022-04-28:
[https://downloads.openwrt.org/releases/22.03.0-rc1/targets/rockchip/armv8/openwrt-22.03.0-rc1-rockchip-armv8-friendlyarm_nanopi-r4s-ext4-sysupgrade.img.gz](https://downloads.openwrt.org/releases/22.03.0-rc1/targets/rockchip/armv8/openwrt-22.03.0-rc1-rockchip-armv8-friendlyarm_nanopi-r4s-ext4-sysupgrade.img.gz)

[image]
[image]
[image]

Upstream speed: 3Mbps

The same network, I use x86 img upload speed is no problem. Only r4s openwrt has such a problem.

 [https://download.friendlyelec.com/nanopir4s](https://download.friendlyelec.com/nanopir4s)
This is the official firmware provided by r4s, and it is no problem.

f4nff, 2022-05-08:
[https://downloads.openwrt.org/snapshots/targets/rockchip/armv8/openwrt-rockchip-armv8-friendlyarm_nanopi-r4s-ext4-sysupgrade.img.gz](https://downloads.openwrt.org/snapshots/targets/rockchip/armv8/openwrt-rockchip-armv8-friendlyarm_nanopi-r4s-ext4-sysupgrade.img.gz)

[image]
problem still exists.

sthapa, 2022-05-26:
This might be related to the ext4 fs image.  I ran an iperf3 test against my r4s (installed using the 22.03-rc1 squashfs image) and got reasonable nubmers:

```
BusyBox v1.35.0 (2022-04-20 10:37:37 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 22.03.0-rc1, r19302-df622768da
 -----------------------------------------------------
root@r4s-fw:~# iperf3 -s -p 22333
-----------------------------------------------------------
Server listening on 22333 (test #1)
-----------------------------------------------------------
Accepted connection from 192.168.50.146, port 40354
[  5] local 192.168.50.1 port 22333 connected to 192.168.50.146 port 40356
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   109 MBytes   913 Mbits/sec                  
[  5]   1.00-2.00   sec   112 MBytes   939 Mbits/sec                  
[  5]   2.00-3.00   sec   112 MBytes   938 Mbits/sec                  
[  5]   3.00-4.00   sec   111 MBytes   931 Mbits/sec                  
[  5]   4.00-5.00   sec   111 MBytes   929 Mbits/sec                  
[  5]   5.00-6.00   sec   112 MBytes   937 Mbits/sec                  
[  5]   6.00-7.00   sec   112 MBytes   939 Mbits/sec                  
[  5]   7.00-8.00   sec   112 MBytes   939 Mbits/sec                  
[  5]   8.00-9.00   sec   110 MBytes   924 Mbits/sec                  
[  5]   9.00-10.00  sec   112 MBytes   939 Mbits/sec                  
[  5]  10.00-10.00  sec   532 KBytes   931 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec  1.09 GBytes   933 Mbits/sec                  receiver
-----------------------------------------------------------
Server listening on 22333 (test #2)
-----------------------------------------------------------

^Ciperf3: interrupt - the server has terminated
```

f4nff, 2022-05-26:
You don't test the LAN, you test the internet.

`iperf3 -c 104.99.221.22 -p 22111 -t 3333`

The LAN test is fine, but the Internet test will not work.

f4nff, 2022-05-26:
@sthapa 

[https://downloads.openwrt.org/releases/22.03.0-rc1/targets/rockchip/armv8/openwrt-22.03.0-rc1-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img.gz](https://downloads.openwrt.org/releases/22.03.0-rc1/targets/rockchip/armv8/openwrt-22.03.0-rc1-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img.gz)

```
root@OpenWrt:~# iperf3 -c de.7mb.xyz
Connecting to host de.7mb.xyz, port 5201
[  5] local 192.168.2.2 port 54764 connected to 31.25.89.164 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   559 KBytes  4.58 Mbits/sec    0    142 KBytes
[  5]   1.00-2.00   sec   377 KBytes  3.09 Mbits/sec    0    167 KBytes
[  5]   2.00-3.00   sec   377 KBytes  3.09 Mbits/sec    0    167 KBytes
[  5]   3.00-4.00   sec   440 KBytes  3.61 Mbits/sec    0    167 KBytes
[  5]   4.00-5.00   sec   377 KBytes  3.09 Mbits/sec    0    167 KBytes
[  5]   5.00-6.00   sec   377 KBytes  3.09 Mbits/sec    0    167 KBytes
[  5]   6.00-7.00   sec   377 KBytes  3.09 Mbits/sec    0    167 KBytes
[  5]   7.00-8.00   sec   440 KBytes  3.61 Mbits/sec    0    167 KBytes
[  5]   8.00-9.00   sec   440 KBytes  3.61 Mbits/sec    0    178 KBytes
[  5]   9.00-10.00  sec   252 KBytes  2.06 Mbits/sec    0    178 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  3.92 MBytes  3.29 Mbits/sec    0             sender
[  5]   0.00-10.18  sec  3.60 MBytes  2.96 Mbits/sec                  receiver

iperf Done.
```

```
root@OpenWrt:~# iperf3 -c 192.168.2.4
Connecting to host 192.168.2.4, port 5201
[  5] local 192.168.2.2 port 41144 connected to 192.168.2.4 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   114 MBytes   958 Mbits/sec    0    346 KBytes
[  5]   1.00-2.00   sec   114 MBytes   954 Mbits/sec    0    362 KBytes
[  5]   2.00-3.00   sec   113 MBytes   949 Mbits/sec    0    379 KBytes
[  5]   3.00-4.00   sec   113 MBytes   949 Mbits/sec    0    379 KBytes
[  5]   4.00-5.00   sec   113 MBytes   947 Mbits/sec    0    379 KBytes
[  5]   5.00-6.00   sec   114 MBytes   953 Mbits/sec    0    379 KBytes
[  5]   6.00-7.00   sec   113 MBytes   950 Mbits/sec    0    379 KBytes
[  5]   7.00-8.00   sec   113 MBytes   950 Mbits/sec    0    402 KBytes
[  5]   8.00-9.00   sec   113 MBytes   948 Mbits/sec    0    519 KBytes
[  5]   9.00-10.00  sec   114 MBytes   956 Mbits/sec    0    519 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.11 GBytes   951 Mbits/sec    0             sender
[  5]   0.00-10.00  sec  1.10 GBytes   949 Mbits/sec                  receiver

iperf Done.
```

The LAN test is fine, but the Internet test will not work.

sthapa, 2022-05-27:
I don't have a iperf3 server on the internet handy to test against but some of my other network gear does a speedtest regularly and that's getting about 200Mb/s (which is the speed of my connection).  I'll try against an iperf3 server when I get the chance.  Might be an issue with iperf3 and the r4s since I don't think the general upstream connection is being limited. 
[image]

f4nff, 2022-05-27:
I have changed other firmwares and there is no problem. Only the firmware released by openwrt has problems. This seems to limit qos for the Internet, but it does not limit the local area network.

sthapa, 2022-05-28:
I tried replicating this from an internet host and got no speed degradations. I'm using the 2022.03-rc1 image using squashfs and the cake/piece.of.cake sqm.  There's obviously some config difference causing @f4nff's issue.  I'd be happy to provide help in narrowing it down. 

root@ubuntu:/# iperf3  -c XXXXX
Connecting to host XXXXX, port 5201
[  5] local XXXX port 35604 connected to XXXXX port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  6.11 MBytes  51.3 Mbits/sec    0   3.36 MBytes       
[  5]   1.00-2.00   sec  21.2 MBytes   178 Mbits/sec    1   4.20 MBytes       
[  5]   2.00-3.00   sec  23.8 MBytes   199 Mbits/sec    0   4.21 MBytes       
[  5]   3.00-4.00   sec  21.2 MBytes   178 Mbits/sec    1   2.96 MBytes       
[  5]   4.00-5.00   sec  17.5 MBytes   147 Mbits/sec    1   2.13 MBytes       
[  5]   5.00-6.00   sec  18.8 MBytes   157 Mbits/sec    0   2.24 MBytes       
[  5]   6.00-7.00   sec  21.2 MBytes   178 Mbits/sec    0   2.35 MBytes       
[  5]   7.00-8.00   sec  21.2 MBytes   178 Mbits/sec    0   2.41 MBytes       
[  5]   8.00-9.00   sec  21.2 MBytes   178 Mbits/sec    0   2.46 MBytes       
[  5]   9.00-10.00  sec  21.2 MBytes   178 Mbits/sec    0   2.49 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   194 MBytes   162 Mbits/sec    3             sender
[  5]   0.00-10.13  sec   193 MBytes   160 Mbits/sec                  receiver

f4nff, 2022-05-28:
[image]

OpenWrt 22.03.0-rc3 r19378-9f415792e1 / LuCI openwrt-22.03 branch git-22.140.66268-ef99568

f4nff, 2022-05-28:
I have repeatedly confirmed that it is the openwrt firmware ([https://downloads.openwrt.org/releases/22.03.0-rc3/targets/rockchip/armv8/openwrt-22.03.0-rc3-rockchip-armv8-friendlyarm_nanopi-r4s-ext4 -sysupgrade.img.gz](https://downloads.openwrt.org/releases/22.03.0-rc3/targets/rockchip/armv8/openwrt-22.03.0-rc3-rockchip-armv8-friendlyarm_nanopi-r4s-ext4 -sysupgrade.img.gz)), I tried to flash other firmware, but nothing happened.

The following link firmware does not have such a bug.
[https://drive.google.com/file/d/1drrJ86LNJZKmcA7117ioUlPabBpD-vCi/view?usp=sharing](https://drive.google.com/file/d/1drrJ86LNJZKmcA7117ioUlPabBpD-vCi/view?usp=sharing)

f4nff, 2022-08-15:
rc6 problem still exists