openwrt/openwrt #9800: resize.f2fs from f2fs-tools and f2fs-tools-selinux from master/21.02.x/22.03.x broken

erdoukki, 2022-04-28:
All tries of expanding f2fs according to https://openwrt.org/docs/guide-user/installation/openwrt_x86#resizing_f2fs_overlay will fail with 'more segment needed' error message !
```
Error: Device size is not sufficient for F2FS volume, more segment needed =12561root@OpenWrt:/#                    
```
This look like to be an already known problem with the release 1.4.x which is included in all latest OpenWrt releases.

https://bugs.archlinux.org/task/71801
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1002034
https://bugzilla.redhat.com/show_bug.cgi?id=1979307

This may have been fixed in archlinux with this commit: https://github.com/archlinux/svntogit-packages/commit/a0ac2bbff1dab7a7c73b3ce1ecb87cf4cc149f0f

commited PKGBUILD details: https://github.com/archlinux/svntogit-packages/commit/a0ac2bbff1dab7a7c73b3ce1ecb87cf4cc149f0f#diff-37538beb61ff63edebbf735dfcf39e5d732f49183d6beb097169d971875ca422R22-R23

Look to be only a cherry-pick f056fbeff08d30a6d9acdb9e06704461ceee3500 from tag d41dcbdf46dc3841cd0a0507e6573e38cb6c55bb on GIT https://git.kernel.org/pub/scm/linux/kernel/git/jaegeuk/f2fs-tools.git/about
```
	# resize.f2fs: fix wrong ovp calculation
	git cherry-pick -n f056fbeff08d30a6d9acdb9e06704461ceee3500
```

I have not tested and looked in 19.x but master, 21.02.x, 22.03.x still include only this version from 2020.
```
root@OpenWrt:/# resize.f2fs -V
resize.f2fs 1.14.0 (2020-08-24)
```
```
root@OpenWrt:/# uname -ar
Linux OpenWrt 5.10.112 #0 SMP Wed Apr 27 18:29:37 2022 aarch64 GNU/Linux                                                   
```

```
root@OpenWrt:/# cat /etc/os-release                                                                                        
NAME="OpenWrt"                                                                                                             
VERSION="SNAPSHOT"                                                                                                         
ID="openwrt"                                                                                                               
ID_LIKE="lede openwrt"                                                                                                     
PRETTY_NAME="OpenWrt SNAPSHOT"                                                                                             
VERSION_ID="snapshot"                                                                                                      
HOME_URL="https://openwrt.org/"                                                                                            
BUG_URL="https://bugs.openwrt.org/"                                                                                        
SUPPORT_URL="https://forum.openwrt.org/"                                                                                   
BUILD_ID="r19529-8fa4361f55"                                                                                               
OPENWRT_BOARD="mvebu/cortexa72"                                                                                            
OPENWRT_ARCH="aarch64_cortex-a72"                                                                                          
OPENWRT_TAINTS=""                                                                                                          
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"                                                                                      
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"                                                                     
OPENWRT_DEVICE_PRODUCT="Generic"                                                                                           
OPENWRT_DEVICE_REVISION="v0"                                                                                               
OPENWRT_RELEASE="OpenWrt SNAPSHOT r19529-8fa4361f55"                                                                       
```

Here are the FIXME:
patch [resize.f2fs: fix wrong ovp calculation](https://git.kernel.org/pub/scm/linux/kernel/git/jaegeuk/f2fs-tools.git/patch/?id=f056fbeff08d30a6d9acdb9e06704461ceee3500)

I have seen an opened PR https://github.com/openwrt/openwrt/pull/4978 but I will proposed one with only this patch included for my own testing, depending from the results.

hauke, 2022-04-30:
This was fixed in 1aac1b36d3cf44c8bfa8d4a6d8df6e815fc06529

closed 2022-04-30 by hauke

erdoukki, 2022-05-01:
@hauke 
Thanks you for the commit, I just test it this morning and can confirm now it is full working !

### TEST DETAILS
```
root@OpenWrt:~# resize.f2fs ${LOOP2}
Info: Segments per section = 1
Info: Sections per zone = 1
Info: sector size = 512
Info: total sectors = 30734464 (15007 MB)
Info: MKFS version
  "Linux version 5.10.112 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r19529-8fa4361f55) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Wed Apr 27 18:29:37 2022"
Info: FSCK version
  from "Linux version 5.10.112 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r19529-8fa4361f55) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Wed Apr 27 18:29:37 2022"
    to "Linux version 5.10.112 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r19529-8fa4361f55) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Wed Apr 27 18:29:37 2022"
Info: superblock features = 0 : 
Info: superblock encrypt level = 0, salt = 00000000000000000000000000000000
Info: total FS sectors = 207488 (101 MB)
Info: CKPT version = 4c00816e
Info: Duplicate valid checkpoint to mirror position 1024 -> 512
Info: Write valid nat_bits in checkpoint
[migrate_main: 204] Info: Done to migrate Main area: main_blkaddr = 0x1000 -> 0x6c00
[migrate_ssa: 270] Info: Done to migrate SSA blocks: sum_blkaddr = 0xe00 -> 0x4e00
[migrate_nat: 386] Info: Done to migrate NAT blocks: nat_blkaddr = 0xa00 -> 0xa00
[migrate_sit: 444] Info: Done to restore new SIT blocks: 0x600
Info: Write valid nat_bits in checkpoint
[rebuild_checkpoint: 585] Info: Done to rebuild checkpoint blocks
[update_superblock: 695] Info: Done to update superblock

Done: 5.069105 secs
```
I can try more from my MOCHABIN now...