openwrt/openwrt #9811: RA messages with outdated ipv6 prefixes were sent to downstreams

IrineSistiana, 2022-04-30:
Hi.

I found that openwrt was sending outdated ipv6 prefixes via RA messages to downstreams. "outdated" means those ipv6 prefixes were previously assigned by upstreams, but were changed and not valid later (For example, upstream interfaces were restarted and a new prefix was assigned).

Which means, if upstreams restarted and the ipv6 prefix was changed, the downstream ipv6 is gonna broke (or partially).

---

Device: Xiaomi Mi Router 3G
Firmware Version: Reproduced in OpenWrt 22.03.0-rc1 and 21.02.2
Clean firmware, no additional packages installed, no additional firewall settings.

`network`

```TXT
config interface 'lan'
	option proto 'static'
	option netmask '255.255.255.0'
	option device 'br-lan'
	option ipaddr '192.168.1.1'
	option delegate '0'
	option ip6assign '64'
	list ip6class 'wan6'

config interface 'wan'
	option proto 'pppoe'
	option device 'wan'
	option username '******'
	option password '******'
	option delegate '0'
	option peerdns '0'
	option ipv6 '1'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'

config interface 'wan6'
	option proto 'dhcpv6'
	option device '@wan'
	option peerdns '0'
	option sourcefilter '0'
	option reqaddress 'try'
	option reqprefix 'auto'
```

---

Reproduce:

1. When `wan6` got an ipv6-PD, say `aaaa/64`. `lan` then got a delegated prefix `aaaa/64`. Downstreams got `aaaa:xxxx/64`.

`# ifstatus lan`
```txt
{
	"up": true,
	"pending": false,
	"available": true,
	"autostart": true,
	"dynamic": false,
	"uptime": 3122,
	"l3_device": "br-lan",
	"proto": "static",
	"device": "br-lan",
	"updated": [
		"addresses"
	],
	"metric": 0,
	"dns_metric": 0,
	"delegation": false,
	"ipv4-address": [
		{
			"address": "192.168.1.1",
			"mask": 24
		}
	],
	"ipv6-address": [
		
	],
	"ipv6-prefix": [
		
	],
	"ipv6-prefix-assignment": [
		{
			"address": "aaaa::",
			"mask": 64,
			"preferred": 169519,
			"valid": 255919,
			"local-address": {
				"address": "aaaa::1",
				"mask": 64
			}
		}
	],
	"route": [
		
	],
	"dns-server": [
		
	],
	"dns-search": [
		
	],
	"neighbors": [
		
	],
	"inactive": {
		"ipv4-address": [
			
		],
		"ipv6-address": [
			
		],
		"route": [
			
		],
		"dns-server": [
			
		],
		"dns-search": [
			
		],
		"neighbors": [
			
		]
	},
	"data": {
		
	}
}
```

2. Restart `wan6`, it got a new ipv6-PD `bbbb/64`. `lan` then got a delegated prefix `bbbb/64`. But downstreams got both `aaaa:xxxx/64` and `bbbb:xxxx/64` RA messages. 

`# ifstatus lan` shows there is no `aaaa/64` anymore.
```txt
{
	"up": true,
	"pending": false,
	"available": true,
	"autostart": true,
	"dynamic": false,
	"uptime": 3145,
	"l3_device": "br-lan",
	"proto": "static",
	"device": "br-lan",
	"updated": [
		"addresses"
	],
	"metric": 0,
	"dns_metric": 0,
	"delegation": false,
	"ipv4-address": [
		{
			"address": "192.168.1.1",
			"mask": 24
		}
	],
	"ipv6-address": [
		
	],
	"ipv6-prefix": [
		
	],
	"ipv6-prefix-assignment": [
		{
			"address": "bbbb::",
			"mask": 64,
			"preferred": 172774,
			"valid": 259174,
			"local-address": {
				"address": "bbbb::1",
				"mask": 64
			}
		}
	],
	"route": [
		
	],
	"dns-server": [
		
	],
	"dns-search": [
		
	],
	"neighbors": [
		
	],
	"inactive": {
		"ipv4-address": [
			
		],
		"ipv6-address": [
			
		],
		"route": [
			
		],
		"dns-server": [
			
		],
		"dns-search": [
			
		],
		"neighbors": [
			
		]
	},
	"data": {
		
	}
}
```

Please help me to investigate this issue. Thanks!

ptpt52, 2022-04-30:
maybe the `aaaa/64` should be disappear only if timeout?

IrineSistiana, 2022-04-30:
Hi @ptpt52,

> maybe the `aaaa/64` should be disappear only if timeout?

I'm not sure. I haven't tested this yet.

In the step.2, `ifstatus lan` is aware that the new `bbbb/64` address is assigned, and outdated `aaaa/64` is no longer there. So I think this may be a bug.

I also found that if I manually restart `lan` (`ifup lan`) after step.2.  The outdated `aaaa/64` disappeared in `ifconfig br-lan`. And no more outdated RA messages.

ptpt52, 2022-04-30:
> Hi @ptpt52,
> 
> > maybe the `aaaa/64` should be disappear only if timeout?
> 
> I'm not sure. I haven't tested this yet.
> 
> In the step.2, `ifstatus lan` is aware that the new `bbbb/64` address is assigned, and outdated `aaaa/64` is no longer there. So I think `lan` still using `aaaa/64` may be a bug.
> 
> I also found that if I manually restart `lan` (`ifup lan`) after step.2. The outdated `aaaa/64` disappeared in `ifconfig br-lan`. And no more outdated RA messages.

true

ptpt52, 2022-04-30:
for my test, it looks like the aaaa/64 on br-lan would be clear after hours, looks like timeouted or what.

ptpt52, 2022-04-30:
But I think it would be better to clear as soon as possible if it is invalid

IrineSistiana, 2022-05-01:
> But I think it would be better to clear as soon as possible if it is invalid

Agree. I've noticed some strange IPv6 connection issues in downstream devices years ago. I've never thought this is because of the openwrt router until I found there were multiple prefixes in downstream.

Manually delete the outdated aaaa/64 from br-lan using `ip -6 addr del`

IrineSistiana, 2022-05-01:
It seems the source of the problem is that the outdated prefix in interface `br-lan` was not removed when the upstream updated their prefix. And the odhcpd is using the interface address to serve downstream RA/dhcpv6. (? not sure, I'm guessing) 

So my current workaround is to restart the `lan` via hotplug so the outdated prefix in `br-lan` can be flushed when the upstream restarts.

I also found some posts from the forum that mentioned their ipv6 are dead or acting weird after the upstream restart and a new prefix was assigned. Hope my workaround could also help.

```sh
cat >/etc/hotplug.d/iface/99-restart-lan-for-ipv6-update <<EOF 
IPV6_INTERFACE="wan6"
TRIGER_ACTION="ifup"
RESTART_INTERFACE="lan"

if [ "$INTERFACE" = "$IPV6_INTERFACE" -a "$ACTION" = "$TRIGER_ACTION" ]
then
    logger -t hotplug restarting lan
    ifup "$RESTART_INTERFACE"
fi

EOF
```

cwbsw, 2022-05-04:
What's the preferred lifetime? If preferred lifetime is 0, it may be expected.
https://github.com/openwrt/odhcpd/commit/83e14f45581757a4ea1418a64d80885c79db5ef0
* RFC7084 § 4.3 :
				 *    L-13:  If the delegated prefix changes, i.e., the current prefix is
				 *           replaced with a new prefix without any overlapping time
				 *           period, then the IPv6 CE router MUST immediately advertise the
				 *           old prefix with a Preferred Lifetime of zero and a Valid
				 *           Lifetime of either a) zero or b) the lower of the current
				 *           Valid Lifetime and two hours (which must be decremented in
				 *           real time) in a Router Advertisement message as described in
				 *           Section 5.5.3, (e) of [RFC4862].

IrineSistiana, 2022-05-04:
Hi @cwbsw 

You are right. The preferred lifetime is 0 in 22.03-rc1.

Sorry, my bad. I noticed clients were using outdated addresses in 21.02, but I upgraded to 22.03-rc1 and tried to debug the problem. It seems https://github.com/openwrt/odhcpd/commit/83e14f45581757a4ea1418a64d80885c79db5ef0 already fixed that.

closed 2022-05-04 by IrineSistiana

dengxinfa, 2023-03-31:
> What's the preferred lifetime? If preferred lifetime is 0, it may be expected. [openwrt/odhcpd@83e14f4](https://github.com/openwrt/odhcpd/commit/83e14f45581757a4ea1418a64d80885c79db5ef0)
> 
> * RFC7084 § 4.3 :
>   *    L-13:  If the delegated prefix changes, i.e., the current prefix is
>   *           replaced with a new prefix without any overlapping time
>   *           period, then the IPv6 CE router MUST immediately advertise the
>   *           old prefix with a Preferred Lifetime of zero and a Valid
>   *           Lifetime of either a) zero or b) the lower of the current
>   *           Valid Lifetime and two hours (which must be decremented in
>   *           real time) in a Router Advertisement message as described in
>   *           Section 5.5.3, (e) of [RFC4862].
I have a question about `If preferred lifetime is 0, it may be expected`, which means that odhcpd needs to set preferred lifetime is 0, Or the preferred lifetime of the current lan prefix is 0?

No matter how hard I test, the other invalid addresses in the lan still exist.
```
root@GL-MT1300:~# cat /etc/banner
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 22.03.2, r19803-9a599fee93
 -----------------------------------------------------
root@GL-MT1300:~# opkg list-installed | grep odhcp
odhcp6c - 2022-08-05-7d21e8d8-18
odhcpd-ipv6only - 2022-03-22-860ca900-1

root@GL-MT1300:~# 
root@GL-MT1300:~# ifconfig br-lan
br-lan    Link encap:Ethernet  HWaddr 94:83:C4:0F:34:2E  
          inet addr:192.168.8.1  Bcast:192.168.8.255  Mask:255.255.255.0
          inet6 addr: dd3d:f4c8:e43d:4::1/62 Scope:Global
          inet6 addr: 240e:3b6:3253:f1ce::1/63 Scope:Global
          inet6 addr: fe80::9683:c4ff:fe0f:342e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:31383 errors:0 dropped:0 overruns:0 frame:0
          TX packets:20498 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:3275428 (3.1 MiB)  TX bytes:3054063 (2.9 MiB)
root@GL-MT1300:~# ifstatus lan
{
        "up": true,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "uptime": 72536,
        "l3_device": "br-lan",
        "proto": "static",
        "device": "br-lan",
        "updated": [
                "addresses"
        ],
        "metric": 0,
        "dns_metric": 0,
        "delegation": true,
        "ipv4-address": [
                {
                        "address": "192.168.8.1",
                        "mask": 24
                }
        ],
        "ipv6-address": [

        ],
        "ipv6-prefix": [

        ],
        "ipv6-prefix-assignment": [
[... 139 lines cut ...]
        "inactive": {
                "ipv4-address": [

                ],
                "ipv6-address": [

                ],
                "route": [

                ],
                "dns-server": [

                ],
                "dns-search": [

                ],
                "neighbors": [

                ]
        },
        "data": {
                "passthru": "00170010dd3df4c8e43d00000000000000000001"
        }
}
```

fightforlife, 2025-02-15:
Strangely I also have this problem in 24.10. my clients keep all their old ipv6 addresses.
Will try a few things and report back.