openwrt/openwrt #9812: tun crashed

fjqingyou, 2022-04-30:
Because my English is very poor, the following content comes from Baidu translation!

When zerotier is installed and used, the Tun module it depends on will be crashed.

Because it involves at kernel / module c:1140 module_ Put + 0x48 / 0x5C I'm not sure if I should feed back this problem in this place!

# system log
```
Sun May  1 00:47:16 2022 user.info kernel: [  144.905309] procd: - ubus -
Sun May  1 00:47:16 2022 user.info kernel: [  148.651608] procd: - init -
Sun May  1 00:47:16 2022 kern.info kernel: [  158.160825] EXT4-fs (sda1): recovery complete
Sun May  1 00:47:16 2022 kern.info kernel: [  158.272644] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts:
Sun May  1 00:47:16 2022 user.info kernel: [  163.757122] kmodloader: loading kernel modules from /etc/modules.d/*
Sun May  1 00:47:16 2022 kern.info kernel: [  166.960439] mpls_gso: MPLS GSO support
Sun May  1 00:47:16 2022 kern.info kernel: [  167.437230] tun: Universal TUN/TAP device driver, 1.6
Sun May  1 00:47:16 2022 kern.warn kernel: [  167.556315] ------------[ cut here ]------------
Sun May  1 00:47:16 2022 kern.warn kernel: [  167.658396] WARNING: CPU: 0 PID: 715 at kernel/module.c:1140 module_put+0x48/0x5c
Sun May  1 00:47:16 2022 kern.warn kernel: [  167.822951] Modules linked in: tun mpls_iptunnel mpls_router mpls_gso vfat fat nls_utf8 nls_iso8859_1 nls_cp936 nls_cp932 nls_cp437 eeprom_93cx6 sdhci_pltfm sdhci mtk_sd leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug f2fs ext4 mbcache jbd2 exfat usbcore nls_base usb_common crc32c_generic crc32_generic
Sun May  1 00:47:16 2022 kern.warn kernel: [  168.526124] CPU: 0 PID: 715 Comm: kmodloader Not tainted 4.14.241 #0
Sun May  1 00:47:16 2022 kern.warn kernel: [  168.665401] Stack : 00000000 8f1cdf28 805c0000 80072b1c 805f0000 80587d78 00000000 00000000
Sun May  1 00:47:16 2022 kern.warn kernel: [  168.849211]         80550d6c 8f113ca4 8fed763c 805c4927 8054bb00 00000001 8f113c48 ac07f5e8
Sun May  1 00:47:16 2022 kern.warn kernel: [  169.032813]         00000000 00000000 80720000 00005c50 00000000 0000011d 00000007 00000000
Sun May  1 00:47:16 2022 kern.warn kernel: [  169.216367]         00000000 805d0000 0008072c 00000000 80000000 805f0000 00000000 805533ec
Sun May  1 00:47:16 2022 kern.warn kernel: [  169.399917]         800a3f2c 00000474 00000028 8f1cdf28 00000003 802b58c8 00000000 80720000
Sun May  1 00:47:16 2022 kern.warn kernel: [  169.583567]         ...
Sun May  1 00:47:16 2022 kern.warn kernel: [  169.637366] Call Trace:
Sun May  1 00:47:16 2022 kern.warn kernel: [  169.691705] [<8000c810>] show_stack+0x58/0x100
Sun May  1 00:47:16 2022 kern.warn kernel: [  169.789683] [<80485954>] dump_stack+0xa4/0xe0
Sun May  1 00:47:16 2022 kern.warn kernel: [  169.885448] [<8002f788>] __warn+0xe0/0x138
Sun May  1 00:47:16 2022 kern.warn kernel: [  169.975338] [<8002f870>] warn_slowpath_null+0x1c/0x2c
Sun May  1 00:47:16 2022 kern.warn kernel: [  170.086332] [<800a3f2c>] module_put+0x48/0x5c
Sun May  1 00:47:16 2022 user.info kernel: [  170.141695] urngd: v1.0.2 started.
Sun May  1 00:47:16 2022 kern.warn kernel: [  170.182347] [<800a84bc>] do_init_module+0x108/0x1ec
Sun May  1 00:47:16 2022 kern.warn kernel: [  170.363098] [<800a7498>] load_module+0x20e8/0x2494
Sun May  1 00:47:16 2022 kern.warn kernel: [  170.468192] [<800a7974>] SyS_init_module+0x130/0x18c
Sun May  1 00:47:16 2022 kern.warn kernel: [  170.577486] [<80016578>] syscall_common+0x34/0x58
Sun May  1 00:47:16 2022 kern.warn kernel: [  170.683439] ---[ end trace 0a0c0b5c17404a85 ]---
Sun May  1 00:47:16 2022 kern.info kernel: [  170.909587] ipip: IPv4 and MPLS over IPv4 tunneling driver
Sun May  1 00:47:16 2022 kern.info kernel: [  171.236832] ip6_tables: (C) 2000-2006 Netfilter Core Team
Sun May  1 00:47:16 2022 kern.info kernel: [  171.576856] i2c /dev entries driver
Sun May  1 00:47:16 2022 kern.info kernel: [  171.972219] fuse init (API version 7.26)
Sun May  1 00:47:16 2022 kern.info kernel: [  172.349336] Loading modules backported from Linux version v4.19.193-0-g1722257b8ece
Sun May  1 00:47:16 2022 kern.info kernel: [  172.518066] Backport generated by backports.git v4.19.193-1-0-gaa2cc8ea
Sun May  1 00:47:16 2022 kern.info kernel: [  172.865189] ip_tables: (C) 2000-2006 Netfilter Core Team
Sun May  1 00:47:16 2022 kern.info kernel: [  173.093033] lib80211: common routines for IEEE802.11 drivers
Sun May  1 00:47:16 2022 kern.debug kernel: [  173.219217] lib80211_crypt: registered algorithm 'NULL'
Sun May  1 00:47:16 2022 kern.debug kernel: [  173.254485] lib80211_crypt: registered algorithm 'CCMP'
Sun May  1 00:47:16 2022 kern.debug kernel: [  173.293594] lib80211_crypt: registered algorithm 'TKIP'
Sun May  1 00:47:16 2022 kern.debug kernel: [  173.329100] lib80211_crypt: registered algorithm 'WEP'
Sun May  1 00:47:16 2022 kern.info kernel: [  173.583483] nf_conntrack version 0.5.0 (8192 buckets, 32768 max)
Sun May  1 00:47:16 2022 kern.info kernel: [  176.766784] openvswitch: Open vSwitch switching datapath
Sun May  1 00:47:16 2022 kern.info kernel: [  177.252476] xt_time: kernel timezone is -0000
Sun May  1 00:47:16 2022 kern.info kernel: [  181.860329] PPP generic driver version 2.4.2
Sun May  1 00:47:16 2022 kern.info kernel: [  181.995472] NET: Registered protocol family 24
Sun May  1 00:47:16 2022 kern.warn kernel: [  183.342774] bus=0x1, slot = 0x0, irq=0xff
Sun May  1 00:47:16 2022 kern.debug kernel: [  193.013238] ath: EEPROM regdomain: 0x6a
Sun May  1 00:47:16 2022 kern.debug kernel: [  193.013580] ath: EEPROM indicates we should expect a direct regpair map
Sun May  1 00:47:16 2022 kern.debug kernel: [  193.014190] ath: Country alpha2 being used: 00
Sun May  1 00:47:16 2022 kern.debug kernel: [  193.014308] ath: Regpair used: 0x6a
Sun May  1 00:47:16 2022 kern.debug kernel: [  193.510281] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
Sun May  1 00:47:16 2022 kern.info kernel: [  193.585322] ieee80211 phy0: Atheros AR9280 Rev:2 mem=0xc08f0000, irq=24
Sun May  1 00:47:16 2022 user.info kernel: [  193.806437] kmodloader: done loading kernel modules from /etc/modules.d/*
```

# build commit id
```
3b6ce4f634 (originOrigin/openwrt-19.07, origin/openwrt-19.07, openwrt-19.07) kernel: bump 4.14 to 4.14.266
e7596ce0b0 hostapd: Apply SAE/EAP-pwd side-channel attack update 2
1691c1168d mbedtls: Update to version 2.16.12
419b9f4c45 mbedtls: update to 2.16.11
bfa4cccd46 tcpdump: libpcap: Remove http://www.us.tcpdump.org mirror
e92a4e5458 tcpdump: Fix CVE-2018-16301
606106fb29 kernel: bump 4.14 to 4.14.265
524cbcf6f5 build: store SOURCE_DATE_EPOCH in JSON info files
57293f51c4 kernel: bump 4.14 to 4.14.264
```

jow-, 2022-04-30:
Did you install kmod-tun using `opkg --force-depends` ?

fjqingyou, 2022-05-01:
I not using --force-depends, but I build squashfs-sysupgrade image has modify include/kernel-defaults.mk file

#  include/kernel-defaults.mk modify keep kernel identical md5
```
diff --git a/include/kernel-defaults.mk b/include/kernel-defaults.mk
index cc1e2361be..ebb5133830 100644
--- a/include/kernel-defaults.mk
+++ b/include/kernel-defaults.mk
@@ -105,7 +105,9 @@ define Kernel/Configure/Default
                cp $(LINUX_DIR)/.config.set $(LINUX_DIR)/.config.prev; \
        }
        $(_SINGLE) [ -d $(LINUX_DIR)/user_headers ] || $(KERNEL_MAKE) INSTALL_HDR_PATH=$(LINUX_DIR)/user_headers headers_install
-       grep '=[ym]' $(LINUX_DIR)/.config.set | LC_ALL=C sort | mkhash md5 > $(LINUX_DIR)/.vermagic
+
+       # grep '=[ym]' $(LINUX_DIR)/.config.set | LC_ALL=C sort | mkhash md5 > $(LINUX_DIR)/.vermagic
+       cp $(TOPDIR)/local/res/.vermagic $(LINUX_DIR)/.vermagic
 endef
 
 define Kernel/Configure/Initramfs
diff --git a/package/kernel/linux/Makefile b/package/kernel/linux/Makefile
index 39b9e82c27..aa9e73ee07 100644
--- a/package/kernel/linux/Makefile
+++ b/package/kernel/linux/Makefile
@@ -24,7 +24,8 @@ export SHELL:=/bin/sh
 include $(INCLUDE_DIR)/package.mk
 
 ifeq ($(DUMP),)
-  STAMP_BUILT:=$(STAMP_BUILT)_$(shell $(SCRIPT_DIR)/kconfig.pl $(LINUX_DIR)/.config | mkhash md5)
+       # STAMP_BUILT:=$(STAMP_BUILT)_$(shell $(SCRIPT_DIR)/kconfig.pl $(LINUX_DIR)/.config | mkhash md5)
+       STAMP_BUILT:=$(STAMP_BUILT)_$(shell cat $(TOPDIR)/local/res/.vermagic)
   -include $(LINUX_DIR)/.config
 endif
```

# local/res/.vermagic file body
```
2e88863ccdd594fb8e842df3c25842ee
```

# install
```
root@HL:~# opkg update
Downloading http://downloads.openwrt.org/releases/19.07.8/targets/ramips/mt7621/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading http://downloads.openwrt.org/releases/19.07.8/targets/ramips/mt7621/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.8/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading http://downloads.openwrt.org/releases/19.07.8/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.8/packages/mipsel_24kc/freifunk/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_freifunk
Downloading http://downloads.openwrt.org/releases/19.07.8/packages/mipsel_24kc/freifunk/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.8/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading http://downloads.openwrt.org/releases/19.07.8/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.8/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading http://downloads.openwrt.org/releases/19.07.8/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.8/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading http://downloads.openwrt.org/releases/19.07.8/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.8/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/19.07.8/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
root@HL:~# opkg install zerotier
Installing zerotier (1.6.5-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.8/packages/mipsel_24kc/packages/zerotier_1.6.5-1_mipsel_24kc.ipkInstalling kmod-tun (4.14.241-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.8/targets/ramips/mt7621/packages/kmod-tun_4.14.241-1_mipsel_24kc.ipk
Installing libmnl0 (1.0.4-2) to root...
Downloading http://downloads.openwrt.org/releases/19.07.8/packages/mipsel_24kc/base/libmnl0_1.0.4-2_mipsel_24kc.ipk
Installing ip-tiny (5.0.0-2.1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.8/packages/mipsel_24kc/base/ip-tiny_5.0.0-2.1_mipsel_24kc.ipk
Configuring kmod-tun.
Configuring libmnl0.
Configuring ip-tiny.
Configuring zerotier.
disabled in config
root@HL:~#
```

fjqingyou, 2022-05-01:
Now I try to recompile the image and directly build kmod Tun in the image to see how the result will be

fjqingyou, 2022-05-01:
Thank you for your patience.

I recompile the firmware, select compile kmod Tun directly from the compile configuration, and install it.

Now it works well

closed 2022-05-01 by fjqingyou