openwrt/openwrt #9817: EAP225 outdoor v1 mac80211 steacktraces

Noki, 2022-05-01:
I have an EAP225 outdoor v1 running on OpenWrt 21.02-SNAPSHOT, r16554-1d4dea6d4f that frequently produces mac80211 stacktraces. Maybe someone can help to identify the underlying issues.

```
Sun May  1 12:43:46 2022 kern.warn kernel: [166205.849878] ------------[ cut here ]------------
Sun May  1 12:43:46 2022 kern.warn kernel: [166205.854791] WARNING: CPU: 0 PID: 13398 at backports-5.10.110-1/net/mac80211/sta_info.c:275 0x87688120 [mac80211@698d25e7+0x7f3a0]
Sun May  1 12:43:46 2022 kern.warn kernel: [166205.867155] sta_info_move_state() returned -5
Sun May  1 12:43:46 2022 kern.warn kernel: [166205.871791] Modules linked in: ath9k ath9k_common iptable_nat ath9k_hw ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_ct nft_counter nf_tables_set nf_tables nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY nfnetlink nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ipt_ECN ip_tables compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 gpio_button_hotplug
Sun May  1 12:43:46 2022 kern.warn kernel: [166205.950482] CPU: 0 PID: 13398 Comm: kworker/0:0 Tainted: G        W         5.4.188 #0
Sun May  1 12:43:46 2022 kern.warn kernel: [166205.958828] Workqueue: events_freezable 0x876807a4 [mac80211@698d25e7+0x7f3a0]
Sun May  1 12:43:46 2022 kern.warn kernel: [166205.966414] Stack : 80640000 85684680 80595e64 87688120 00000000 00000000 00000000 00000000
Sun May  1 12:43:46 2022 kern.warn kernel: [166205.975164]         00000000 00000000 00000000 00000000 00000000 00000001 823e5940 e40491b0
Sun May  1 12:43:46 2022 kern.warn kernel: [166205.983895]         823e59d8 00000000 00000000 00000885 00000038 8057c5e4 6d616338 30323131
Sun May  1 12:43:46 2022 kern.warn kernel: [166205.992627]         00000885 807b0000 00000000 64323565 00000000 823e5920 00000000 87688120
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.001368]         00000009 00000000 00000001 876491bc 00000002 803210b8 00000000 80790000
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.010101]         ...
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.012714] Call Trace:
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.012730] [<87688120>] 0x87688120 [mac80211@698d25e7+0x7f3a0]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.021696] [<8057c5e4>] 0x8057c5e4
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.025412] [<87688120>] 0x87688120 [mac80211@698d25e7+0x7f3a0]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.031693] [<803210b8>] 0x803210b8
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.035496] [<80069364>] 0x80069364
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.039224] [<8006936c>] 0x8006936c
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.042931] [<800824f4>] 0x800824f4
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.046643] [<87688120>] 0x87688120 [mac80211@698d25e7+0x7f3a0]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.052966] [<800825cc>] 0x800825cc
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.056744] [<87700000>] 0x87700000 [ath10k_core@7159513a+0x5a9f0]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.063322] [<87688120>] 0x87688120 [mac80211@698d25e7+0x7f3a0]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.069595] [<876899d0>] 0x876899d0 [mac80211@698d25e7+0x7f3a0]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.075829] [<803d7a70>] 0x803d7a70
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.079566] [<87689e60>] 0x87689e60 [mac80211@698d25e7+0x7f3a0]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.085825] [<876e495c>] 0x876e495c [mac80211@698d25e7+0x7f3a0]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.092070] [<876a3578>] 0x876a3578 [mac80211@698d25e7+0x7f3a0]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.098341] [<87638dfc>] 0x87638dfc [cfg80211@163cb28f+0x43e40]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.104603] [<8760292c>] 0x8760292c [cfg80211@163cb28f+0x43e40]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.110844] [<87602e08>] 0x87602e08 [cfg80211@163cb28f+0x43e40]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.117162] [<800aac0c>] 0x800aac0c
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.120902] [<800aa284>] 0x800aa284
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.124617] [<800ab8c0>] 0x800ab8c0
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.128336] [<80503528>] 0x80503528
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.132103] [<805067a0>] 0x805067a0
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.135846] [<805030d0>] 0x805030d0
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.139634] [<805066f4>] 0x805066f4
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.143394] [<80504174>] 0x80504174
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.147100] [<80504244>] 0x80504244
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.150900] [<80098c9c>] 0x80098c9c
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.154678] [<80097e20>] 0x80097e20
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.158394] [<8049c090>] 0x8049c090
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.162109] [<805066f4>] 0x805066f4
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.165894] [<805034e0>] 0x805034e0
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.169633] [<80560290>] 0x80560290
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.173353] [<800a08c0>] 0x800a08c0
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.177052] [<8050a628>] 0x8050a628
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.180769] [<800a09b8>] 0x800a09b8
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.184478] [<800a08c0>] 0x800a08c0
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.188180] [<803daf00>] 0x803daf00
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.191891] [<800a09b8>] 0x800a09b8
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.195622] [<803db044>] 0x803db044
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.199326] [<8006936c>] 0x8006936c
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.203038] [<803db108>] 0x803db108
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.206764] [<876c3988>] 0x876c3988 [mac80211@698d25e7+0x7f3a0]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.213000] [<876021ec>] 0x876021ec [cfg80211@163cb28f+0x43e40]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.219389] [<876c30c0>] 0x876c30c0 [mac80211@698d25e7+0x7f3a0]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.225668] [<8057f98c>] 0x8057f98c
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.229391] [<875a14d4>] 0x875a14d4 [nf_tables_set@b5565a11+0x4700]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.236036] [<87680870>] 0x87680870 [mac80211@698d25e7+0x7f3a0]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.242359] [<8057d7f8>] 0x8057d7f8
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.246138] [<80099c08>] 0x80099c08
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.249856] [<80099fd0>] 0x80099fd0
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.253640] [<80099e68>] 0x80099e68
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.257401] [<8057da30>] 0x8057da30
[... 114 lines cut ...]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.948354] [<8769c064>] 0x8769c064 [mac80211@698d25e7+0x7f3a0]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.954602] [<803daf54>] 0x803daf54
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.958413] [<800a09b8>] 0x800a09b8
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.962192] [<803db044>] 0x803db044
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.965906] [<8006936c>] 0x8006936c
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.969643] [<803db108>] 0x803db108
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.973356] [<876c3988>] 0x876c3988 [mac80211@698d25e7+0x7f3a0]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.979642] [<876021ec>] 0x876021ec [cfg80211@163cb28f+0x43e40]
Sun May  1 12:43:46 2022 kern.warn kernel: [166206.986009] [<876c30c0>] 0x876c30c0 [mac80211@698d25e7+0x7f3a0]
[... cut ...]
Sun May  1 12:43:46 2022 kern.warn kernel: [166207.027692] [<80099e68>] 0x80099e68
Sun May  1 12:43:46 2022 kern.warn kernel: [166207.031394] [<8009ed20>] 0x8009ed20
Sun May  1 12:43:46 2022 kern.warn kernel: [166207.035088] [<8009ebe8>] 0x8009ebe8
Sun May  1 12:43:46 2022 kern.warn kernel: [166207.038806] [<8009ebe8>] 0x8009ebe8
Sun May  1 12:43:46 2022 kern.warn kernel: [166207.042529] [<80064bf8>] 0x80064bf8
Sun May  1 12:43:46 2022 kern.warn kernel: [166207.046232]
Sun May  1 12:43:46 2022 kern.warn kernel: [166207.047858] ---[ end trace ac0d945835bcd682 ]---
```

pepe2k, 2022-05-04:
@Noki have you checked whether master and/or 22.03-rc1 also include this issue?

Cheers,
Piotr

Noki, 2022-05-04:
@pepe2k I haven't checked. It's a Freifunk Firmware build based on 21.02-SNAPSHOT and I can't upgrade it to master / 22.03-rc1 as migration of the firmware functions to 22.03 / master is not done yet.

pepe2k, 2022-05-04:
@Noki I'm not familiar with of modifications/changes done on top of OpenWrt in Freifunk thus I'm asking you to confirm the problem first on a unmodified/clean OpenWrt version. And please, next time at least mention that you are not running a generic OpenWrt image :)

Noki, 2023-01-10:
issue resolved by switching to ath10k-ct (https://github.com/freifunk-berlin/bbb-configs/pull/379)

closed 2023-01-10 by Noki