openwrt/openwrt #9822: Linksys E5600 (ramips/mt7621) fails to boot into openwrt.

chinmaythosar, 2022-05-02:
The E5600 fails to boot into openwrt. Upon 3 unsuccessful boot failures it will boot int alternate firmware. You can the boot the device back into openwrt by pressing the wps button immediately on power on (suggesting it can still boot into openwrt by forcing it). However upon reboot (either by power failure or rebooting via logging in) it will again fail to boot back into openwrt.

chinmaythosar, 2022-05-08:
The only way to open the case I think is to break it.

chinmaythosar, 2022-05-15:
I was able to flash it from stock on the latest stock firmware. I first installed the latest 21.x factory and then upgraded to 22.x rc. So far no issues. Seems like flashing incorrectly was the problem.

closed 2022-05-15 by chinmaythosar

smcpeck, 2023-09-13:
For anyone else who ends up here, it is clearly documented that an initial upgrade of the stock firmware is needed.

Reference: https://openwrt.org/toh/linksys/e5600#oem_easy_installation

> This device has 2 partitions for the firmware called `firmware` and `alt_firmware`. To successfully flash and boot the device, the device should have been running from `alt_firmware` partition. To get the device booted through `alt_firmware` partition, download the OEM firmware from Linksys website and upgrade the firmware from web GUI. Once this is done, flash the OpenWrt Factory firmware from web GUI.