openwrt/openwrt #9824: openwrt-22.03 bricks Samknows SK-WB8

isidorosss, 2022-05-02:
openwrt-22.03.0-rc1-ramips-mt7621-samknows_whitebox-v8-squashfs-sysupgrade.bin seems that bricks Samknows SK-WB8 (ramips/mt7621).
I flashed it on luci while on 21.02.2 and bricked it once. Then i unbricked it and flashed 21.02.3 then i tried one more time to flash 22.03 and bricked it again.
I'm pasting the code below.

`
## Booting image at bce30000 ...
Bad Magic Number,85190320
`
whole debug log here: [https://wtools.io/paste-code/bB8c](url)

pepe2k, 2022-05-04:
@isidorosss I don't know this device but looking at your logs clearly shows that the U-Boot is trying to boot from a wrong address/offset, see:

Booting from 0xbce30000 fails:
```
## Booting image at bce30000 ...
Bad Magic Number,85190320
```

Booting from 0xbc050000 works:
```
## Booting image at bc050000 ...
   Image Name:   MIPS OpenWrt Linux-5.4.188
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2381084 Bytes =  2.3 MB
   Load Address: 80001000
   Entry Point:  80001000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK
No initrd
## Transferring control to Linux (at address 80001000) ...
## Giving linux memsize in MB, 128
```
And at least once the image with 5.10 kernel (22.03 image?) was almost booted:
```
## Booting image at bc050000 ...
   Image Name:   MIPS OpenWrt Linux-5.10.111
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2694082 Bytes =  2.6 MB
   Load Address: 80001000
   Entry Point:  80001000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... LZMA ERROR 1 - must RESET board to recover
```
The `LZMA ERROR 1` could be related to kernel size and probably this board would require similar change as introduced in e83f7e5d76ebf74aa0a1f686d2941950fa5f22e2 and 12f3d1466ad783fded1747fa2ee521a9bea1ff73.

Cheers,
Piotr

closed 2022-05-13 by jow-

pepe2k, 2022-05-13:
@isidorosss I pushed a fix, to master and 22.03 branches. Could you try in a few days (when new images become available) a 22.03 snapshot from here: https://downloads.openwrt.org/releases/22.03-SNAPSHOT/targets/ramips/mt7621/ and let me know if it fixes the problem? Thanks!

Cheers,
Piotr

isidorosss, 2022-05-13:
@pepe2k I confirm that when a new snapshot is available, i'll check if everything works well on the device during the next days. Thanks a lot for the help Piotr!

isidorosss, 2022-05-14:
@pepe2k Hey Piotr! I flashed successfully OpenWrt 22.03-SNAPSHOT r19340-1c6a179e1a / LuCI openwrt-22.03 branch .
Everything seems to work fine! Thanks a lot!
[image]

pepe2k, 2022-05-15:
@isidorosss thanks for the feedback!

commit 064e7e57 (2022-05-13): ramips: fix booting on Samknows SK-WB8

This fixes a well known "LZMA ERROR 1" error, reported previously on
numerous of similar devices.

Fixes: #9824
commit 1c6a179e (2022-05-13): ramips: fix booting on Samknows SK-WB8

This fixes a well known "LZMA ERROR 1" error, reported previously on
numerous of similar devices.

Fixes: #9824

(cherry picked from commit 064e7e57b483e6879de0facef4f1fce86ec4ad47)