openwrt/openwrt #9827: IPv6 router advertisements stop after about 1 day

ianmtaylor1, 2022-05-03:
**Issue**: after about 1 day of router uptime, IPv6 unsolicited router advertisements stop being received by client devices.

**OpenWrt version:** 21.02.3
**Device:** Linksys WRT1900ACS

**Steps to Replicate:** 

1. Install OpenWRT 21.02.3 on a Linksys WRT1900ACS. 
2. After abut 24 hours of router up time, notice a drop in ipv6 connectivity. 
3. Open Wireshark on a client device (Windows 10 laptop) and wait for the max RA interval and don't receive any router advertisements. 
4. Disconnect and reconnect to wifi, get a router advertisement in response to a router solicitation, but still no unsolicited advertisements (i.e. with dest IP ff02::1).

**Additional Information:**

This causes client devices to lose their IPv6 default route and routes to other LAN subnets. They can still ping the router and their own /64. WAN IPv6 connectivity is still present, all the IPv6 ping and traceroute diagnostics in LuCI still work.

I noticed this after I updated from 19.07. I updated to 21.02.1 with a fresh configuration and recreated all interfaces manually, then updated to 21.02.3 by uploading the configuration archive from 21.02.1.

I created a firewall rule in `/etc/firewall.user` log outgoing router advertisements:

```
ip6tables -A output_rule -p icmpv6 --icmpv6-type router-advertisement -d ff02::1 -j LOG --log-prefix "UNSOLICITED RA:  "
ip6tables -A output_rule -p icmpv6 --icmpv6-type router-advertisement ! -d ff02::1 -j LOG --log-prefix "SOLICITED RA:  "
```

During periods when my device doesn't receive any unsolicited router advertisements, they are still being logged in the firewall at the expected intervals on every expected interface. So somewhere between traversing the firewall and going out on the wire/antenna, they are disappearing.

Is this related to #8239? Software flow offloading is not enabled on my device, but the symptoms of dropping IPv6 packets are similar.

**Workarounds:**

Rebooting the router restores router advertisements and default routes on all devices for a day or two until they stop again.

**Configuration:**

/etc/config/network (with some sensitive info redacted)

```
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fdxx:xxxx:xxxx::/48'

config interface 'lan'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '64'
        option ip6hint '01'
        option device 'bridge.101'

config device
        option name 'wan'

config interface 'wan'
        option proto 'dhcp'
        option device 'bridge.201'

config interface 'wan6'
        option proto '6rd'
        option peeraddr '---'
        option ip6prefix '2602::'
        option ip6prefixlen '24'

config device
        option type 'bridge'
[... 367 lines cut ...]
        option src_dport '32400'
        option dest 'services'
        option dest_ip '192.168.10.xxx'
        option dest_port '32400'

config zone
        option input 'REJECT'
        option output 'ACCEPT'
        option forward 'REJECT'
        list network 'dmz'
        option name 'dmzone'

config forwarding
        option dest 'wan'
        option src 'dmzone'

config forwarding
        option src 'lan'
        option dest 'dmzone'

config forwarding
        option src 'services'
        option dest 'dmzone'

```

tiagogaspar8, 2022-05-08:
Hi,

From the configuration I'm seeing you are using bridge vlan filtering right?

ianmtaylor1, 2022-05-08:
Yes, I combined all five of my Ethernet ports into a bridge and use vlan filtering on that bridge.

tiagogaspar8, 2022-05-08:
So, I've had similar issues with IPv6 and bridge vlan filtering. As it seems by default the switch looks at multicast for the bridge as a whole and not per vlan.
This commit adds this feature but I can't seem to find out how it works....

https://lwn.net/Articles/863487/

ianmtaylor1, 2022-05-08:
Thanks for this information, I'm glad I'm not the only one with this problem. Do you know if this commit will be included in future OpenWrt releases? I upgrade to 22.03.0-rc1, has this been included?

If it's not included in any future releases yet, are there any configuration workarounds to avoid using vlan filtering? For example, creating vlan interfaces on the individual ports (e.g. lan1.102) and bridging those into separate bridges for each interface?

ianmtaylor1, 2022-05-08:
A couple of interesting updates:

1. This might affect IPv4 broadcasts as well. I noticed that after IPv6 router advertisements stop, my phone's Roku app is unable to locate my Roku (IPv4-only device) despite being on the same subnet. After rebooting the router, they can find each other.
2. It might be not just after "about" one day that the advertisements stop, but exactly 1 day. I happened to have Wireshark open on my laptop today when they stopped arriving and the last one came at 23hr59min of router uptime.

ianmtaylor1, 2022-05-09:
I hope it's okay to answer my own question here, partially. One of the commits in your link, according to the linux kernel github repository, [made it into the linux kernel version 5.15 and on](https://github.com/torvalds/linux/commit/9632233e7de8da43711bb7cd3e054af32fedcc38). OpenWrt 22.03.0-rc1 is using 5.10.111, so I don't know if it will be included. It's not appearing anywhere in the [5.10.x changelogs](https://cdn.kernel.org/pub/linux/kernel/v5.x/) through 5.10.111.

ianmtaylor1, 2022-05-14:
I've reconfigured my router to use no bridge vlan filtering and and am still experiencing the issue (updated configuration below). I still have to create a vlan device since my ISP sends packets tagged with vlan id 201 for some reason but I don't have any filtering on the br-lan bridge.

With this new configuration, though, I've noticed something new. The most recent time when unsolicited RA's stopped being received by clients, it only affected wireless clients. I tested this by attaching my Windows 10 laptop both to the Wifi network and with ethernet with a Wireshark instance listening on each. The laptop received the router advertisements on the ethernet interface but not the wireless interface. I'm not sure if this was true before because I had never checked. For this reason I've also included my wireless configuration below.

Also, I increased the frequency of unsolicited RAs from the default (min 200 - max 600 seconds) to a min 60 seconds max 180 seconds, and that didn't improve the problem. In fact, the RAs stopped being received at the same 24 hour mark as before despite the increased frequency.

In summary, I still think these broadcast packets are disappearing below layer 3, but I no longer think bridge vlan filtering is the culprit. It also may be a wifi-specific issue, either with the wireless interfaces themselves or with the bridge containing the wireless interface.

---

Updated `/etc/config/network`:

```
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fdxx:xxxx:xxxx::/48'

config interface 'lan'
        option proto 'static'
        option ip6assign '64'
        option ipaddr '192.168.0.1'
        option ip6hint '00'
        option netmask '255.255.254.0'
        option device 'br-lan'

config device
        option name 'wan'

config interface 'wan'
        option proto 'dhcp'
        option device 'wan.201'
        option hostname '*'

config interface 'wan6'
        option proto '6rd'
        option peeraddr '---'
        option ip6prefix '2602::'
        option ip6prefixlen '24'
        option auto '0'

config interface 'henet'
        option proto '6in4'
        option peeraddr '---'
        option ip6addr '2001:yyyy:yyyy:yyyy::2/64'
        list ip6prefix '2001:zzzz:zzzz::/48'
        option tunnelid '---'
        option username '---'
        option password '----'
        option peerdns '0'
        list dns '---'
        list dns '---'
        option mtu '1480'

... wireguard config omitted ...

config device
        option type '8021q'
        option ifname 'wan'
        option vid '201'
        option name 'wan.201'

config device
        option type 'bridge'
        option name 'br-lan'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
```

Updated `/etc/config/dhcp`:

```
config dnsmasq
        option domainneeded '1'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option expandhosts '1'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
        option localservice '1'
        option ednspacket_max '1232'
        option local '/example.com/'
        option domain 'example.com'
        option local_ttl '60'

config dhcp 'lan'
        option interface 'lan'
        option dhcpv4 'server'
        option ra 'server'
        option leasetime '2h'
        option dhcpv6 'server'
        list ra_flags 'managed-config'
        list ra_flags 'other-config'
        option start '256'
        option limit '255'
        list dhcp_option '6,192.168.0.zzz'
        option ra_mininterval '60'
        option ra_maxinterval '180'
        list dns 'fdxx:xxxx:xxxx::zzz'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'
        list ra_flags 'none'

config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/tmp/hosts/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'
        option loglevel '4'

... static leases omitted ...
```

`/etc/config/wireless`:

```
config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11a'
        option path 'soc/soc:pcie/pci0000:00/0000:00:01.0/0000:01:00.0'
        option htmode 'VHT80'
        option country 'US'
        option cell_density '0'
        option channel '153'

config wifi-device 'radio1'
        option type 'mac80211'
        option hwmode '11g'
        option path 'soc/soc:pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
        option country 'US'
        option cell_density '0'
        option htmode 'HT20'
        option channel '11'

config wifi-iface 'wifinet0'
        option device 'radio0'
        option mode 'ap'
        option ssid 'myssid-5'
        option key 'verysecurewifipassword'
        option network 'lan'
        option encryption 'sae-mixed'

config wifi-iface 'wifinet1'
        option device 'radio1'
        option mode 'ap'
        option ssid 'myssid-2.4'
        option key 'verysecurewifipassword'
        option network 'lan'
        option encryption 'sae-mixed'
```

quarkysg, 2022-08-23:
@ianmtaylor1 are you still having this issue with IPv6 network disappearing?

I make my own build for my R7800 off the 21.02 branch, and I noticed a few months back that my router's IPv6 network disappears a few hours after my router rebooted.

For me, the issue seems to be the router did not get the RA multicast from my ISP.  To work-around this issue, I added the following command in my rc.local startup script:

`ip link set eth0.2 allmulticast on`

My wan interface is `eth0.2`, so you would have to replace it with whatever interface that you are using.

I don't remember having this issue before.  Something must have been updated in the 5.4 kernel that affected the multicast packets.

HTH.

patrakov, 2022-09-23:
R7800 mentioned -> this bug is not specific to mvebu. @pepe2k please remove the target/mvebu tag.

ianmtaylor1, 2022-09-27:
My IPv6 upstream is a 6in4 tunnel, so I'm not sure if your work-around would apply to me.

fate615030, 2022-11-18:
Hi @ianmtaylor1 

I also had a similar problem and found a solution. I modified the parameter of dnsmasq when it's running and added "--interface=br-lan".

Hope this information can help you.

kevinh-csalabs, 2022-11-23:
After a ton of digging I found that setting noserverunicast on the wan was the solution:
`option noserverunicast '1'`
https://blog.printk.io/2018/08/ipv6-renew-issue-with-fiber7-and-openwrt/

tiagogaspar8, 2023-01-06:
Does this fix the issue for anyone/everyone?

ianmtaylor1, 2023-01-06:
Thank you all! I will try to check soon, I don't have much spare time at the moment.

tiagogaspar8, 2023-01-06:
I'm asking because if it does solve the issue it could be added to the wiki 😁

ianmtaylor1, 2023-01-16:
I still haven't been able to try the solution from @kevinh-csalabs 's comment. After reading the linked post, I'm not sure it would help my specific case. My IPv6 upstream interface is a 6in4 tunnel, not a dhcpv6 client, and the issue I'm having is not with the router advertisements received from my ISP (which AFAIK don't happen with a 6in4 tunnel) but with the router advertisements that are supposed to be sent out from my router on the LAN interface.

Since I started having this issue, I've just been using 19.07.10 on which IPv6 worked fine. When going to reinstall 22.03 to try these solutions, I noticed the WRT1900ACS has been removed from 22.03.3 due to the MV88E6176 switch issue #11077. Now I have two options:

1. Install 22.03.**2** and try the `option noserverunicast '1'` fix.
2. Install snapshot, which upgrades to kernel 5.15. See if the new kernel version resolves the issue with its bugfixes on the switch. If not, try `option noserverunicast '1'` on snapshot.

Since it seems like using the WRT1900ACS on 22.03 is generally discouraged at this point, I'll use snapshot and report back.

tiagogaspar8, 2023-01-16:
I'd you're using 6in4 that won't work... Have you tried OpenWRT vanilla to see if the issue persists?
I am using 6in4 and I do not face that issue

ianmtaylor1, 2023-01-16:
What do you mean by OpenWRT vanilla?

tiagogaspar8, 2023-01-16:
I mean, restore the default OpenWrt configuration and then configure the bare minimum, eg. 6in4 tunnel, dns... then see if you can still reproduce the issue. It can be some random config that has nothing to do, been there 😄

ianmtaylor1, 2023-02-23:
Okay, I've tested the Linksys WRT1900ACS with a minimal config and I can't replicate the issue. Details about how to repeat my test below. I depend on the internet in my house because I work from home, so I purchased another cheap router to use in my home temporarily so that I could have the this one free to test 😄.

**Repeating the test**

These are the steps I performed. I'm not sure if all of them matter.

1. Install OpenWrt 21.02.5 on a Linksys WRT1900ACS (v1, but the image is the same for v2).
2. Install the luci-app-advanced-reboot app.
3. Upload the attached config archive (nothing sensitive is included, all passwords and keys are temporary for testing and not used anywhere else. Feel free to inspect before using it or I can paste the contents here).
4. Reboot the router.
5. On a client device connected to the OpenWrt-5 wifi network, open Wireshark and filter for icmpv6 type 134 messages (router advertisements).
6. Result behavior: Messages continue to be received beyond 24 hours after reboot, as expected.

**Test Notes**

The WAN here is a dhcpv6 client instead of a 6in4 tunnel. So it is unchanged from the default configuration and different than my configuration before. The test router will need to be able to get an IPv6 address and prefix delegation via DHCPv6 to repeat the test. For my test, the tested Linksys WRT1900ACS was connected via its wan port to the lan side of a Netgear WAX202, running OpenWrt 22.03.3.

**Other hypotheses/next steps**

I want to start building up to my earlier configuration one step at a time. I think the most relevant differences are probably the type of ipv6 upstream interface (6in4 tunnel vs dhcpv6) and the fact that the default 'wan6' interface was disabled in my previous config. The configurations I want to test with the changes that seem the most relevant are:

- 6in4 tunnel instead of DHCPv6 on the wan6 interface
- 6in4 tunnel on a second interface, with the default wan6 interface disabled
- 6in4 tunnel on a second interface, with the default wan6 interface disabled, and sae-mixed security on the wireless SSIDs.

If I'm unable to replicate the issue after making these changes, then I'll have to think of something else to test. Beyond these, the additional configuration changes I had made were for wireguard, static leases, dhcp options on the lan interface (lease time, DNS server), and ddns-scripts.

If you have any suggestions for things to check, I would be happy to hear them! Thank you for all of your help so far.

ianmtaylor1, 2023-02-24:
I decided to test the WiFi security first, and with sae-mixed security I was able to replicate the issue.

**Steps to Replicate**

1. Install OpenWrt 21.02.5 on a Linksys WRT1900ACS.
2. Install the luci-app-advanced-reboot app.
3. Upload the attached config archive (nothing sensitive is included, all passwords and keys are temporary for testing and not used anywhere else). It activates two Wifi SSIDs _with `sae-mixed` security_ and changes very little from the defaults. Feel free to inspect before using it or I can paste the contents here. [Config experiencing the issue: backup-wrt1900acs-2023-02-24.tar.gz](https://github.com/openwrt/openwrt/files/10825987/backup-wrt1900acs-2023-02-24.tar.gz)
4. Reboot the router.
5. On a client device connected to the OpenWrt-5 wifi network, open Wireshark and filter for icmpv6 type 134 messages (router advertisements).
6. Expected behavior: Router advertisements with destination IP `ff02::1` are received at semi-regular intervals. Result behavior: Messages are received between 200 and 600 seconds apart until 24 hours have passed, after which no more router advertisements are received.

Follow-up test: repeat the above except have the client device connected to the router via an ethernet cable to a LAN port. Result: router advertisements are received by the client device as expected.

**Conclusion**

Based on these two tests with psk2 vs sae-mixed security, I think the issue is with the wifi configuration. A workaround would be to use only psk2 security instead of sae or sae-mixed security. This is [already recommended on the OpenWrt device Wiki](https://openwrt.org/toh/linksys/wrt_ac_series#marvell_wifi).

closed 2023-02-24 by ianmtaylor1

kevinh-csalabs, 2023-02-24:
I don't doubt this but there are definitely other causes of this behavior as I've seen the IPv6 stop working where the only wifi is completely separate from the openwrt and all openwrt connections are wired.

ianmtaylor1, 2023-02-24:
@kevinh-csalabs Is the behavior you experience the same as I do? I.e. what happens when you use packet capture software like Wireshark on a client device to listen for router advertisements sent out of the lan side of your router? Do they stop after 24 hours as I've described? You may be experiencing the same or an unrelated issue.

ianmtaylor1, 2023-02-26:
I've found issues for other devices, due to frames with multicast destination MAC addresses not traversing the bridge to the wifi interface. Example here: https://github.com/openwrt/mt76/issues/410