openwrt/openwrt #9829: BT Home Hub 5A (HH5a) - 22.03.0-rc1 - crashes/reboots during boot up sequence

bill888uk, 2022-05-03:
I've witnessed an issue on two HH5a with v22.03.0-rc1 where the hubs 'may' some times take a considerable time to boot up.

Normally for HH5a, the power light flashes green colour then changes on constant blue after about 35 seconds with OpenWrt up and running.

However, with 22.03.0-rc1, the sequence starts off with flashing green colour, then the light may go out for considerable time, and the flashing green light appears, before eventually turning solid blue after about 2 minutes.

I reproduced the issue on HH5a which has a serial connection and captured the serial console.

**Use a pin to perform a hard reset.**

Here is partial log:
`

[   12.978795] random: ubusd: uninitialized urandom read (4 bytes read)
[   12.987617] random: ubusd: uninitialized urandom read (4 bytes read)
[   13.004014] procd: - init -
Please press Enter to activate this console.
[   14.538961] kmodloader: loading kernel modules from /etc/modules.d/*
[   14.938541] urngd: v1.0.2 started.
[   15.231591] IFXOS, Version 1.7.1 (c) Copyright 2009, Lantiq Deutschland GmbH
[   15.243113] NET: Registered protocol family 8
[   15.246019] NET: Registered protocol family 20
[   15.260745] PPP generic driver version 2.4.2
[   15.282942] Lantiq (VRX) DSL CPE MEI driver, version 1.5.17.6, (c) 2007-2015 Lantiq Beteiligungs-GmbH & Co. KG
[   15.293882]
[   15.293882]
[   15.293882] Lantiq CPE API Driver version: DSL CPE API V4.17.18.6
[   15.311616]
[   15.311616] Predefined debug level: 3
[   15.333878] Loading modules backported from Linux version v5.15.33-0-g06f50ca83ace
[   15.340137] Backport generated by backports.git v5.15.33-1-0-g183c4ab2
[   15.353225] Infineon Technologies DEU driver version 2.0.0
[   15.360608] IFX DEU DES initialized (multiblock).
[   15.367953] IFX DEU AES initialized (multiblock).
[   15.371762] IFX DEU SHA1 initialized.
[   15.375306] IFX DEU MD5 initialized.
[   15.388884] IFX DEU SHA1_HMAC initialized.
[   15.391978] IFX DEU MD5_HMAC initialized.
[   15.417049] NET: Registered protocol family 24
[   15.423093] ath9k_pci_owl_loader 0000:00:0e.0: enabling device (0000 -> 0002)
[   15.432846] ath9k_pci_owl_loader 0000:00:0e.0: fixup device configuration
[   36.488592] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[   36.493101] rcu:     1-...!: (1 GPs behind) idle=6d2/1/0x40000000 softirq=759/760 fqs=3
[   36.500915]  (detected by 0, t=2102 jiffies, g=-111, q=26)
[   36.506399] Sending NMI from CPU 0 to CPUs 1:

ROM VER: 1.1.4
CFG 06
NAND
NAND Read OK

ROM VER: 1.1.4
CFG 06
NAND
NAND Read OK

etc etc

`
**It eventually starts and completes booting of OpenWrt.**

**I power cycled the HH5a again.  It booted up normally within 35 seconds**

**I power cycled the HH5a once more, and captured the problem again**

**Here is partial log up to the point where router appears to crash and reboot**

`

Please press Enter to activate this console.
[   14.492110] kmodloader: loading kernel modules from /etc/modules.d/*
[   14.923248] urngd: v1.0.2 started.
[   15.192023] IFXOS, Version 1.7.1 (c) Copyright 2009, Lantiq Deutschland GmbH
[   15.204926] NET: Registered protocol family 8
[   15.207833] NET: Registered protocol family 20
[   15.221231] PPP generic driver version 2.4.2
[   15.243233] Lantiq (VRX) DSL CPE MEI driver, version 1.5.17.6, (c) 2007-2015 Lantiq Beteiligungs-GmbH & Co. KG
[   15.254917]
[   15.254917]
[   15.254917] Lantiq CPE API Driver version: DSL CPE API V4.17.18.6
[   15.272597]
[   15.272597] Predefined debug level: 3
[   15.293478] Loading modules backported from Linux version v5.15.33-0-g06f50ca83ace
[   15.299607] Backport generated by backports.git v5.15.33-1-0-g183c4ab2
[   15.312669] Infineon Technologies DEU driver version 2.0.0
[   15.319876] IFX DEU DES initialized (multiblock).
[   15.327406] IFX DEU AES initialized (multiblock).
[   15.331094] IFX DEU SHA1 initialized.
[   15.334913] IFX DEU MD5 initialized.
[   15.338324] IFX DEU SHA1_HMAC initialized.
[   15.342493] IFX DEU MD5_HMAC initialized.
[   15.368097] NET: Registered protocol family 24
[   15.373730] ath9k_pci_owl_loader 0000:00:0e.0: enabling device (0000 -> 0002)
[   15.383367] ath9k_pci_owl_loader 0000:00:0e.0: fixup device configuration
[   15.531332] random: crng init done
[   36.442384] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[   36.446890] rcu:     1-...!: (0 ticks this GP) idle=716/1/0x40000000 softirq=707/707 fqs=0
[   36.454966]  (detected by 0, t=2102 jiffies, g=-135, q=51)
[   36.460449] Sending NMI from CPU 0 to CPUs 1:

ROM VER: 1.1.4
CFG 06
NAND
NAND Read OK

ROM VER: 1.1.4
CFG 06
NAND
NAND Read OK

etc etc
`
**It eventually starts and completes booting of OpenWrt.**

When I install 21.02.01 on both HH5a, they behave and boot up normally in about 35 seconds when I power cycle the routers.

mkresin, 2022-05-03:
TL;DR confirmed

I haven't updated my tree for a very long time and only started to play with 22.03 last weekend. Despite a lot of reboots, I saw the issue only once during the weekend.

Different story today, I barely can get the HH5a to successfully boot.

So far I've no idea what causes CPU1 to hang. As far as I remember, the `ath9k_pci_owl_loader` triggers a rescan of the PCI bus and the ath9k chip should be detected.

```
[   16.143835] ath9k_pci_owl_loader 0000:00:0e.0: enabling device (0000 -> 0002)
[   16.158403] ath9k_pci_owl_loader 0000:00:0e.0: fixup device configuration
[   16.171044] pci 0000:00:0e.0: [168c:002d] type 00 class 0x028000
[   16.175778] pci 0000:00:0e.0: reg 0x10: [mem 0x18000000-0x1800ffff]
[   16.182008] pci 0000:00:0e.0: PME# supported from D0 D3hot
[   16.190225] pci 0000:00:0e.0: BAR 0: assigned [mem 0x18000000-0x1800ffff]
```

@xdarklight @hauke FYI

xdarklight, 2022-05-03:
I never caught a `Sending NMI from CPU 0 to CPUs 1` (on the serial console) but I am wondering if the "random reboots" which I have been seeing are caused by the same problem: https://openwrt.ebilan.co.uk/viewtopic.php?t=1223

bill888uk, 2022-05-03:
@xdarklight What version of OpenWrt were you using when you reported the "random reboots"?

I just installed latest snapshot.  (I did not Keep Settings).  After OpenWrt successfully started up, I power cycled the HH5a

OpenWrt SNAPSHOT, r19592-4509b790f0    (3rd May 2022)

`
[   15.228857] IFX DEU MD5_HMAC initialized.
[   15.257521] NET: Registered protocol family 24
[   15.263338] ath9k_pci_owl_loader 0000:00:0e.0: enabling device (0000 -> 0002)
[   15.279126] ath9k_pci_owl_loader 0000:00:0e.0: fixup device configuration
[   36.331931] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[   36.336421]  (detected by 0, t=2102 jiffies, g=-99, q=35)
[   36.341816] rcu: All QSes seen, last rcu_sched kthread activity 2102 (-26371--28473), jiffies_till_next_fqs=1, root ->qsmask 0x0
[   36.353394] rcu: rcu_sched kthread starved for 2102 jiffies! g-99 f0x2 RCU_GP_WAIT_FQS(5) ->state=0x0 ->cpu=0
[   36.363304] rcu:     Unless rcu_sched kthread gets sufficient CPU time, OOM is now expected behavior.
[   36.372266] rcu: RCU grace-period kthread stack dump:
[   36.377319] task:rcu_sched       state:R  running task     stack:    0 pid:   11 ppid:     2 flags:0x00100000
[   36.387239] Stack : 808eac50 80880000 80880000 800a2fec 00000000 81005d40 80c5be20 80c3bac0
[   36.395593]         80880000 00000001 00000001 808eac50 80880000 80880000 8087e1ec 806bf9cc
[   36.403948]         808eac50 80880000 80880000 8087e1ec ffff90c8 806c28ac 81014480 00000001
[   36.412304]         80880000 00000000 00000000 810124bc ffff90c8 800a24d4 02400001 80c3bac0
[   36.420660]         808eab20 808eab20 00000000 8009d0b4 80c5be94 8006d7e8 8093ad40 81005d40
[   36.429016]         ...
[   36.431455] Call Trace:
[   36.433933] [<806bf65c>] __schedule+0x3dc/0x6f8
[   36.438438] [<806bf9cc>] schedule+0x54/0xf8
[   36.442616] [<806c28ac>] schedule_timeout+0x68/0xe0
[   36.447495] [<8009d0b4>] rcu_gp_kthread+0x9b0/0xdb8
[   36.452371] [<800544d8>] kthread+0x13c/0x144
[   36.456633] [<80006eb8>] ret_from_kernel_thread+0x14/0x1c
[   36.462016]

ROM VER: 1.1.4
CFG 06
NAND
NAND Read OK

`

abajk, 2022-05-03:
On the current master, I have seen such reboots on average once every 10 starts.

xdarklight, 2022-05-03:
> @xdarklight What version of OpenWrt were you using when you reported the "random reboots"?

I always run `master` (custom image, not a snapshot).
From what I remember these reboots started at some point last year, but they are not consistent for me (in the past 4 weeks I didn't notice a single unexpected reboot).

Also I have not been able to reproduce these problems on the HH5A which I use for testing - yet, maybe I just need to reboot it more often?

mkresin, 2022-05-05:
It tend to say it's more likely to hit the issue if the HH5A does a cold boot (was powered off). Or in other words, I can't remember to have seen the issue after reboots.

But to be honest, it's more a gut feeling.

xdarklight, 2022-05-24:
I just hit this on my test HH5A with `maxvpes=1 maxtcs=1 nosmp` added to the bootargs.
That means at least that the secondary CPU core doesn't seem to be causing this issue.

I'll now try to reproduce this with
```
&pci0 {
    status = "disabled";
};
```

xdarklight, 2022-06-16:
I just had this reboot:
```
...
[   16.524815] xt_time: kernel timezone is -0000
[   16.531738] ath9k_pci_owl_loader 0000:00:0e.0: enabling device (0000 -> 0002)
[   16.541455] ath9k_pci_owl_loader 0000:00:0e.0: fixup device configuration
[   22.447972] CPU 1 Unable to handle kernel paging request at virtual address 00000000, epc == 00000000, ra == 00000000
[   22.457180] Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: console_unlock+0x548/0x548
[   22.467424] Rebooting in 1 seconds..

U-Boot SPL 2014.01-openwrt4-gcffe577 (Dec 02 2016 - 23:48:59)
SPL: applying tuned DDR SDRAM settings
MEM: DDR Echo DLL delay:  26/26
MEM: DDR Write DQS delay: 65/65
MEM: DDR Read DQS delay:  1f/1f
SPL: loading U-Boot to RAM
SPL: checking U-Boot image
SPL: invalid data CRC
SPL: cannot start U-Boot

ROM VER: 1.1.4
CFG 06
NAND
NAND Read OK

U-Boot SPL 2014.01-openwrt4-gcffe577 (Dec 02 2016 - 23:48:59)
SPL: applying tuned DDR SDRAM settings
MEM: DDR Echo DLL delay:  26/26
MEM: DDR Write DQS delay: 65/65
MEM: DDR Read DQS delay:  1f/1f
SPL: loading U-Boot to RAM
SPL: checking U-Boot image
SPL: decompressing U-Boot with LZO
SPL: jumping to U-Boot

U-Boot 2014.01-openwrt4-gcffe577 (Dec 02 2016 - 23:48:59) BTHOMEHUBV5A

Board: BT Home Hub 5A
SoC:   Lantiq VRX268 v1.2
CPU:   500 MHz
IO:    250 MHz
BUS:   250 MHz
BOOT:  NAND
DRAM:  128 MiB
NAND:  128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Phy not found
PHY reset timed out
ltq-eth
Hit any key to stop autoboot:  0 
UBI: attaching mtd1 to ubi0
...
```
It shows that after watchdog reset the system u-boot was unable to read the u-boot image.
After the next reset everything worked fine again.

With the PCI controller disabled I cannot reproduce the system lockup.
Also when booting from an initramfs image (meaning that NAND is basically never accessed) I am unable to reproduce the problem.
So I think the problem may be EBU access as this is what NAND and PCI have in common.

@hauke I don't understand EBU at all (I think it's some configurable bus access subsystem - but I have no idea how NAND and PCI are integrated). Do you have any idea/hint?

xdarklight, 2022-06-16:
I also tried adding the reset line for the PCI controller: [0001-lantiq-dts-Add-the-reset-line-for-the-PCI-controller.patch.txt](https://github.com/openwrt/openwrt/files/8922071/0001-lantiq-dts-Add-the-reset-line-for-the-PCI-controller.patch.txt)
And then make the PCI driver use it: [kernel-patch-reset-pci.patch.txt](https://github.com/openwrt/openwrt/files/8922069/kernel-patch-reset-pci.patch.txt)

The idea is to mimic what the ifxmips_pci vendor driver does. Unfortunately for me it doesn't seem to change anything - the hangs are still preset.

abajk, 2022-06-16:
> So I think the problem may be EBU access as this is what NAND and PCI have in common.

I have observed several times a situation where the "old image" could not boot properly. Then one time the router booted properly. The booting problems occurred 95% of the time. I later installed a "new image" using the sysupgrade command. The new image was identical to the old one. Then the problem went away for a while. After some time the problem returned and the situation began again.

These are observations from when I was testing the router extensively. I tested the stability and performance of the wired interfaces. During the tests I often uploaded different images and performed many reboots.

In addition to the panic mentioned in the thread, I've definitely seen two more panics from rcu related to "Unaligned access". I think it occurred once every few dozen compilations. Unfortunately I can't find that stack trace. If it is useful then I can try to look for that image.

xdarklight, 2022-06-20:
summary of a chat with @hauke (thanks a lot for your time again!):
- EBU is used to share the actual (electrical) pins for various EBU (external bus unit) interfaces
  - this includes NAND and PCI
  - "it handles a up to 26 address lines and up to 16 data lines"
  - PCI was only used till around xRX200 (VR9), xRX300 (AR10) doesn't have PCI support anymore
- this means: 1) either the PCI controller is configured incorrectly 2) the NAND controller is configured incorrectly 3) EBU is configured incorrectly 4-n) a mix of 1-3
- assumption: concurrent access of the PCI and NAND bus leads to lockups
- https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/mips/pci/pci-lantiq.c#n138
  - 0xa matches what the AVM driver does: they OR together IFX_PCI_CLK_CTRL_EBU_PCI_SWITCH_EN = 0x2 and IFX_PCI_CLK_CTRL_FPI_NORMAL_ACK = 0x8
  - this seems to be for auto-switching between EBU (NAND) and PCI
- it seems that there are two NAND controllers in xRX200 and later
  - or is it just that they added DMA transfers on top of the original NAND controller?
  - the AVM code has ifxmips_mtd_dmanand.c (only used on "VR9" and "AR10") as well as ifxmips_mtd_nand.c (used on all SoCs)
  - some google search for ifx_hsnand shows that AVM uses this driver on newer boards
- the EBU has some NAND specific extensions and it was extended for every SoC
- https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/mtd/nand/raw/intel-nand-controller.c
  - intel-nand-controller seems to be for a (slightly?) different "HSNAND" core
  - some xRX200 HSNAND registers seem similar but some are totally different (for example: IFX_RX_CNT seems to be gone in the new IP)
- TODO: fully compare the EBU setup in xway_nand with the UGW code but it looks fishy: UGW either uses LTQ_EBU_BUSCON0 or LTQ_EBU_BUSCON1 depending on "CS0" or "CS1". also we're not initializing LTQ_EBU_BUSCON1[15:14]. that's two differences I spotted... so I think it's worth digging further

mjsage, 2022-07-02:
Not sure if this is worth mentioning, as hopefully Darklight is on to the root cause, but I have noticed this issue seems to gets far worse when more / larger packages are included in the firmware sysupgrade.bin, to the point that a recent build I made from the master branch with lots of stuff, like wpad-wolfssl, kmod-batman-adv, luci, possibly wireguard, etc, etc, took over an hour to finally boot successfully.

As the 22.03.0-rc4 release was rebooting multiple times on the BT hub I tried it on, I built  22.03.0-rc4 so that I could use the  package feeds, but without including Luci and and most of the other optional packages in the firmware (I left out far more than intended as I inadvertently built them using the 'm' option in the .config), and after flashing the router and checking it would boot reliably, used opkg to install everything I'd omitted, including luci. 

I've rebooted it multiple times and **so far** haven't noticed it fail to boot first time.

xdarklight, 2022-07-04:
Dropping `target/linux/lantiq/patches-5.15/0018-MTD-nand-lots-of-xrx200-fixes.patch` didn't help and neither did dropping the `ebu_lock` lock `xway_select_chip()` help

The code from `target/linux/lantiq/patches-5.15/0018-MTD-nand-lots-of-xrx200-fixes.patch` is still strange as the vendor NAND driver has no references to PCI interfering with NAND.
Instead the vendor `drivers/mtd/maps/ifxmips_mtd_nor.c` driver uses `CONFIG_DANUBE_EBU_PCI_SW_ARBITOR`.
Also it's not clear to me why `target/linux/lantiq/files/arch/mips/boot/dts/lantiq/ar9_bt_homehub-v3a.dts` is enabling the `req-mask` hack for what is believed to be a workaround only for Danube boards (specifically the BT Home Hub 2B).

jonathanunderwood, 2022-08-01:
Is this one a release blocker for 22.03?

rbubley, 2022-08-02:
From the boards over the last few years it would seem there are a lot of these in use in the UK, as they are cheap (second hand) and functional.

OlyAndy, 2022-08-02:
Not sure if this is related, but the switch fails to initialize too. I use the imagebuilder to add in a few extra packages, but I can never get the 22.03 branch to work (tried up to RC6). I can connect to the hub via wifi, but the switch remains unoperational.

xdarklight, 2022-08-02:
> Not sure if this is related, but the switch fails to initialize too

In my opinion this is not related. Also this "works for me" so it would be great if you could create a separate ticket for this so we can investigate it separately

rbubley, 2022-09-08:
I’ve just experienced this reboot loop issue with the released version of 22.03 (standard upgrade not a custom build)

alfredonodo, 2022-09-10:
Hello,
I too am interested in this topic. I use many packages, so I think it is best to wait for some progress.

rbubley, 2022-09-10:
My experience using a HH5a with 22.03.0, configured as an Access Point with just a single lightweight package added (nano), was the same with booting sometimes working, sometimes repeating and eventually working and sometime repeating and then failing.

One point I noticed in the middle case (repeated reboots and then appearing to work) is that various things were not working quite right: e.g. the clock drifted (ntpd appeared to be running error free, yet not working), unexpected sporadic delays using LUCI, etc. 

The non-determinism variety of issues makes me wonder if there is a buffer overflow or corrupted pointer somewhere leading to random memory corruption.

patrakov, 2022-09-10:
I think that this might be related to the way how the voltages on the board are regulated.

My BT HomeHub 5A booted successfully in my house (after some attempts), but not in my friend's house that is in a different part of the city. Eventually I fixed this by replacing the power supply.

cybern0id, 2022-09-10:
My own anecdotal experience: the boot loop issue has occurred a multitude of times over the past year or so using various of my own custom builds from git master, my own custom builds from openwrt-21.02 branch as well as most of the 22.03 RC releases.

I just upgraded to 22.03 stable from 22.03 RC6 today, using the openwrt.org build-bot to include curl, dnsmasq-full, luci-ssl, luci-app-wireguard and luci-app-vpn-policy-routing packages and did not experience a boot loop afterwards.

I've tried various scenarios over the past year in an attempt to isolate the issue (thinking it was related to attached devices rather than a bug in the code) and I *consistently have no boot loop problems* if I first disable the Wifi on my Google Pixel 3 mobile phone or ensure this phone is not in the same room as the router.

I'm wondering whether having certain Wifi enabled client devices within short range of the router are interfering with the boot process after upgrade somehow?

(I swear there are no Voodoo or Druidic incantations involved)

mjsage, 2022-09-10:
I've made a non boot looping custom build of the 22.03.0 branch with a load of packages omitted and installed them after flashing with opkg from the repositories (using another router to connect).  As mentioned in my post above, I had noticed including more/larger packages increased the occurrence of repeated failures to boot. 

So far it has booted in about 60 seconds each time without getting stuck in a bootloop and I now must have tried rebooting the routers I flashed with it a couple of dozen times.  I've also not had any boot looping problem with previous builds done from release candidate branches in the same way, this one has got larger though so might not be as reliable.

If anyone want to try themselves, I've uploaded a copy of the image I flashed as well as a diffconfig, a manifest and the list of package I installed afterwards here:-

https://github.com/mjsage/bthub5a-22.03

jaimet, 2022-09-22:
Would there be value in bisecting this to find the offending commit? (I'm just trying to think of ways to progress this issue!) Or, alternatively, are people already reasonably certain where it started?

bill888uk, 2022-09-23:
fwiw, I tried the image created by mjsage 

Unfortunately, the issue still exists.

Turn off power.
Wait 5 seconds.
Turn on power.

For my HH5a, bootloop occurs perhaps 1 in every 5 cold boots with the mjsage image.

```
[   15.576657] random: 26 urandom warning(s) missed due to ratelimiting
[   15.585054] ath9k_pci_owl_loader 0000:00:0e.0: enabling device (0000 -> 0002)
[   15.595038] ath9k_pci_owl_loader 0000:00:0e.0: fixup device configuration

ROM VER: 1.1.4
CFG 06
NAND
NAND Read OK

ROM VER: 1.1.4
CFG 06
NAND
NAND Read OK
```

On one occasion, I did observe this
```
[   15.516192] ath9k_pci_owl_loader 0000:00:0e.0: enabling device (0000 -> 0002)
[   15.525519] ath9k_pci_owl_loader 0000:00:0e.0: fixup device configuration
[   15.556489] random: crng init done
[   15.558538] random: 26 urandom warning(s) missed due to ratelimiting
[   36.592243] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[   36.596744] rcu:     1-...!: (0 ticks this GP) idle=3b6/1/0x40000000 softirq=750/750 fqs=0
[   36.604818]  (detected by 0, t=2102 jiffies, g=-115, q=36)
[   36.610303] Sending NMI from CPU 0 to CPUs 1:
[   49.887153] rcu: rcu_sched kthread starved for 2102 jiffies! g-115 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x402 ->cpu=1
[   49.895880] rcu:     Unless rcu_sched kthread gets sufficient CPU time, OOM is now expected behavior.
[   49.904843] rcu: RCU grace-period kthread stack dump:
[   49.909894] task:rcu_sched       state:I stack:    0 pid:   11 ppid:     2 flags:0x00100000
[   49.918248] Stack : 80858c70 807f0000 807f0000 800a29dc 0000041a 00000020 80c5be20 80c3bac0
[   49.926602]         807f0000 00000001 00000001 80858c70 807f0000 807f0000 807ec1ec 806be55c
[   49.934957]         80858c70 807f0000 807f0000 807ec1ec ffff90e2 806c143c 81014500 00000001
[   49.943313]         807f0000 00000000 00000000 81012524 ffff90e2 800a1ec4 08c00001 80c3bac0
[   49.951669]         80858b40 80858b40 00000000 8009ca84 80c5be94 8006d1bc 808a9dc0 81005dc0
[   49.960025]         ...
[   49.962463] Call Trace:
[   49.962486] [<800a29dc>] 0x800a29dc
[   49.968395] [<806be55c>] 0x806be55c
[   49.971874] [<806c143c>] 0x806c143c
[   49.975356] [<800a1ec4>] 0x800a1ec4
[   49.978837] [<8009ca84>] 0x8009ca84
[   49.982316] [<8006d1bc>] 0x8006d1bc
[   49.985804] [<800a0000>] 0x800a0000
[   49.989278] [<800a0000>] 0x800a0000
[   49.992770] [<8009c0d4>] 0x8009c0d4
[   49.996245] [<80053f58>] 0x80053f58
[   49.999724] [<80053e1c>] 0x80053e1c
[   50.003207] [<80053e1c>] 0x80053e1c
[   50.006689] [<80053e1c>] 0x80053e1c
[   50.010169] [<80006e18>] 0x80006e18
[   50.013670]
[   99.642257] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[   99.646761] rcu:     1-...0: (0 ticks this GP) idle=3b6/1/0x40000000 softirq=750/750 fqs=2482
[   99.655097]  (detected by 0, t=8407 jiffies, g=-115, q=38)
[   99.660581] Sending NMI from CPU 0 to CPUs 1:
[   99.664926] Unable to send backtrace IPI to CPU1 - perhaps it hung?
```

As far as I'm aware, the boot loop issue has existed since the switch to DSA a year ago reported by another user.

mjsage, 2022-09-23:
Here's a comparison of boot times on the same BT hub 5A with the 22.03 release installed and the reduced size build I posted. 

I'm using default settings (keep settings not checked) with no cables connected, other than power, of course and timing from when I press the button on the back to when the power LED lights up solid blue. I'm typically leaving it off 5-10 seconds between boots. I did not include the first reboot after installing each firmware as I've never noticed the first reboot to fail. 

Boot time for: Openwrt Release Firmware:- OpenWrt 22.03.0 r19685-512e76967f / LuCI openwrt-22.03 branch git-22.245.77528-487e58a

1 minute 36 seconds.
1:36
50 seconds 
6:00+ +++gave up, stuck on solid green light, no network connection+++
0:47 
1:36
0:47
3:04
8 minutes+ +++gave up, repeatedly bootlooping+++            
47 seconds.  ***router was turned off for about 1 minute before this ***

My small build, default settings and I've not installed the modules I omitted.

33 seconds
33 seconds
33 seconds
33 seconds
33 seconds
33 seconds
33 seconds
33 seconds
33 seconds
33 seconds
33 seconds

After installing the modules

41 seconds
42 seconds
42 seconds
42 seconds
42 seconds

I thought my uploaded image was probably still occasionally rebooting early on in the start-up causing a slight delay from observing the LED, but didn't have serial access to check.  However, I did have a serial connected when I tried a dozen reboots on one of my previous builds which would have been smaller, and didn't see any evidence of a crash/reboot in the serial terminal, I suspect it just makes a crash during boot it far less likely to occur, rather than preventing it.

For what it is worth, I first noticed rare, occasional restarts during boot shortly after the change of kernel from 5.4 to 5,10 in the master branch, I'd been building firmware from the master branch with the DSA patch included for several months before that and didn't notice any reboots with 5.4 with DSA, however I wouldn't be surprised if it was just the increase in kernel size, and also I had been adding extra modules into my custom build each time such as wpad-openssl, batman-adv, openvpn etc... and eventually ended up with a build that took about an hour of bootlooping before booting up correctly.  

I would try creating a minimal build using the custom file option in the firmware selector, but I tried that a while back and ended up with no ethernet and having to connect to the serial pads.

jaimet, 2022-09-23:
> As far as I'm aware, the boot loop issue observed since the snapshots about a year ago, reported by another user. I don't know when Lantiq devices switched to kernel 5/DSA.

@bill888uk Interesting: release 21.02.2 (r16495) has linux 5.4.179 and dsa, and since I started using it (r16495), I must have rebooted it 50 times without running into this problem. If no-one else can trigger this problem on 21.02.2, then I don't think that this started with the switch to kernel 5 and dsa. If that is true (and that's my 2nd big "if") then perhaps a bisection might help...?

bill888uk, 2022-09-24:
@jaimet 21.02.x does Not use DSA.

alfredonodo, 2022-09-24:
Hi,
I have a couple of questions/considerations. Since the major changes between 22.03 and 21.02 are kernel 5.10 vs 5.4 and firewall nftables vs iptables:

1. Does BT work well with 21.02, kernel 5.4 and DSA?
2. Does BT work well with 22.03, kernel 5.10 without DSA?

According to what I read, the answer to the first question should be yes, while I don't know the second. If we can rule out the problem being DSA-related, then it will be the kernel. Otherwise, if the answer to the second question is positive, it could be a problem between the kernel 5.10 and DSA.

jaimet, 2022-09-24:
>     1. Does BT work well with 21.02, kernel 5.4 and DSA?

I _thought_ (simple past tense i.e. I used to think, but I am longer sure) that the stock release of 21.02.2 used/uses dsa. (I am not talking about mjsage's builds "with the DSA patch included" - instead I am talking about [the release build](https://downloads.openwrt.org/releases/21.02.2/targets/lantiq/xrx200/openwrt-21.02.2-lantiq-xrx200-bt_homehub-v5a-squashfs-sysupgrade.bin).)

I thought this, because there was a syntax change which I noticed in `/etc/config/network` - the following stanza makes its first appearance:

```
config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth0.1'

```
and I thought (incorrectly?) that the syntax change was due to the inclusion of dsa (the [dsa tutorial](https://openwrt.org/docs/guide-user/network/dsa/dsa-mini-tutorial) uses that exact syntax.)

For **anyone** who understands dsa sufficiently - how do I tell if the home hub 5a that I am connected to is using dsa or not? (Preferably a test with 100% [sensitivity and specificity](https://en.wikipedia.org/wiki/Sensitivity_and_specificity) please!)

bill888uk, 2022-09-24:
Official 21.02.x stable for Lantiq devices does NOT use DSA.  Period.
In LuCI, swconfig still exists in **LuCI > Network > Switch.** in 21.02.x.

21.02.x does introduce new UCI syntax for **/etc/network/config**.

22.03.x is first time DSA introduced to Lantiq devices
[https://openwrt.org/releases/22.03/notes-22.03.0#more_targets_converted_to_dsa](url)

alfredonodo, 2022-09-24:
Hi,
so that rules out my first question. Is it possible to compile 22.03 without DSA or with it disabled?

jaimet, 2022-09-25:
> P.S. the device has 128 MBytes of RAM and NAND, how can this be a problem for loading the kernel? The [minimum requirements](https://openwrt.org/supported_devices) are 64/8 MBytes.

Why do you think that the size of RAM / NAND would be a problem for the kernel? (As far as I am aware, it isn't.)

alfredonodo, 2022-09-25:
> > P.S. the device has 128 MBytes of RAM and NAND, how can this be a problem for loading the kernel? The [minimum requirements](https://openwrt.org/supported_devices) are 64/8 MBytes.
> 
> Why do you think that the size of RAM / NAND would be a problem for the kernel? (As far as I am aware, it isn't.)

According to @mjsage message:

> however I wouldn't be surprised if it was just the increase in kernel size, and also I had been adding extra modules into my custom build each time such as wpad-openssl, batman-adv, openvpn etc..

I do not believe that increasing the size of the kernel can be the cause of the problem.

mjsage, 2022-09-25:
>

> I do not believe that increasing the size of the kernel can be the cause of the problem.

I'm not suggesting it is the underlying cause, read Xdarklight's posts for that. 

I'm just stating that I have observed that increasing the number of built-in packages in custom builds makes rebooting happen far more frequently, and removing default packages in recent release candidates and the current release makes it far less, or not at-all, on the three routers I've tested it on.  I was speculating that the (presumably) larger kernel in 22.03  may have helped tip the balance in making default builds prone to rebooting. 

Also, Cybern0id's reports observing the boot loop issue in his own custom builds from openwrt-21.02 branch, which suggests it is not DSA and not the 5.10 kernel itself. Presumably the bug is present in all openwrt releases for the BT hub 5,even if it previously hadn't reared its ugly head before, unless the code for handling the Infineon EBU bus has been changed at some point?

cybern0id, 2022-09-25:
> Also, Cybern0id's reports observing the boot loop issue in his own custom builds from openwrt-21.02 branch, which suggests it is not DSA and not the 5.10 kernel itself. Presumably the bug is present [...]

I wouldn't take my non-precise anecdotal report too seriously. My recollections were from memory only and if I do actually remember correctly, my 21.02- builds had all sorts of cherry-picked patches, uncommitted pull-requests and backported code added. Unfortunately the SSD containing all my builds died suddenly recently so I can't retest with them.

However, I did commission a newly acquired HH5A yesterday, sys-upgrading to 22.03 from the outset during the install process. I then tested a multitude of cold boots and captured several serial dumps when it started boot looping. The errors I got are the same as others have already reported. My "mobile phone in the same room" theory isn't as consistent as I had originally thought as it had no effect on this bug yesterday.

```
[   16.502536] NET: Registered protocol family 24
[   16.538436] xt_time: kernel timezone is -0000
[   16.544639] ath9k_pci_owl_loader 0000:00:0e.0: enabling device (0000 -> 0002)
[   16.558976] ath9k_pci_owl_loader 0000:00:0e.0: fixup device configuration
[   16.699796] random: crng init done
[   16.701752] random: 27 urandom warning(s) missed due to ratelimiting
[   10.242504] CPU 1 Unable to handle kernel paging request at virtual address 00000000, epc == 00000000, ra == 00000000
[   10.251676] CPU 1 Unable to handle kernel paging request at virtual address 00000000, epc == 00000000, ra == 00000000
[   10.262254] Oops[#1]:
[   10.264519] CPU: 1 PID: 890 Comm: kworker/u4:3 Not tainted 5.10.138 #0
[   10.271018] $ 0   : 000 890 Comm: kworker/u4:3 Not taint
NAND
NAND Read OK

ROM VER: 1.1.4
CFG 06
NAND
NAND Read OK
```

What I noticed is that if you enter fail-safe mode by pressing the [f] key then [enter] at the designated time, which interrupts the boot process before the second kmodloader is run, the device seems to behave. It drops to a terminal where I can run various commands and interact with the OS. Running mount_root works as expected but running`kmodloader` manually results in the same bug, albeit not triggering a reboot, just a repeat of the following or similar:
```
[   37.721410] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[   37.725915] rcu: 	1-...!: (1 GPs behind) idle=656/1/0x40000000 softirq=845/848 fqs=0 
[   37.733728] 	(detected by 0, t=2102 jiffies, g=25, q=22)
[   37.739038] Sending NMI from CPU 0 to CPUs 1:
```

Based on this, my money's on a bug in a kernel module or the kernel itself, rather than it being an image size or kernel size problem. Perhaps EBU, PCI, NAND, ath9k interaction as mentioned previously.

It would be interesting to see how git master with 5.15.x experimental kernel fares - I'll try a build sometime this week.

Some more serial console captured errors follow...

```
[   16.650286] ath9k_pci_owl_loader 0000:00:0e.0: enabling device (0000 -> 0002)
[   16.681445] ath9k_pci_owl_loader 0000:00:0e.0: fixup device configuration
[   16.717579] random: crng init done
[   16.719589] random: 27 urandom warning(s) missed due to ratelimiting
[   37.732513] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[   37.737017] rcu: 	1-...!: (1 ticks this GP) idle=3d2/1/0x40000000 softirq=850/851 fqs=0 
[   37.745091] 	(detected by 0, t=2102 jiffies, g=1, q=30)
[   37.750314] Sending NMI from CPU 0 to CPUs 1:
```

```
[   16.595990] xt_time: kernel timezone is -0000
[   16.601164] ath9k_pci_owl_loader 0000:00:0e.0: enabling device (0000 -> 0002)
[   16.615836] ath9k_pci_owl_loader 0000:00:0e.0: fixup device configuration
[   22.438541] CPU 1 Unable to handle kernel paging request at virtual address 00000004, epc == 00000004, ra == 00000004
[   22.447703] Oops[#1]:
[   22.449963] CPU: 1 PID: 887 Comm: kworker/u4:0 Not tainted 5.10.138 #0
[   22.456460] $ 0   : 00000000 00800108 00000000 00000000
[   22.461667] $ 4   : 807df2b8 00000002 fffffffe 00000020
[   22.466876] $ 8   : 00000ffffe0 80c08400fffffe 00000000
```

```
[   16.567672] xt_time: kernel timezone is -0000
[   16.573790] ath9k_pci_owl_loader 0000:00:0e.0: enabling device (0000 -> 0002)
[   16.588823] ath9k_pci_owl_loader 0000:00:0e.0: fixup device configuration
[   16.767200] random: crng init done
[   16.769177] random: 27 urandom warning(s) missed due to ratelimiting
[... 31 lines cut ...]
[   37.729846] [<8006d1bc>] 0x8006d1bc
[   37.733334] [<800a0000>] 0x800a0000
[   37.736809] [<800a0000>] 0x800a0000
[   37.740301] [<8009c0d4>] 0x8009c0d4
[   37.743775] [<80053f58>] 0x80053f58
[   37.747255] [<80053e1c>] 0x80053e1c
[   37.750738] [<80053e1c>] 0x80053e1c
[   37.754219] [<80053e1c>] 0x80053e1c
[   37.757699] [<80006e18>] 0x80006e18
[   37.761200] 
```

```
[   16.581740] xt_time: kernel timezone is -0000
[   16.587959] ath9k_pci_owl_loader 0000:00:0e.0: enabling device (0000 -> 0002)
[   16.603022] ath9k_pci_owl_loader 0000:00:0e.0: fixup device configuration
[   16.725134] random: crng init done
[   16.727092] random: 27 urandom warning(s) missed due to ratelimiting
[   10.263013] CPU 1 Unable to handle kernel paging request at virtual address 00000000, epc == 00000000, ra == 00000000
[   10.272186] CPU 1 Unable to handle kernel paging request at virtual address 00000000, epc == 00000000, ra == 00000000
[   10.282763] Oops[#1]:
[   10.285030] CPU: 1 PID: 896 Comm: kworker/u4:0 Not tainted 5.10.138 #0
[   10.291527] $ U: 1 PID: 896 Comm: kworker/u4:0 Not taint
NAND
```

alfredonodo, 2022-09-26:
> Based on this, my money's on a bug in a kernel module or the kernel itself, rather than it being an image size or kernel size problem. Perhaps EBU, PCI, NAND, ath9k interaction as mentioned [...]

I agree and think the 5.15.x kernel is an interesting test.

jaimet, 2022-09-26:
Here's something. I used the 22.03.0 imagebuilder to build a "minimal" image (I never use LuCI, so that's ok by me) and I flashed it onto a spare 5A. I then rebooted it few times, and sure enough, got the bootloops. I then started rem'ing out modules in `/etc/modules.d` to see what would happen. When I rem'd out `ath9k_pci_owl_loader` which is the only line in `/etc/modules.d/owl-loader` then hey, no more bootloops!

Could someone else please confirm whether they can reproduce this?

bill888uk, 2022-09-27:
If you can post your 22.03 imagebuilder minimal image, I can quickly test it on my spare HH5a for bootloops.

jaimet, 2022-09-27:
> If you can post your 22.03 imagebuilder minimal image, I can quickly check it on my spare Hub One for bootlooping.

[Here you go.](https://github.com/openwrt/openwrt/files/9652175/openwrt-22.03.0-lantiq-xrx200-bt_homehub-v5a-squashfs-sysupgrade.bin.zip) It's just the result of running `make image PROFILE="bt_homehub-v5a"` and so it has the minimal set of packages. The md5sum ends in `0089`.

(Quick guide to re-flashing _without_ LuCI: push the new rom onto the router using ssh: `dd if=<localRomFilename> | ssh <routerIP> dd of=/tmp/s.bin` then ssh into the router and flash the rom using `sysupgrade -n -v /tmp/s.bin`. If you get an `Image version mismatch` error message, you can force the flash using `sysupgrade -F -n -v /tmp/s.bin`. HTH)

bill888uk, 2022-09-27:
I installed you image using LuCI from 21.02.3.  Forced install. Do not Keep Settings.

Upon restart, it reported 22.03.0 installed at serial console.

I then Power Cycled the 2021 Hub One.

Bootloop appeared.

```
 -----------------------------------------------------
 OpenWrt 22.03.0, r19685-512e76967f
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:/#

************ Power Cycled

ROM VER: 1.1.4
CFG 06
NAND
NAND Read OK

DDR autotuning Rev 0.3d
DDR size from 0xa0000000 - 0xa7ffffff
DDR check ok... start booting...

U-Boot 2010.06-LANTIQ-v-2.2.46 (Feb 11 2015 - 16:18:32 on bat@rowlf-jungo)

CLOCK CPU 500M RAM 250M
secure boot
DRAM:  125 MiB
NAND:  ONFI flash detected
ONFI param page 0 valid
NAND device: Manufacturer ID: 0xef, Chip ID: 0xf1 (Unknown W29N01HV)
128 MiB
Bad block table found at page 65472, version 0x01
Bad block table found at page 65408, version 0x01
In:    serial
Out:   serial
Err:   serial
Net:   Internal phy(GE) firmware version: 0x841b
vr9 Switch

Type "run flash_nfs" to mount root filesystem over NFS

Hit any key to stop autoboot:  0
Creating 1 MTD partitions on "nand0":
0x000000100000-0x000007f80000 : "mtd=0"
UBI: attaching mtd1 to ubi0
UBI: physical eraseblock size:   131072 bytes (128 KiB)
UBI: logical eraseblock size:    129024 bytes
UBI: smallest flash I/O unit:    2048
UBI: sub-page size:              512
UBI: VID header offset:          512 (aligned 512)
UBI: data offset:                2048
UBI: attached mtd1 to ubi0
UBI: MTD device name:            "mtd=0"
UBI: MTD device size:            126 MiB
UBI: number of good PEBs:        1012
UBI: number of bad PEBs:         0
UBI: max. allowed volumes:       128
UBI: wear-leveling threshold:    4096
UBI: number of internal volumes: 1
UBI: number of user volumes:     4
UBI: available PEBs:             10
UBI: total number of reserved PEBs: 1002
UBI: number of PEBs reserved for bad PEB handling: 10
UBI: max/mean erase counter: 8/5
Volume kernel found at volume id 0
read 0 bytes from volume 0 to 80800000(buf address)
Read [2709504] bytes
## Booting kernel from Legacy Image at 80800000 ...
   Image Name:   MIPS OpenWrt Linux-5.10.138
   Created:      2022-09-03   2:55:34 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2697269 Bytes = 2.6 MiB
   Load Address: 80002000
   Entry Point:  80002000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Linux version 5.10.138 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r19685-512e76967f) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Sat Sep 3 02:55:34 2022
[    0.000000] SoC: xRX200 rev 1.2
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019556 (MIPS 34Kc)
[    0.000000] MIPS: machine is BT Home Hub 5A
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Detected 1 available secondary CPU(s)

.....

[   12.115344] procd: - early -
[   12.117198] procd: - watchdog -
[   12.819345] procd: - watchdog -
[   12.823414] procd: - ubus -
[   12.908100] random: ubusd: uninitialized urandom read (4 bytes read)
[   12.916368] random: ubusd: uninitialized urandom read (4 bytes read)
[   12.922170] random: ubusd: uninitialized urandom read (4 bytes read)
[   12.939026] procd: - init -
Please press Enter to activate this console.
[   14.172119] random: jshn: uninitialized urandom read (4 bytes read)
[   14.255656] random: ubusd: uninitialized urandom read (4 bytes read)
[   14.413375] kmodloader: loading kernel modules from /etc/modules.d/*
[   14.745487] urngd: v1.0.2 started.
[   14.922075] IFXOS, Version 1.7.1 (c) Copyright 2009, Lantiq Deutschland GmbH
[   14.933304] NET: Registered protocol family 8
[   14.936241] NET: Registered protocol family 20
[   14.950388] PPP generic driver version 2.4.2
[   14.976140] Lantiq (VRX) DSL CPE MEI driver, version 1.5.17.6, (c) 2007-2015 Lantiq Beteiligungs-GmbH & Co. KG
[   14.989635]
[   14.989635]
[   14.989635] Lantiq CPE API Driver version: DSL CPE API V4.17.18.6
[   15.007339]
[   15.007339] Predefined debug level: 3
[   15.029050] Loading modules backported from Linux version v5.15.58-0-g7d8048d4e064
[   15.035327] Backport generated by backports.git v5.15.58-1-0-g42a95ce7
[   15.048438] Infineon Technologies DEU driver version 2.0.0
[   15.056693] IFX DEU DES initialized (multiblock).
[   15.064876] IFX DEU AES initialized (multiblock).
[   15.068665] IFX DEU SHA1 initialized.
[   15.082243] IFX DEU MD5 initialized.
[   15.084908] IFX DEU SHA1_HMAC initialized.
[   15.088959] IFX DEU MD5_HMAC initialized.
[   15.118926] NET: Registered protocol family 24
[   15.125334] ath9k_pci_owl_loader 0000:00:0e.0: enabling device (0000 -> 0002)
[   15.134654] ath9k_pci_owl_loader 0000:00:0e.0: fixup device configuration
[   15.342368] random: crng init done
[   15.344321] random: 26 urandom warning(s) missed due to ratelimiting
[    9.147163] CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 00000000, ra == 00000000
 Read OK
[    9.156331] CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 00000000, ra == 00000000
[    9.156343] Oops[#1]:
[    9.156361] CPU: 0 PID: 821 Comm: kworker/u4:3 Not tainted 5.10.138 #0
[    9.156367] $ 0   : 00000000 00000001 00000069 808586ec
AND
NAND Read OK

ROM VER: 1.1.4
CFG 06
NAND
NAND Read OK

ROM VER: 1.1.4
CFG 06
NAND
NAND Read OK
```

Out of 6 times I power cycled, the router, it boot looped 4 times.

jaimet, 2022-09-27:
> Out of 6 times I power cycled, the router, it boot looped 4 times.

Perfect. What do you get if you rem-out the single line in `/etc/modules.d/owl-loader` ?

bill888uk, 2022-09-27:
My bad. I didn't realise the 'fix' had to be applied to your image.

I've REMmed out the single lin in /etc/modules.d/owl-loader, and can confirm your image subsequently powered up from cold for 5 times.   Taking 41 seconds each time when viewed via serial link.

I'm still waiting for stock 22.03.0 to start normally so I can try same fix....

jonathanunderwood, 2022-09-27:
This [message](https://www.spinics.net/lists/linux-wireless/msg187670.html) contains a nice description of what that module is doing:

```
On Thursday, June 27, 2019 8:13:47 PM CEST Kalle Valo wrote:
> Christian Lamparter <chunkeey@xxxxxxxxx> writes:
> > Atheros cards with a AR92XX generation (and older) chip usually
> > store their pci(e) initialization vectors on an external eeprom chip.
> > However these chips technically don't need the eeprom chip attached,
> > the AR9280 Datasheet in section "6.1.2 DEVICE_ID" describes that
> > "... if the EEPROM content is not valid, a value of 0xFF1C returns
> > when read from the register". So, they will show up on the system's
> > pci bus. However in that state, ath9k can't load, since it relies
> > on having the correct pci-id, otherwise it doesn't know what chip it
> > actually is. This happens on many embedded devices like routers
> > and accesspoint since they want to keep the BOM low and store the
> > pci(e) initialization vectors together with the calibration data
> > on the system's FLASH, which is out of reach of the ath9k chip.
> >
> > Furthermore, Some devices (like the Cisco Meraki Z1 Cloud Managed
> > Teleworker Gateway) need to be able to initialize the PCIe wifi device.
> > Normally, this should be done as a pci quirk during the early stages of
> > booting linux. However, this isn't possible for devices which have the
> > init code for the Atheros chip stored on NAND in an UBI volume.
> > Hence, this module can be used to initialize the chip when the
> > user-space is ready to extract the init code.
```

So that (perhaps) ties together the observation that not loading `ath9k_pci_owl_loader` prevents the bootloops with @xdarklight 's earlier suggestion that the crashes occur during simultaneous PCI and UBI access - that's exactly what the implicated module is trying to do.

jaimet, 2022-09-27:
Does anyone know what we do now? :rofl:

petzah, 2022-09-27:
I can confirm that commenting/disabling load of `ath9k_pci_owl_loader` module fixes the kernel oops (bootloop). Good find @jaimet !

```
root@btr6nq:~# cat /etc/modules.d/owl-loader 
#ath9k_pci_owl_loader
```
As a dirty workaround to load this module back I have added hotplug file:
```
root@btr6nq:~# cat /etc/hotplug.d/firmware/50-bug9829 
#!/bin/sh
modprobe ath9k_pci_owl_loader
```

I'm running recent(yesterday) snapshot:
```
root@btr6nq:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r20753-4ed90e84f8'
DISTRIB_TARGET='lantiq/xrx200'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r20753-4ed90e84f8'
DISTRIB_TAINTS=''

```

jaimet, 2022-09-27:
> As a dirty workaround to load this module back I have added hotplug file:
> 
> ```
> root@btr6nq:~# cat /etc/hotplug.d/firmware/50-bug9829 
> #!/bin/sh
> modprobe ath9k_pci_owl_loader
> ```

Exactly! I've just done 10 "reboot/login/modprobe" cycles, and I've got a successful ath9k phy (without accompanying reboot) every time. It's definitely a timing issue.

cybern0id, 2022-09-27:
I'd be surprised if the hotplug file is a permanent fix as it doesn't 
address the concurrent / conflicting PCI + NAND access which appears to 
be the root cause of this issue and is counter to my own experience 
described previously - when I entered fail safe mode (loads kernel 
without modules) I could interact with shell etc without issue, I could 
run mount_root, but as soon as I ran kmodloader, which loads modules in 
/etc/modules.d/* i.e. ath9k_pci_owl_loader etc (basically the same as 
the hotplug file) I got no ath9k phy, only the reported kernel errors 
repeating.

cybern0id, 2022-09-27:
> 
> Exactly! I've just done 10 "reboot/login/modprobe" cycles, and I've got a successful ath9k phy (without accompanying reboot) every time. It's definitely a timing issue.

@jaimet @petzah Could you perhaps try repeated cold boot rather than reboot / warm boot just to check you still don't experience the bug? - by all accounts this bug seems to manifest more often during cold boot.

jaimet, 2022-09-27:
> @jaimet @petzah Could you perhaps try repeated cold boot rather than reboot / warm boot just to check you still don't experience the bug? - by all accounts this bug seems to manifest more often during cold boot.

Will do. Back in a bit.

petzah, 2022-09-27:
@cybern0id 
I've just done 3 cold boots and got no bootloop. Must be some kind of race condition as having ath9 drivers loading just little bit later (e.g via hotplug) is preventing it somehow.
```
root@btr6nq:~# dmesg |grep ath
[   14.473568] ath10k_pci 0000:02:00.0: enabling device (0000 -> 0002)
[   14.478632] ath10k_pci 0000:02:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   15.216207] ath9k_pci_owl_loader 0000:00:0e.0: enabling device (0000 -> 0002)
[   15.224035] ath9k_pci_owl_loader 0000:00:0e.0: fixup device configuration
[   15.837319] ath10k_pci 0000:02:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[   15.845240] ath10k_pci 0000:02:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   15.863174] ath10k_pci 0000:02:00.0: firmware ver 10.2.4-1.0-00047 api 5 features no-p2p,raw-mode,mfp,allows-mesh-bcast crc32 35bd9258
[   16.957803] ath10k_pci 0000:02:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[   18.009700] ath10k_pci 0000:02:00.0: htt-ver 2.1 wmi-op 5 htt-op 2 cal file max-sta 128 raw 0 hwcrypto 1
[   18.145247] ath: EEPROM regdomain: 0x833a
[   18.145264] ath: EEPROM indicates we should expect a country code
[   18.145282] ath: doing EEPROM country->regdmn map search
[   18.145293] ath: country maps to regdmn code: 0x37
[   18.145302] ath: Country alpha2 being used: GB
[   18.145309] ath: Regpair used: 0x37
[   18.210484] ath9k 0000:00:0e.0: enabling device (0000 -> 0002)
[   18.222211] ath: phy1: Ignoring endianness difference in EEPROM magic bytes.
[   18.229665] ath: EEPROM regdomain: 0x833a
[   18.229689] ath: EEPROM indicates we should expect a country code
[   18.229711] ath: doing EEPROM country->regdmn map search
[   18.229730] ath: country maps to regdmn code: 0x37
[   18.229747] ath: Country alpha2 being used: GB
[   18.229759] ath: Regpair used: 0x37
[   44.498910] ath10k_pci 0000:02:00.0: pdev param 0 not supported by firmware
[   49.873762] ath10k_pci 0000:02:00.0: pdev param 0 not supported by firmware
[   49.879314] ath10k_pci 0000:02:00.0: failed to enable peer stats info: -122
```

cybern0id, 2022-09-27:
Interesting! Maybe module load order is important too? I see that with your hotplug file, the ath10k driver loads before the ath9k one, which is the opposite of a normal boot.

jaimet, 2022-09-27:
> Will do. Back in a bit.

All done on [22.03.0 stock release](https://downloads.openwrt.org/releases/22.03.0/targets/lantiq/xrx200/openwrt-22.03.0-lantiq-xrx200-bt_homehub-v5a-squashfs-sysupgrade.bin) **with `ath9k_pci_owl_loader` rem'd out**.

Power-off, power-on, boot to solid blue led after 47 seconds, consisting of:

1) 14 seconds of solid green for uboot/uboot wait, followed by
2) 4 seconds of fast flashing green for failsafe trigger wait, followed by
3) 29 seconds of slow flashing green for normal boot.

Login, manually run `modprobe ath9k_pci_owl_loader` and check phy (`iw phy | grep Wiphy | wc -l`).

The whole lot repeated 10 times. 100% success.

xdarklight, 2022-09-27:
> Does anyone know what we do now?

I think we know more than before your experiments:
- your approach confirms that PCI communication is related to our issue (which we already suspected)
- your patch rules out the PCI controller's initialization code as removing `ath9k_pci_owl_loader` still results in the PCI controller getting probed (this finding is new to me)

So thank you for this idea and getting a few more insights :-)

I believe that delaying when `ath9k_pci_owl_loader` gets loaded doesn't fully fix the problem.
On my own HH5A I sometimes also have random reboots after a few days/hours of uptime.

I noticed that the out-of-tree UGW PCI driver sets `IFX_PCI_CLK_CTRL_FPI_NORMAL_ACK` (bit 3 when counting from 0) in IFX_PCI_CLK_CTRL. Doing this on mainline results in bus errors. Also `ifx_pci_core_int_isr` in UGW handles a "PCI timeout interrupt" which seems "interesting". Whether any of these are related to our issue: I don't know.
Unfortunately I don't have much time (for any investigation) at the moment. I'll try to work on it mid October when I have a few days off.

jaimet, 2022-09-27:
> So thank you for this idea and getting a few more insights :-)

Thank _you_ - it's a pleasure to be able to contribute back into the project.

> I believe that delaying when `ath9k_pci_owl_loader` gets loaded doesn't fully fix the problem. On my own HH5A I sometimes also have random reboots after a few days/hours of uptime.

I presume that more people will install 22.03.x now that we have a work-around for the boot-looping - it will be interesting to get more feedback regarding uptime and stability.

> Unfortunately I don't have much time (for any investigation) at the moment. I'll try to work on it mid October when I have a few days off.

No worries. The work-around is easy to apply, so we're good for the time being. Thanks again.

cybern0id, 2022-09-27:
Building on everyone's great suggestions and discoveries so far, I've done some more testing and have found that moving `/etc/modules.d/owl-loader` to earlier in the boot process also works (for me at least), but most importantly, moving it to before `/etc/modules.d/51-ltq-vdsl-vr9`.

e.g.

`mv /etc/modules.d/owl-loader /etc/modules.d/50-owl-loader`

I've done 10x cold boots and 10x warm reboot with no boot loops. Will probably require more testing and bootings.

If I set owl-loader to 51-owl-loader or any number later, I still get intermittent boot loops.

Perhaps a race condition between drv_dsl_cpe_api and ath9k_pci_owl_loader kernel modules?

bill888uk, 2022-09-28:
I tried @cybern0id suggestion on my 2021 Hub One but it didn't work well for me.
`mv /etc/modules.d/owl-loader /etc/modules.d/50-owl-loader`

On 3rd cold boot, I encountered:

`
[   15.513900] ath9k_pci_owl_loader 0000:00:0e.0: enabling device (0000 -> 0002)
[   15.523283] ath9k_pci_owl_loader 0000:00:0e.0: fixup device configuration
[   15.643584] random: crng init done
[   15.645569] random: 26 urandom warning(s) missed due to ratelimiting
[   36.582529] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[   36.587020]  (detected by 0, t=2102 jiffies, g=-111, q=36)
[   36.592502] rcu: All QSes seen, last rcu_sched kthread activity 2102 (-26346--28448), jiffies_till_next_fqs=1, root ->qsmask 0x0
[   36.604081] rcu: rcu_sched kthread starved for 2102 jiffies! g-111 f0x2 RCU_GP_WAIT_FQS(5) ->state=0x200 ->cpu=1
[   36.614252] rcu:     Unless rcu_sched kthread gets sufficient CPU time, OOM is now expected behavior.
[   36.623214] rcu: RCU grace-period kthread stack dump:
[   36.628267] task:rcu_sched       state:R stack:    0 pid:   11 ppid:     2 flags:0x00100000
[   36.636620] Stack : 80858c70 807f0000 807f0000 800a29dc 00000000 81005dc0 80c5be20 80c3bac0
[   36.644973]         807f0000 00000001 00000001 80858c70 807f0000 807f0000 807ec1ec 806be55c
[   36.653329]         80858c70 807f0000 807f0000 807ec1ec ffff90e1 806c143c 81014500 00000001
[   36.661685]         807f0000 00000000 00000000 81012520 ffff90e1 800a1ec4 08800001 80c3bac0
[   36.670041]         80858b40 80858b40 00000000 8009ca84 80c5be94 8006d1bc 808a9dc0 81005dc0
[   36.678397]         ...
[   36.680834] Call Trace:
[   36.680859] [<800a29dc>] 0x800a29dc
[   36.686766] [<806be55c>] 0x806be55c
[   36.690245] [<806c143c>] 0x806c143c
[   36.693728] [<800a1ec4>] 0x800a1ec4
[   36.697208] [<8009ca84>] 0x8009ca84
[   36.700688] [<8006d1bc>] 0x8006d1bc
[   36.704175] [<800a0000>] 0x800a0000
[   36.707650] [<800a0000>] 0x800a0000
[   36.711142] [<8009c0d4>] 0x8009c0d4
[   36.714616] [<80053f58>] 0x80053f58
[   36.718096] [<80053e1c>] 0x80053e1c
[   36.721579] [<80053e1c>] 0x80053e1c
[   36.725060] [<80053e1c>] 0x80053e1c
[   36.728540] [<80006e18>] 0x80006e18
[   36.732041]

`
Flashing green power LED.
I had to turn router off and on.

Repeated cold booting a few more times and did encounter the usual boot loop.

bill888uk, 2022-09-28:
I tried @cybern0id suggestion on my 2021 Hub One.
`mv /etc/modules.d/owl-loader /etc/modules.d/50-owl-loader`

It seems to work for me, but...

When I factory reset the router and execute these commands which I presume performs the same action (I'm no master of LInux), something weird happens.
`
cd /etc/modules.d
ls
mv owl-loader 50-owl-loader
ls
`
I see just 50-owl-loader listed.

When I reboot, I see ubifs errors
```
[   13.558150] UBIFS error (ubi0:2 pid 663): 0x80271094: failed to read inode 214, error -2
[   13.564930] UBIFS error (ubi0:2 pid 663): 0x8026c5b4: dead directory entry 'owl-loader', error -2
[   13.573806] UBIFS warning (ubi0:2 pid 663): 0x802745f0: switched to read-only mode, error -2
[   13.582118] CPU: 0 PID: 663 Comm: find Not tainted 5.10.138 #0
[   13.587950] Stack : 00000000 00000000 8074c1cc 81e3f710 00000000 00000000 00000000 00000000
[   13.596298]         00000000 00000000 00000000 00000000 00000000 00000001 81e1bc10 d6ef7c59
[   13.604654]         81e1bca8 00000000 00000000 81e1bab0 00000038 8038f6c4 00000000 ffffffea
[   13.613010]         000000db 81e1babc 000000db 807f46e0 81e1bbf0 8071a8bc 80000000 80720000
[   13.621361]         81cd3000 81e50600 00000094 81e3f710 00000018 00000030 00000000 809b0000
[   13.629721]         ...
[   13.632154] Call Trace:
[   13.632175] [<8038f6c4>] 0x8038f6c4
[   13.638088] [<8000b938>] 0x8000b938
[   13.641561] [<8000b940>] 0x8000b940
[   13.645048] [<803749d4>] 0x803749d4
[   13.648524] [<8026c5b4>] 0x8026c5b4
[   13.652006] [<8026c5c0>] 0x8026c5c0
[   13.655492] [<801bf96c>] 0x801bf96c
[   13.658973] [<801af248>] 0x801af248
[   13.662453] [<801b028c>] 0x801b028c
[   13.665939] [<801b8aa8>] 0x801b8aa8
[   13.669413] [<802abf1c>] 0x802abf1c
[   13.672895] [<8019bb78>] 0x8019bb78
[   13.676384] [<802acb94>] 0x802acb94
[   13.679858] [<802ad8c8>] 0x802ad8c8
[   13.683346] [<802ac76c>] 0x802ac76c
[   13.686825] [<802ad8f8>] 0x802ad8f8
[   13.690307] [<801b8998>] 0x801b8998
[   13.693790] [<8019d5e4>] 0x8019d5e4
[   13.697269] [<801b901c>] 0x801b901c
[   13.700748] [<801b8c1c>] 0x801b8c1c
[   13.704238] [<80018678>] 0x80018678
[   13.707719]
[   13.719727] UBIFS error (ubi0:2 pid 663): 0x80271094: failed to read inode 214, error -2
[   13.726477] UBIFS error (ubi0:2 pid 663): 0x8026c5b4: dead directory entry 'owl-loader', error -2
[   14.467647] random: jshn: uninitialized urandom read (4 bytes read)
[   14.513580] random: ubusd: uninitialized urandom read (4 bytes read)
[   14.811598] kmodloader: loading kernel modules from /etc/modules.d/*
[   14.817218] UBIFS error (ubi0:2 pid 740): 0x80271094: failed to read inode 214, error -2
[   14.824684] UBIFS error (ubi0:2 pid 740): 0x8026c5b4: dead directory entry 'owl-loader', error -2
[   15.154955] urngd: v1.0.2 started.
```
and when I list the /etc/modules.d folder, I see both! 
`50-owl-loader
owl-loader`
I'm unable to delete owl-loader.

I've repeated this two further times and see same results.

Firstboot and Reset button on back of router does not restore factory settings.  I can use LuCI to reinstall 22.03.0 sysupgrade.bin with Keep Settings set to off, to restore factory default settings.

alanswanson, 2022-09-30:
The only significant change to the ath9k owl loader for years is commit [ath9k: nvmem for ath9k caldata](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=d91318662d5c290ce97bfb18c84e4cddc514f33a) which seems to correspond with when boot loops started. (This was upstreamed into Linux kernel 5.16). So maybe a NVMEM access conflict or timing with drv_dsl_cpe_api not releasing perhaps but not familiar enough to say. Commit possibly only tested on standard routers.

Disabling the owl_nvmem_probe() as an workaround (without reordering module loading) should hopefully also work.

xdarklight, 2022-09-30:
> Disabling the owl_nvmem_probe() as an workaround (without reordering module loading) should hopefully also work.

we cannot use NVMEM to access the ath9k caldata since it's part of a UBI volume (and NVMEM access to data inside a UBI volume is not supported). so disabling `owl_nvmem_probe()` is a no-op (unless there's a serious memory corruption bug inside it which even affects systems where NVMEM is not used)

[54 more replies]

closed 2023-04-22 by bill888uk