openwrt/openwrt #9837: io error

f4nff, 2022-05-05:
`Linux version 5.10.111 (builder@buildhost) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r19302-df622768da) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Wed Apr 20 10:37:37 2022`

download links：[https://downloads.openwrt.org/releases/22.03.0-rc1/targets/x86/64/openwrt-22.03.0-rc1-x86-64-generic-ext4-combined-efi.img.gz](https://downloads.openwrt.org/releases/22.03.0-rc1/targets/x86/64/openwrt-22.03.0-rc1-x86-64-generic-ext4-combined-efi.img.gz)

ssh log in to openwrt, enter top, then keep it on and wait, after about 30 minutes, there will be an io error, you can only restart openwrt.

jow-, 2022-05-05:
IO error reported by what exactly? By top? In the syslog? Is the system running on bare metal or virtualized?

f4nff, 2022-05-06:
I run on r4s and x86 physical hosts, and this happens. Basically, I connect to ssh, and then always open mtr or top. It will appear after a while.