openwrt/openwrt #9842: [Zyxel NBG-419n v2 / 21.02.x boot fails / bootloop

juhench, 2022-05-06:
I would like to report that the kernel from the downloads section (all, the factory image, sysupgrade, and the kernel that comes with the image builder don't boot.

I have been able to prepare a working kernel when compiling from source with the build system by setting some kernel options. In particular, when deselecting the ramdisk feature, bootable kernels result.

See also forum entry. [Zyxel NBG-419n v2 / 21.02.x boot fails (bootloop)](https://forum.openwrt.org/t/zyxel-nbg-419n-v2-21-02-x-boot-fails-bootloop/119566)

Once a bootable kernel has been loaded, everything else works fine for me, at least the packages I have tried.

I have not yet tried 22.x images.

Ntemis, 2022-05-07:
I confirm the issue on this model too. Tried all 21.x based firmwares and none worked!
19.07 and < works fine.

Ntemis, 2022-05-17:
@juhench https://www.androidfilehost.com/?fid=14655340768118475218
@pepe2k The issue is the lzma compression tool, using lz4 worked fine.
Maybe lzma support was removed by mistake?

Ntemis, 2022-05-18:
@DragonBluep yes this appears to be the fix, thank you

juhench, 2022-05-18:
@DragonBluep - Thank you very much. Can this be fixed in future releases to restore the functionality of the pre-compiled images that are publicly available?

DragonBluep, 2022-05-19:
If you can confirm that this is the correct fix, maybe @pepe2k will be happy to help us submit a commit to upstream. Also link to #8964

juhench, 2022-05-19:
Thanks. Just to confirm:
I will download the current buildsystem, modify the file
`target/linux/ramips/image/rt305x.mk`
by adding 
`$(Device/uimage-lzma-loader)`
in the respective section, which will then look like:
```
define Device/zyxel_nbg-419n-v2
  $(Device/uimage-lzma-loader)
  SOC := rt3352
  IMAGE_SIZE := 7872k
  DEVICE_VENDOR := ZyXEL
  DEVICE_MODEL := NBG-419N
  DEVICE_VARIANT := v2
  SUPPORTED_DEVICES += nbg-419n2
endef
```
I will then build the image without modification and flash a 419n-v2 device and test. Do you agree?

DragonBluep, 2022-05-23:
Yes, you can have a try. In recently PR #9916, it is very similar to Zyxel NG-419N's hardware and I believe the lzma-Loader is a useful fix.

pepe2k, 2022-05-23:
@juhench @DragonBluep ok, please let me know if that fixes the problem and whether it should be applied also to `22.03` and `master` branches. Also, if you like, please add here a line with your e-mail and full name:

```
Reported-by: Full Name <e-mail>
Tested-by: Full Name <e-mail>
```

I will include that in the commit with the fix.

Cheers,
Piotr

juhench, 2022-05-23:
Dear @DragonBluep, @pepe2k, and @Ntemis: I can confirm that the fix, adding `$(Device/uimage-lzma-loader)`, solves the problem. Thanks a lot for your effort. I have just tested it on an NBG-419n v2.

The image (sysupgrade) was flashable and booted without problems!

```
Reported-by: Juergen Hench <jurgen.hench@gmail.coml>
Tested-by: Juergen Hench <jurgen.hench@gmail.coml>
```

pepe2k, 2022-05-23:
@juhench did you try only `21.02` or also `22.03` and `master`?

juhench, 2022-05-23:
@pepe2k : I tried only `21.02.1`. I have not yet tried anything else yet.

Here is what I did:
```
# Download and update the sources
git clone https://git.openwrt.org/openwrt/openwrt.git
cd openwrt
git pull

# Select a specific code revision
git branch -a
git tag
git checkout v21.02.1

# Update the feeds
./scripts/feeds update -a
./scripts/feeds install -a

# Configure the firmware image and the kernel
#make menuconfig

# or use the default config
wget https://downloads.openwrt.org/releases/21.02.1/targets/ramips/rt305x/config.buildinfo -O .config

make -j $(nproc) kernel_menuconfig
# used the UI to select NBG-419n v2

# Build the firmware image
make -j $(nproc) defconfig download clean world
```

Just let me know which other versions I should try, i.e., adaptations to the above script.

closed 2022-06-20 by jow-

pepe2k, 2022-06-20:
Fix pushed to `master` and backported to `22.03` and `21.02`: https://github.com/openwrt/openwrt/commit/fd72e595c2b2a46bab8cbc7e9415fbfeae7b5b0d, https://github.com/openwrt/openwrt/commit/2726c8c3151dc09614ed12a0e8729934e602821e, https://github.com/openwrt/openwrt/commit/09a35619b85ecb381368a5a344a7f981ccde6ca9

Sorry for the delay!

Cheers,
Piotr

commit fd72e595 (2022-06-20): ramips: fix booting on ZyXEL NBG-419N v2

This fixes a well known "LZMA ERROR 1" error, reported previously on
numerous of other devices from 'ramips' target.

Fixes: #9842
Fixes: #8964

Reported-by: Juergen Hench <jurgen.hench@gmail.com>
Tested-by: Juergen Hench <jurgen.hench@gmail.com>
commit 2726c8c3 (2022-06-20): ramips: fix booting on ZyXEL NBG-419N v2

This fixes a well known "LZMA ERROR 1" error, reported previously on
numerous of other devices from 'ramips' target.

Fixes: #9842
Fixes: #8964

Reported-by: Juergen Hench <jurgen.hench@gmail.com>
Tested-by: Juergen Hench <jurgen.hench@gmail.com>

(cherry picked from commit fd72e595c2b2a46bab8cbc7e9415fbfeae7b5b0d)
commit 09a35619 (2022-06-20): ramips: fix booting on ZyXEL NBG-419N v2

This fixes a well known "LZMA ERROR 1" error, reported previously on
numerous of other devices from 'ramips' target.

Fixes: #9842
Fixes: #8964

Reported-by: Juergen Hench <jurgen.hench@gmail.com>
Tested-by: Juergen Hench <jurgen.hench@gmail.com>

(cherry picked from commit fd72e595c2b2a46bab8cbc7e9415fbfeae7b5b0d)