openwrt/openwrt #9847: firewall4: regress? interface wildcard

obsy, 2022-05-07:
Nftables allow to use wildcard in interface name, ex:
```
root@OpenWrt:/tmp# nft add rule inet fw4 input_wan iifname eth* tcp dport 22 cou
nter 
root@OpenWrt:/tmp# nft list ruleset | grep "dport 22"
		iifname "eth*" tcp dport 22 counter packets 0 bytes 0
```
But when I use it in network config, like:
```
config interface 'vpn'
	option device 'tun*'
	option proto 'static'
	option ipaddr '192.168.3.1'
	option netmask '255.255.255.0'
```
fw4 produces this ruleset:
```
root@OpenWrt:~# nft list ruleset | grep vpn
	chain input_vpn {
		jump reject_from_vpn
	chain output_vpn {
		jump accept_to_vpn
	chain forward_vpn {
		jump accept_to_lan comment "!fw4: Accept vpn to lan forwarding"
		jump reject_to_vpn
	chain accept_to_vpn {
	chain reject_from_vpn {
	chain reject_to_vpn {
	chain srcnat_vpn {
		meta nfproto ipv4 masquerade comment "!fw4: Masquerade IPv4 vpn traffic"
```

For normal name:
```
config interface 'vpn'
	option device 'tun1'
	option proto 'static'
	option ipaddr '192.168.3.1'
	option netmask '255.255.255.0'
```

```
root@OpenWrt:~# nft list ruleset | grep vpn
		iifname "tun1" jump input_vpn comment "!fw4: Handle vpn IPv4/IPv6 input traffic"
		iifname "tun1" jump forward_vpn comment "!fw4: Handle vpn IPv4/IPv6 forward traffic"
		oifname "tun1" jump output_vpn comment "!fw4: Handle vpn IPv4/IPv6 output traffic"
	chain input_vpn {
		jump reject_from_vpn
	chain output_vpn {
		jump accept_to_vpn
	chain forward_vpn {
		jump accept_to_lan comment "!fw4: Accept vpn to lan forwarding"
		jump reject_to_vpn
	chain accept_to_vpn {
		oifname "tun1" counter packets 0 bytes 0 accept comment "!fw4: accept vpn IPv4/IPv6 traffic"
	chain reject_from_vpn {
		iifname "tun1" counter packets 0 bytes 0 jump handle_reject comment "!fw4: reject vpn IPv4/IPv6 traffic"
	chain reject_to_vpn {
		oifname "tun1" counter packets 0 bytes 0 jump handle_reject comment "!fw4: reject vpn IPv4/IPv6 traffic"
		oifname "tun1" jump srcnat_vpn comment "!fw4: Handle vpn IPv4/IPv6 srcnat traffic"
	chain srcnat_vpn {
		meta nfproto ipv4 masquerade comment "!fw4: Masquerade IPv4 vpn traffic"
```

Shouldn't fw4 generate rules for interface with wildcard?

jow-, 2022-05-07:
fw4 uses `+` instead of `*` as wildcard character to stay backwards compatible with existing uci configurations.

closed 2022-05-07 by jow-

obsy, 2022-05-07:
No, + not working. Tested as above.

obsy, 2022-05-07:
```
root@OpenWrt:/tmp# fw4 print | grep vpn
	chain input_vpn {
		jump reject_from_vpn
	chain output_vpn {
		jump accept_to_vpn
	chain forward_vpn {
		jump accept_to_lan comment "!fw4: Accept vpn to lan forwarding"
		jump reject_to_vpn
	chain accept_to_vpn {
	chain reject_from_vpn {
	chain reject_to_vpn {
	chain srcnat_vpn {
		meta nfproto ipv4 masquerade comment "!fw4: Masquerade IPv4 vpn traffic"
```
```
config interface 'vpn'
	option device 'tun+'
	option proto 'static'
	option ipaddr '192.168.3.1'
	option netmask '255.255.255.0'
````

jow-, 2022-05-07:
Ah you use it in interface sections. That will not work. The firewall acquires its state from ubus/netifd which requires specific device names. You can only use wildcard ifnames in zone declarations of the firewall config.