openwrt/openwrt #9848: [nftables] Poor network performance when using nftables with ipq40xx

PussAzuki, 2022-05-07:
When using fw3 and software offload on, I got around 680Mbps in speedtest test.

When using fw3 and software offload off, I got around 260Mbps in speedtest test.

When using fw4 and software offload on, I got around 220~260Mbps in speedtest test.

When using fw4 and software offload off, I got around 190~220Mbps in speedtest test.

When using fw4 and software offload off, and using iptables-nft with 
`iptables -A FORWARD -m conntrack --ctstate RELATED,ESTABLISHED -j FLOWOFFLOAD` 
I got around 220Mbps in speedtest test.

I'm sure I have kmod-ipt-offload and kmod-nft-offload installed.

As a comparison, when using [kmod-natflow](https://github.com/ptpt52/natflow) and adjust the irq of tx/rx, I got around 900+Mbps in speedtest test, and this solution is valid when using fw3/fw4.

I do think there is an error in the nftables based offload, is there a solution that would allow me to debug what is wrong?

abajk, 2022-05-07:
@PussAzuki 
[On lantiq](https://github.com/openwrt/openwrt/pull/9733#issuecomment-1116483329) I also saw a significant drop in performance after switching to nftables.
Try these commands:
```
$ nft delete flowtable inet fw4 ft { devices = { br-lan }\; }
$ nft add flowtable inet fw4 ft { devices = { lan1, lan2, lan3, lan4, wlan0, wlan1 }\; }
```

abajk, 2022-05-07:
This Issue is similar to #9676.

PussAzuki, 2022-05-07:
But ipq40xx is still using essedma, not converted to DSA, so only br-lan no lanX.

https://github.com/openwrt/openwrt/pull/4721 <- This is the DSA driver PR for ipq40xx, I have tried to use it. The default is also to bridge lanX together into br-lan.

When using br-lan on DSA, the performance is similar to essedma, but even with kmod-natflow, the performance is only 860Mbps max, which is still relatively worse...

https://github.com/openwrt/openwrt/commit/726ef8ba2dbe4d4a693c4d9300bc69e234e6d67d At least this one doesn't seem to solve the current problem. I also tried the experimental 5.15 kernel.

abajk, 2022-05-07:
> But ipq40xx is still using essedma, not converted to DSA, so only br-lan no lanX.

What does the brctl show command show on your device?

> #4721 <- This is the DSA driver PR for ipq40xx, I have tried to use it. The default is also to bridge lanX together into br-lan.

It's ok. The ports should be added to br-lan. The idea is to also add all ports belonging to br-lan to flowtable, instead of br-lan.

PussAzuki, 2022-05-07:
[image]

abajk, 2022-05-07:
> [image]

Thanks.
What the command shows:
``

PussAzuki, 2022-05-07:
[image]

abajk, 2022-05-07:
Try these commands (consider that their state is not stored during reset):
```
$ nft delete flowtable inet fw4 ft { devices = { br-lan }\; }
$ nft add flowtable inet fw4 ft { devices = { eth0 }\; }
```

abajk, 2022-05-07:
I am also using one device that has not been converted to DSA, when I have a free moment I will check if it works on it too.

abajk, 2022-05-07:
After these operations, the flowtable should look like this:
```
root@LEDE:~#  nft list flowtable  inet fw4 ft
table inet fw4 {
        flowtable ft {
                hook ingress priority filter
                devices = { br-lan, cf, eth0, eth1, pppoe-wan }
        }
}
```

PussAzuki, 2022-05-07:
[image]
[image]

still bad ...

abajk, 2022-05-07:
Are you using kernel 5.10 or 5.15?

PussAzuki, 2022-05-07:
5.15

PussAzuki, 2022-05-07:
[image]

abajk, 2022-05-07:
According to [1]:
```
You do not need to add the PPPoE and the VLAN devices to your flowtable, instead the real device is sufficient for the flowtable to track your flows.
```
What is your real WAN device?

1. https://www.kernel.org/doc/html/latest/networking/nf_flowtable.html

abajk, 2022-05-07:
Can you show a screenshot of ifconfig?

PussAzuki, 2022-05-07:
[image]
Is this enough? I also tried this 
[image]
but looks nothing changed

abajk, 2022-05-07:
> [image] Is this enough? I also tried this [image] but looks nothing changed

Looks good. Eth0 and eth1 should be sufficient in flowtable.

My previous tests were done in an IP over Ethernet environment . I did not test PPPoE.

abajk, 2022-05-07:
I would also experiment with packet steering.

PussAzuki, 2022-05-07:
> I would also experiment with packet steering.

I've tried packet steering, but it shouldn't work on the ipq40xx

PussAzuki, 2022-05-14:
I think ft is working, but I'm not sure why only a quarter of the performance.

PussAzuki, 2022-05-15:
> Try to use irqbalance (or manually control with /proc/irq/xx/smp_affinity, see what's happening at watch cat /proc/interrupts and htop. May be useful to watch from busybox or procps-ng-watch and also install htop.) Seems to work for me, but only tested up to 200 Mbit/s. There could be other bottlenecks, but the cpu is only lightly loaded and nicely spread. (I know irqbalance may not work on IPQ40xx) (need to do additional checks myself)

OEM firmware does run the full limit of the net, so I think there should be a program to improve performance. Then when I restored the configuration from an existing backup after returning from the DSA driver version firmware to the essedma+swconfig version driver version firmware, I found that the performance was limited to 680Mbps, which is exactly four times the 170Mbps, so I think flowtables does work, but I'm not sure why it behaves like it thinks the machine only has a single CPU core...

The configuration is /etc/config/firewall with software NAT enabled, and kmod-natflow is enabled manually in /etc/rc.local. So I do think it's still a problem with nftables/fw4.

PussAzuki, 2022-05-21:
`ethtool -K eth0 gro off`
`ethtool -K eth1 gro off`
Then it seems to be working "properly", but the speedtest.net score is only 550Mbps (510Mbps on the client side).

jmv2009, 2022-05-21:
Interesting. Also see https://access.redhat.com/solutions/20278, where it seems to be a firmware issues of the NIC. https://github.com/ntop/ntopng/issues/2618. Not sure how these issues pertain to the ipq40xx and the AR8031/AR8033/AR8035 PHY.

PussAzuki, 2022-05-21:
Maybe I should keep this issue open and change the title?

PussAzuki, 2022-05-22:
Reopen, perhaps we should debug whether the hardware does not support gro, or the support is incomplete, or nftables is not well compatible

PussAzuki, 2022-05-22:
I did find that in qsdk they use a script to do irq balancing, which by the way also turns off gro, but there is no speed drop when using iptables/fw3 with gro on.

I'm not sure what the problem is, but I'm willing to help get the feedback needed for debugging.

jmv2009, 2022-05-22:
With respect to irqbalancing:
With echo 8 > /proc/irq/90 etc, I get the following from cat /proc/interrupts
 84:  161681876          0          0          0     GIC-0 272 Edge      c080000.ethernet:rxq0
 86:       3235    6287914          0          0     GIC-0 274 Edge      c080000.ethernet:rxq2
 88:       3720          0    6680824          0     GIC-0 276 Edge      c080000.ethernet:rxq4
 90:       3220          0          0    4778861     GIC-0 278 Edge      c080000.ethernet:rxq6
So half of the irqs are still going to cpu0. This may half the possible maximum throughput if the cpu is limiting. Any idea why this happens?

PussAzuki, 2022-05-22:
> With respect to irq balancing: With echo 8 > /proc/irq/90/smp_affinity etc, I get the following from cat /proc/interrupts 84: 161681876 0 0 0 GIC-0 272 Edge c080000.ethernet:rxq0 86: 3235 6287914 0 0 GIC-0 274 Edge c080000.ethernet:rxq2 88: 3720 0 6680824 0 GIC-0 276 Edge c080000.ethernet:rxq4 90: 3220 0 0 4778861 GIC-0 278 Edge c080000.ethernet:rxq6 So half of the irqs are still going to cpu0. This may half the possible maximum throughput if the cpu is limiting, as mainly rxq0 is being triggered. Any idea why this happens?

At present, the ipq40xx dsa driver can only be considered to work, have to wait for the V3 version of the driver or even a higher version. If there are performance issues I hope to bring them up together as soon as possible.

PussAzuki, 2022-05-22:
Although turning off gro on the essedma+swconfig combo got the nftables offload to work, I still found that it seemed to occasionally not be accelerated and then only had a speedtest.net score of about 450Mbps.

PussAzuki, 2022-05-23:
Added: I noticed that when using natflow to speed up forwarding, sometimes the speed is also low, then I tried several times to observe the irq in top and found that the irq only occupies about 23% when the speed is normal, but when the speed is abnormally low, it occupies about 43% of the irq.

cwbsw, 2022-06-14:
It  seems that kernel 5.15 breaks smp packet steering, can you confirm that?
https://forum.openwrt.org/t/kernel-5-15-breaks-smp-packet-steering/129825

jmv2009, 2022-06-16:
Running 5.15 with packet steering ok on https://github.com/sartura/openwrt/tree/ipq40xx-dsa with single ethernet port.

PussAzuki, 2023-03-27:
Is there a follow up to ipqess? I see no updated patches since last year

abajk, 2024-10-12:
Maybe it was a pppoe bug. It is potentially fixed in https://github.com/torvalds/linux/commit/6db5dc7b351b9569940cd1cf445e237c42cd6d27.
Can someone compare pppoe speeds on OpenWRT 21.02.x, 22.03.x and 23.03.7?