openwrt/openwrt #9849: mtd write /tmp/factory.bin mtd2 Could not open mtd device: mtd2

hello-liang-tao, 2022-05-07:
In the openwrt17 system, mtd mtd2 can be used, but in openwrt21.04 it will report an error 

mtd write /tmp/factory.bin mtd2
Could not open mtd device: mtd2

Ntemis, 2022-05-07:
You should close this down my man.
the correct command is
"mtd -r write  /tmp/factory.bin factory"

closed 2022-05-13 by pepe2k

shripadrayewar, 2023-03-20:
im also facing same issue, both commands doesnt work

john-clark, 2023-12-29:
```
root@OpenWrt:/sbin# mtd -r write /tmp/openwrt-23.05.1-mediatek-mt7622-linksys_e8450-squashfs-sysupgrade.bin factory
Could not open mtd device: factory
Can't open device for writing!
```

or firmware. Documentation [here](https://openwrt.org/docs/guide-user/installation/generic.sysupgrade) not working.