openwrt/openwrt #9853: Cannot configure switch/VLAN through luci on Asus MAP-AC2200 on master 580723e86

tboege, 2022-05-08:
I have compiled openwrt with luci from master (580723e86) for Asus MAP-AC2200.

I would like to setup VLAN - but the menu entry "Switch" is missing under "Network"
It seems, that it is possible to configure the switch with swconfig:

```
root@OpenWrt:~# swconfig list
Found: switch0 - 90000.mdio-1
```

```
root@OpenWrt:~# swconfig dev switch0 show
Global attributes:
        enable_vlan: 1
        enable_mirror_rx: 0
        enable_mirror_tx: 0
        mirror_monitor_port: 0
        mirror_source_port: 0
        linkdown: ???
Port 0:
        mib: Port 0 MIB counters
RxBroad     : 566
RxPause     : 0
RxMulti     : 41
RxFcsErr    : 0
RxAlignErr  : 0
RxRunt      : 0
RxFragment  : 0
Rx64Byte    : 649
Rx128Byte   : 4245
Rx256Byte   : 1022
Rx512Byte   : 674
Rx1024Byte  : 93
Rx1518Byte  : 26
RxMaxByte   : 358
RxTooLong   : 0
RxGoodByte  : 1527750
RxBadByte   : 0
RxOverFlow  : 0
Filtered    : 598
TxBroad     : 154
TxPause     : 0
TxMulti     : 397
TxUnderRun  : 0
Tx64Byte    : 0
Tx128Byte   : 5952
Tx256Byte   : 2810
Tx512Byte   : 65
Tx1024Byte  : 258
Tx1518Byte  : 7400
TxMaxByte   : 80
TxOverSize  : 0
TxByte      : 11939032
TxCollision : 0
TxAbortCol  : 0
TxMultiCol  : 0
TxSingleCol : 0
TxExcDefer  : 0
TxDefer     : 0
TxLateCol   : 0

        pvid: 0
        link: port:0 link:up speed:1000baseT full-duplex txflow rxflow
Port 1:
        mib: Port 1 MIB counters
RxBroad     : 0
RxPause     : 0
RxMulti     : 0
RxFcsErr    : 0
RxAlignErr  : 0
RxRunt      : 0
RxFragment  : 0
Rx64Byte    : 0
Rx128Byte   : 0
Rx256Byte   : 0
Rx512Byte   : 0
Rx1024Byte  : 0
Rx1518Byte  : 0
RxMaxByte   : 0
RxTooLong   : 0
RxGoodByte  : 0
RxBadByte   : 0
RxOverFlow  : 0
Filtered    : 0
TxBroad     : 0
TxPause     : 0
TxMulti     : 0
TxUnderRun  : 0
Tx64Byte    : 0
Tx128Byte   : 0
Tx256Byte   : 0
Tx512Byte   : 0
Tx1024Byte  : 0
Tx1518Byte  : 0
TxMaxByte   : 0
TxOverSize  : 0
TxByte      : 0
TxCollision : 0
TxAbortCol  : 0
TxMultiCol  : 0
TxSingleCol : 0
TxExcDefer  : 0
TxDefer     : 0
TxLateCol   : 0

        pvid: 1
        link: port:1 link:down
Port 2:
        mib: Port 2 MIB counters
RxBroad     : 0
RxPause     : 0
RxMulti     : 0
RxFcsErr    : 0
RxAlignErr  : 0
RxRunt      : 0
RxFragment  : 0
Rx64Byte    : 0
Rx128Byte   : 0
Rx256Byte   : 0
Rx512Byte   : 0
Rx1024Byte  : 0
Rx1518Byte  : 0
RxMaxByte   : 0
RxTooLong   : 0
RxGoodByte  : 0
RxBadByte   : 0
RxOverFlow  : 0
Filtered    : 0
TxBroad     : 0
TxPause     : 0
TxMulti     : 0
TxUnderRun  : 0
Tx64Byte    : 0
Tx128Byte   : 0
Tx256Byte   : 0
Tx512Byte   : 0
Tx1024Byte  : 0
Tx1518Byte  : 0
TxMaxByte   : 0
TxOverSize  : 0
TxByte      : 0
TxCollision : 0
TxAbortCol  : 0
TxMultiCol  : 0
TxSingleCol : 0
TxExcDefer  : 0
TxDefer     : 0
TxLateCol   : 0

        pvid: 1
        link: port:2 link:down
Port 3:
        mib: Port 3 MIB counters
RxBroad     : 0
RxPause     : 0
RxMulti     : 0
RxFcsErr    : 0
RxAlignErr  : 0
RxRunt      : 0
RxFragment  : 0
Rx64Byte    : 0
Rx128Byte   : 0
Rx256Byte   : 0
Rx512Byte   : 0
Rx1024Byte  : 0
Rx1518Byte  : 0
RxMaxByte   : 0
RxTooLong   : 0
RxGoodByte  : 0
RxBadByte   : 0
RxOverFlow  : 0
Filtered    : 0
TxBroad     : 0
TxPause     : 0
TxMulti     : 0
TxUnderRun  : 0
Tx64Byte    : 0
Tx128Byte   : 0
Tx256Byte   : 0
Tx512Byte   : 0
Tx1024Byte  : 0
Tx1518Byte  : 0
TxMaxByte   : 0
TxOverSize  : 0
TxByte      : 0
TxCollision : 0
TxAbortCol  : 0
TxMultiCol  : 0
TxSingleCol : 0
TxExcDefer  : 0
TxDefer     : 0
TxLateCol   : 0

        pvid: 1
        link: port:3 link:down
Port 4:
        mib: Port 4 MIB counters
RxBroad     : 0
RxPause     : 0
RxMulti     : 0
RxFcsErr    : 0
RxAlignErr  : 0
RxRunt      : 0
RxFragment  : 0
Rx64Byte    : 0
Rx128Byte   : 0
Rx256Byte   : 0
Rx512Byte   : 0
Rx1024Byte  : 0
Rx1518Byte  : 0
RxMaxByte   : 0
RxTooLong   : 0
RxGoodByte  : 0
RxBadByte   : 0
RxOverFlow  : 0
Filtered    : 0
TxBroad     : 0
TxPause     : 0
TxMulti     : 0
TxUnderRun  : 0
Tx64Byte    : 0
Tx128Byte   : 0
Tx256Byte   : 0
Tx512Byte   : 0
Tx1024Byte  : 0
Tx1518Byte  : 0
TxMaxByte   : 0
TxOverSize  : 0
TxByte      : 0
TxCollision : 0
TxAbortCol  : 0
TxMultiCol  : 0
TxSingleCol : 0
TxExcDefer  : 0
TxDefer     : 0
TxLateCol   : 0

        pvid: 1
        link: port:4 link:down
Port 5:
        mib: Port 5 MIB counters
RxBroad     : 154
RxPause     : 0
RxMulti     : 397
RxFcsErr    : 0
RxAlignErr  : 0
RxRunt      : 0
RxFragment  : 0
Rx64Byte    : 1951
Rx128Byte   : 4197
Rx256Byte   : 2624
Rx512Byte   : 65
Rx1024Byte  : 258
Rx1518Byte  : 7480
RxMaxByte   : 0
RxTooLong   : 0
RxGoodByte  : 11873412
RxBadByte   : 0
RxOverFlow  : 0
Filtered    : 0
TxBroad     : 0
TxPause     : 0
TxMulti     : 9
TxUnderRun  : 0
Tx64Byte    : 685
Tx128Byte   : 4305
Tx256Byte   : 908
Tx512Byte   : 113
Tx1024Byte  : 94
Tx1518Byte  : 383
TxMaxByte   : 0
TxOverSize  : 0
TxByte      : 1307239
TxCollision : 0
TxAbortCol  : 0
TxMultiCol  : 0
TxSingleCol : 0
TxExcDefer  : 0
TxDefer     : 0
TxLateCol   : 0

        pvid: 2
        link: port:5 link:up speed:1000baseT full-duplex auto
VLAN 1:
        vid: 1
        ports: 0t 1 2 3 4
VLAN 2:
        vid: 2
        ports: 0t 5
```

```
root@OpenWrt:~# cat /etc/board.json
{
        "model": {
                "id": "asus,map-ac2200",
                "name": "ASUS Lyra MAP-AC2200"
        },
        "network": {
                "lan": {
                        "device": "eth1",
                        "protocol": "static"
                },
                "wan": {
                        "device": "eth0",
                        "protocol": "dhcp"
                }
        }
}
```

Do we miss something in board.json or should we look another place for the problem?

tboege, 2022-05-08:
I add to /etc/config/network

```
root@OpenWrt:~# tail  -n 15 /etc/config/network
config switch
        option name 'switch0'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch0'
        option vlan '1'
        option ports '0t 1 2 3 4'

config switch_vlan
        option device 'switch0'
        option vlan '2'
        option ports '0t 5'
```

After reboot the menuitem "Switch" appears, but with a warning - and I think the wrong layout display( I would expect  cpu to have tagget both vlan1 and vlan2 - so basicly the setting for Port 1)

pkgadd, 2022-05-09:
Btw., as a two-port device, you should only see `eth0`/ `eth1` in the network configuration.

tboege, 2022-05-09:
Thank you. I will wait for the inclusion of DSA

ondoteam, 2022-05-20:
Hello! 

https://github.com/sartura/openwrt/pull/27