openwrt/openwrt #9854: firewall4: software offloading not working with bridge devices in flowtables

alanswanson, 2022-05-08:
NFT flowtables need physical devices not bridges, at least not until Linux 5.13 when it'll automatically discover the topology behind bridge (and also VLAN and pppoe) devices.

However with https://git.openwrt.org/?p=project/firewall4.git;a=commitdiff;h=52831a00bec8e3676e2747029c0349f6be9c2593  resolve_lower_devices() checks for hw-tc-offload in devstatus and prevents any recursive resolving of devices for software offloading.The hw-tc-offload check needs removed from resolve_lower_devices() in root/usr/share/firewall4/main.uc.

Issue discovered by [abajk](https://github.com/abajk) in https://github.com/openwrt/openwrt/pull/9733#issuecomment-1116483329.

abajk, 2022-05-08:
@alanswanson 
If you have an idea how to fix it, can you prepare a patch and send it to the mailing list?

alanswanson, 2022-05-08:
I have a patch that fixes software offloading by removing the hw-tc-offload check from resolve_lower_devices() in root/usr/share/ucode/fw4.uc (where it was later moved) but reviewing the code I'm not sure about hardware offloading. Not familiar enough with ucode to confirm what the intention in resolve_lower_devices() here;

`if (!devstatus || devstatus[devname]?.["hw-tc-offload"])`

Should instead, as an alternative fix to removing it, be changed to;

`if (devstatus && devstatus[devname]?.["hw-tc-offload"])`

Need the dev to confirm as I don't have a hardware offloading device to test.

jow-, 2022-05-09:
Fix pushed to https://git.openwrt.org/57984e03 - please confirm that it solves your issue.

Vladdrako, 2022-05-09:
@jow- It causes regression on mt7621
<pre>
root@MI-R3P:~# service firewall restart
Section @rule[11] (Allow-ICMPv6-Forward) is disabled, ignoring section
Section @rule[16] (Support-UDP-Traceroute) is disabled, ignoring section
Section @redirect[2] is disabled, ignoring section
Section @redirect[7] (NTP) is disabled, ignoring section
Hardware flow offloading unavailable, falling back to software offloading
</pre>

jow-, 2022-05-09:
@Vladdrako - please show the list of devices included in the flowtable.

Vladdrako, 2022-05-09:
@jow- 
```
flowtable ft {
		hook ingress priority filter
		devices = { lan1, lan2, lan3, wan, wlan0, wlan0-1, wlan1 }
	}
```

jow-, 2022-05-09:
Hmm, I see. We likely need to filter the wifi ifaces then.

alanswanson, 2022-05-09:
So I update OpenWrt to git master as latest ucode needed after the fw4 modern syntax commit and it seemed to work with flowtable populated on HH5 with;

`devices = { lan1, lan2, lan3, lan4, wan, wlan1 }`

Though was missing wlan0 (ath10k) for some reason. However couldn't investigate further as had no internet connectivity (after DSL was up) for clients and running fw4 was hanging on following process;

`utpl -S /usr/share/firewall4/main.uc`

No errors to show why, and manually nft flushing didn't resolve internet connectivity so needed to quickly revert to old OpenWrt image. Forgot to check nft ruleset output before and after to see if different, so not sure if fw4 or another cause. Will need to investigate another time.

Had also configured interface wan6 with device @wan in the perennial hope my ISPs upstream provider ever provides IPv6 sometime and also had over 60 (sixty) of the following running/hanging too;

```
{dhcpv6.script} /bin/sh /lib/netifd/dhcpv6.script pppoe-wan bound
{fw3} /bin/sh /sbin/fw3 -q network wan6
flock -x /var/run/fw4.lock utpl -S /usr/share/firewall4/main.uc
```

abajk, 2022-05-09:
> Though was missing wlan0 (ath10k) for some reason. However couldn't investigate further as had no internet connectivity (after DSL was up) for clients and running fw4 was hanging on following process;

Maybe it's because flowtable in kernel 5.10 can't encapsulate vlan and pppoe packets. This trick only works on the br-lan side where there is no ppp. Try manually replacing wan with pppoe-wan again or kernel 5.15.

jow-, 2022-05-09:
Another change pushed in https://git.openwrt.org/b2682251

alanswanson, 2022-05-09:
@abajk No, I'd tried clearing the flowtable (and nftables) while it should also return from flowtable back to ingress (such as it does for br-lan before Linux 5.13). Lantiq with Linux 5.15 as experimental is still a WIP issue.

Will retest tomorrow with latest change on a spare router and see if the utpl hanging occurs there too.

alanswanson, 2022-05-10:
Retested successfully with [b2682251](https://git.openwrt.org/b2682251) and the flowtable for software offloading was correctly setup on a HH5;

`devices = { dsl0, lan1, lan2, lan3, lan4, wan, wlan0, wlan1 }`

Had removed the (wishful) wan6 interface which may be why I didn't encounter whatever caused the hanging utpl problem previously. Will retest that separately in the future.

@Vladdrako Can you test your hardware offloading flowtable setup and close this issue if it's ok? Cheers.

Vladdrako, 2022-05-11:
> Another change pushed in [git.openwrt.org/b2682251](https://git.openwrt.org/b2682251)

> Can you test your hardware offloading flowtable setup and close this issue if it's ok? Cheers.

Now it works fine, thanks.
```
flowtable ft {
		hook ingress priority filter
		devices = { lan1, lan2, lan3, wan }
		flags offload
	}
```

pepe2k, 2022-05-13:
Closing this one, based on recent comments it seems the problem was already fixed.

Cheers,
Piotr

closed 2022-05-13 by pepe2k

abajk, 2022-05-17:
@pepe2k At least firewall4 must be bumped to take advantage of these changes.

Unfortunately, I haven't had time to test the performance impact of these changes yet.