openwrt/openwrt #9855: WPA3 not working at all on Netgear R8000 with 22.03.0-rc

acidicX, 2022-05-08:
Hi,

I can't get WPA 3 to work at all with the 22.03.0-rc release on my R8000.

After changing the settings to either SAE or mixed mode, I can't connect any device (iPhones, Android Phones, Linux PCs ...).
802.11r & 802.11w are both disabled.

Log just says:
```
Sun May  8 17:46:59 2022 daemon.info hostapd: wlan0: STA MY:MAC:ADDRESS:DEVICE IEEE 802.11: disassociated
Sun May  8 17:46:59 2022 daemon.info hostapd: wlan0: STA MY:MAC:ADDRESS:DEVICE IEEE 802.11: disassociated
```

Device says the password is wrong.
This might be related to https://github.com/openwrt/openwrt/issues/7858 but I can't get it to work in neither mixed nor SAE modes.

As far as I can remember, this also did not work on 21.02 (but there were several other issues as well, so I waited for 22.03).

Is there a way to debug this further? Get the log output of other components as well?

pkgadd, 2022-05-09:
It's quite unlikely to be related to #7858 (as WPA3/ SAE does work on the r7800/ qca9984 just fine).

brcmfmac is a fullmac driver, meaning the kernel driver is mostly just a message passing agent to the firmware blob running on the wireless chipset. This rather heavy fullmac firmware blob needs to support WPA3/ SAE and its preconditions (predominantly 802.11r), which is not a given - if it doesn't, there's nothing that could be done on the OpenWrt or kernel side, you'd have to ask Broadcom for an updated firmware blob.

acidicX, 2022-05-09:
@pkgadd I see, so it would be possible that it works if I upgrade to the latest stock Netgear firmware (https://kb.netgear.com/000064738/R8000-Firmware-Version-1-0-4-84) **before** flashing OpenWRT 22.03? Because that would also update the firmware blob?

pkgadd, 2022-05-09:
No.
The brcmfmac firmware blob is part of the rootfs (the OpenWrt firmware image), taken from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/. Flashing the OEM firmware won't do anything, not that I'd expect that one to support WPA3/ SAE either (vendors tend to treat it as a 802.11ax-only feature, although there is no technical reason for that), apart from it using a different driver altogether.

pepe2k, 2022-05-13:
@acidicX as already explained by @pkgadd, we are not able to solve this problem on our side, this would need to be handled by chipset vendor, in the proprietary firmware which runs on the WiFi radio.

Cheers,
Piotr

closed 2022-05-13 by pepe2k

acidicX, 2022-05-13:
@pepe2k thx for the information, is it possible to document this in the wiki

https://openwrt.org/toh/netgear/r8000

(or add my GH account to the wiki users, so I can do it)?

pepe2k, 2022-05-13:
@acidicX have a look here: https://forum.openwrt.org/t/applying-for-openwrt-wiki-account/101671