openwrt/openwrt #9856: dnsmasq: dhcp-script option not working

potchin, 2022-05-08:
Model | Raspberry Pi 4 Model B Rev 1.4
Architecture | ARMv8 Processor rev 3
Target Platform | bcm27xx/bcm2711
Firmware Version | OpenWrt 22.03.0-rc1 r19302-df622768da / LuCI openwrt-22.03 branch git-22.083.69105-af8e91c
Kernel Version | 5.10.111
dnsmasq - 2.86-13

Adding the `dhcp-script` option to `/etc/dnsmasq.conf` always results in an error rather than the script being called. For example, setting..

```
dhcp-script = /bin/echo
```
will result in...

```
Sun May  8 19:47:23 2022 daemon.err dnsmasq[1]: failed to execute /bin/echo: No such file or directory
```

Just for sanity...
```
root@gateway:/etc# ls -l /bin/echo
lrwxrwxrwx    1 root     root             7 Apr 20 11:37 /bin/echo -> busybox
root@gateway:/etc# which echo
/bin/echo
root@gateway:/etc# echo test
test
```

Link to the openwrt [documentation](https://openwrt.org/docs/guide-user/base-system/dhcp.dnsmasq#etcdnsmasqconf) which mentions this feature.

This is a regression from 21.02, I have been using this option previously

dave14305, 2022-05-08:
It’s a `ujail` issue perhaps. dnsmasq can’t access the file if it’s outside of its permitted ujail mounts.

jeffqumata, 2022-05-09:
I found that there's a directory already available (`/etc/hotplug.d/dhcp`) which I can just drop scripts in so I have changed my setup to use that now, thanks for the suggestion.

Its probably worth updating the documentation nonetheless, especially as its a regression from 21.02

potchin, 2022-05-09:
I found that there's a directory already available (/etc/hotplug.d/dhcp) which I can just drop scripts in so I have changed my setup to use that now, thanks for the suggestion.

Its probably worth updating the documentation nonetheless, especially as its a regression from 21.02

closed 2022-05-12 by jow-