openwrt/openwrt #9860: Openwrt 22.03 kernel module kmod-mac80211 build fails

shaheenmohamed, 2022-05-09:
I am trying to build a firmware for my x86 device with kmod-mt7915e kernel module. But with the latest openwrt 22.03 version the package package/kernel/mac80211 (kmod-mt7915e) fails to build.
`make[2]: Entering directory '/home/shaheen/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/shaheen/openwrt/scripts/config'
time: target/linux/prereq#0.09#0.01#0.10
 make[1] world
 make[2] tools/compile
 make[2] package/cleanup
 make[3] -C tools/flock compile
 make[3] -C tools/xz compile
 make[3] -C tools/sed compile
 make[3] -C tools/tar compile
 make[3] -C tools/patch compile
 make[3] -C tools/m4 compile
 make[3] -C tools/autoconf-archive compile
 make[3] -C tools/ninja compile
 make[3] -C tools/cpio compile
 make[3] -C tools/lzma compile
 make[3] -C tools/mtools compile
 make[3] -C tools/patch-image compile
 make[3] -C tools/sstrip compile
 make[3] -C tools/xxd compile
 make[3] -C tools/zip compile
 make[3] -C tools/autoconf compile
 make[3] -C tools/missing-macros compile
 make[3] -C tools/meson compile
 make[3] -C tools/pkgconf compile
 make[3] -C tools/zstd compile
 make[3] -C tools/automake compile
 make[3] -C tools/libressl compile
 make[3] -C tools/dosfstools compile
 make[3] -C tools/libtool compile
 make[3] -C tools/cmake compile
 make[3] -C tools/flex compile
 make[3] -C tools/e2fsprogs compile
 make[3] -C tools/fakeroot compile
 make[3] -C tools/gengetopt compile
 make[3] -C tools/mklibs compile
 make[3] -C tools/gmp compile
 make[3] -C tools/patchelf compile
 make[3] -C tools/bison compile
 make[3] -C tools/mpfr compile
 make[3] -C tools/mpc compile
 make[3] -C tools/bc compile
 make[3] -C tools/findutils compile
 make[3] -C tools/mkimage compile
 make[3] -C tools/padjffs2 compile
 make[3] -C tools/quilt compile
 make[3] -C tools/zlib compile
 make[3] -C tools/expat compile
 make[3] -C tools/make-ext4fs compile
 make[3] -C tools/mtd-utils compile
 make[3] -C tools/firmware-utils compile
 make[3] -C tools/squashfskit4 compile
 make[2] toolchain/compile
 make[3] -C toolchain/gdb compile
 make[3] -C toolchain/binutils compile
 make[3] -C toolchain/fortify-headers compile
 make[3] -C toolchain/nasm compile
 make[3] -C toolchain/gcc/initial compile
 make[3] -C toolchain/kernel-headers compile
 make[3] -C toolchain/musl compile
 make[3] -C toolchain/gcc/final compile
 make[2] target/compile
 make[3] -C target/linux compile
 make[2] buildinfo
 make[2] package/compile
 make[3] diffconfig buildversion feedsversion
 make[3] -C package/libs/toolchain compile
 make[3] -C package/libs/libjson-c host-compile
[... 132 lines cut ...]
 make[3] -C feeds/luci/libs/rpcd-mod-luci compile
 make[3] -C feeds/luci/applications/luci-app-wireguard compile
 make[3] -C feeds/luci/protocols/luci-proto-3g compile
 make[3] -C feeds/packages/net/mosquitto compile
 make[3] -C package/network/utils/iproute2 compile
 make[3] -C feeds/luci/modules/luci-base compile
 make[3] -C feeds/packages/net/mosquitto compile
 make[3] -C feeds/packages/net/mosquitto compile
 make[3] -C /ns_source/netsurion/ns_mgr compile
 make[3] -C package/boot/grub2 compile
 make[3] -C /ns_source/netsurion/ns_stats compile
 make[3] -C feeds/luci/modules/luci-compat compile
 make[3] -C feeds/luci/modules/luci-mod-network compile
 make[3] -C feeds/luci/modules/luci-mod-status compile
 make[3] -C feeds/luci/modules/luci-mod-system compile
 make[3] -C feeds/luci/applications/luci-app-uhttpd compile
 make[3] -C feeds/luci/modules/luci-mod-admin-full compile
 make[3] -C feeds/packages/net/mwan3 compile
 make[3] -C feeds/luci/applications/luci-app-mwan3 compile
 make[3] -C package/kernel/mac80211 compile
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
 make[3] -C package/boot/grub2 compile`
Attaching the diffconfig file...
[diffconfig.txt](https://github.com/openwrt/openwrt/files/8650736/diffconfig.txt)
This issue can be reproduced with the latest version of openwrt (22.03)

ynezz, 2022-05-31:
BTW our buildbots don't seem to have this issue as there is [kmod-mt7615e_5.10.116+2022-03-15-053668ac-4_x86_64.ipk](https://downloads.openwrt.org/releases/22.03.0-rc3/targets/x86/64/kmods/5.10.116-1-6703dcafaa268bc75821eea0f02b92ca/kmod-mt7615e_5.10.116+2022-03-15-053668ac-4_x86_64.ipk) available in 22.03-rc3 so it's probably something related to your local changes/config. You can try the [config](https://downloads.openwrt.org/releases/22.03.0-rc3/targets/x86/64/config.buildinfo) used by our buildbots and reiterate from there and find out which config option breaks that kernel module for you.

dangowrt, 2022-06-06:
I've just hit this bug with a clean Debian 11 VM trying to build OpenWrt. The same build tree and configuration works well on my usual build host, just on the Debian build host I see this breakage. So this is **not** config related.

dangowrt, 2022-06-06:
[image]

richard-scott, 2022-09-14:
I'm trying to build on an Ubuntu 20.04.5 LTS host and it fails.
I've even tried the official Docker images, and that also fails.

Here is my error:
```
export CROSS="aarch64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="aarch64-openwrt-linux-musl-nm" STRIP="/home/build/openwrt/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/build/openwrt/scripts/strip-kmod.sh" PATCHELF="/home/build/openwrt/staging_dir/host/bin/patchelf" /home/build/openwrt/scripts/rstrip.sh /home/build/openwrt/build_dir/target-aarch64_generic_musl/linux-layerscape_armv8_64b/backports-5.15.58-1/ipkg-aarch64_generic/kmod-rtlwifi-btcoexist
rstrip.sh: /home/build/openwrt/build_dir/target-aarch64_generic_musl/linux-layerscape_armv8_64b/backports-5.15.58-1/ipkg-aarch64_generic/kmod-rtlwifi-btcoexist/lib/modules/5.10.139/btcoexist.ko: relocatable
(cd /home/build/openwrt/build_dir/target-aarch64_generic_musl/linux-layerscape_armv8_64b/backports-5.15.58-1/ipkg-aarch64_generic/kmod-rtlwifi-btcoexist/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/build/openwrt/bin/targets/layerscape/armv8_64b/packages
/home/build/openwrt/staging_dir/host/bin/fakeroot /home/build/openwrt/staging_dir/host/bin/bash /home/build/openwrt/scripts/ipkg-build -m "" /home/build/openwrt/build_dir/target-aarch64_generic_musl/linux-layerscape_armv8_64b/backports-5.15.58-1/ipkg-aarch64_generic/kmod-rtlwifi-btcoexist /home/build/openwrt/bin/targets/layerscape/armv8_64b/packages
libfakeroot: read: Connection reset by peer
/home/build/openwrt/staging_dir/host/bin/fakeroot: line 1: kill: (601206) - No such process
make[2]: *** [Makefile:567: /home/build/openwrt/bin/targets/layerscape/armv8_64b/packages/kmod-rtlwifi-btcoexist_5.10.139+5.15.58-1-2_aarch64_generic.ipk] Error 1
make[2]: Leaving directory '/home/build/openwrt/feeds/base/package/kernel/mac80211'
time: package/feeds/base/mac80211/regular/compile#6.09#3.60#39.14
    ERROR: package/feeds/base/mac80211 failed to build (build variant: regular).
make[1]: *** [package/Makefile:116: package/feeds/base/mac80211/compile] Error 1
make[1]: Leaving directory '/home/build/openwrt'
make: *** [/home/build/openwrt/include/toplevel.mk:224: package/feeds/base/mac80211/compile] Error 2
```

Ricrdsson1, 2023-01-11:
I also have this problem but only for X86 for another platform it works fine.  use the officialni .config without changes and the compilation will fail.  How to solve it?  I don't understand that the build bot builds it correctly.

Neustradamus, 2024-11-04:
Have you solved this problem?

jjm2473, 2024-12-25:
I recently ran into this bug as well, and it was a very weird one. It's not relate to mac80211 or config, but related to system (Ubuntu 22.04 / 24.04) and elfutils.

Here is a Github Actions workflow to reproduce this bug: https://github.com/jjm2473/test-build/tree/openwrt-22.03-fail-oot-kmods .

Logs:
https://github.com/jjm2473/test-build/issues/1

If you build elfutils first and then build OOT Kmods on Ubuntu 22.04 / 24.04, the bug will be triggered.

After patch elfutils's libelf.pc https://github.com/istoreos/istoreos/commit/82095637a11db1cf7633a7d6727d7db9c9b64a20 , the bug was gone.

Again, it's very weird.