openwrt/openwrt #9862: [build/ath79] Unifi AC Mesh: Calculation for image size check does not include rootfs_data partition, insufficient flash space

dzzinstant, 2022-05-09:
### Bug Report
**What is the problem?**
I run OpenWrt on an Ubiquiti Unifi AC Mesh acting as a Freifunk node (gluon). Recently, an update (official image of the local Freifunk community) reset the device to first boot state, all configuration data were deleted. It was also impossible to configure the device, as all data were lost after reboot.
Details in original bug report: https://github.com/freifunk-gluon/gluon/issues/2473.

Apparently, the flash space allocated to the `rootfs_data` partition is too small, so it can't be mounted as an jffs2 file system. 
Excerpt from `dmesg` log:
```
[    0.333109] 7 fixed-partitions partitions found on MTD device spi0.0
[    0.339677] Creating 7 MTD partitions on "spi0.0":
[    0.344649] 0x000000000000-0x000000060000 : "u-boot"
[    0.350659] 0x000000060000-0x000000070000 : "u-boot-env"
[    0.357032] 0x000000070000-0x000000800000 : "firmware"
[    0.366427] 2 uimage-fw partitions found on MTD device firmware
[    0.372594] Creating 2 MTD partitions on "firmware":
[    0.377741] 0x000000000000-0x000000210000 : "kernel"
[    0.383744] 0x000000210000-0x000000790000 : "rootfs"
[    0.389700] mtd: device 4 (rootfs) set to be root filesystem
[    0.397324] 1 squashfs-split partitions found on MTD device rootfs
[    0.403773] 0x000000760000-0x000000790000 : "rootfs_data"
[    0.410191] 0x000000800000-0x000000f90000 : "kernel1"
[    0.416318] 0x000000f90000-0x000000fb0000 : "bs"
[    0.421969] 0x000000fb0000-0x000000ff0000 : "cfg"
[    0.427787] 0x000000ff0000-0x000001000000 : "art"
[..]
[   10.862669] Too few erase blocks (3)
[   10.867711] Too few erase blocks (3)
[   10.871614] mount_root: failed to mount -t jffs2 /dev/mtdblock5 /tmp/overlay: Invalid argument
[   11.414596] Too few erase blocks (3)
[   11.419254] mount_root: unable to set filesystem state
[   11.424834] mount_root: switching to jffs2 overlay
[   11.429947] mount_root: switching to jffs2 failed - fallback to ramoverlay
```

When creating a somewhat bigger image, the file size check kicks in (as it should):
```
 make[4] -C target/linux install
 make[7] -C target/linux/ath79/image/lzma-loader compile loader.elf
[..]
    WARNING: Image file /home/dustin/gluon-tmp/gluon/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/tmp/openwrt-ath79-generic-ubnt_unifiac-mesh-squashfs-sysupgrade.bin is too big: 8060932 > 7929856
```
However, an image with a maximum size of 7744kB (= 0x790000) would leave no space at all for a `rootfs_data` partition. 

**What is the expected behaviour?**
A reasonable amount of memory should be reserved for the `rootfs_data` partition, and included in the calculation of the maximum allowed image size.
If it is not possible to reserve enough memory, the build process should warn about this and not offer the image for release.
(A further image size check during sysupgrade might be useful as well.)

walterav1984, 2022-05-10:
Is this the [Unifi AC Mesh](https://openwrt.org/toh/ubiquiti/unifiac) hardware you are referring to with current [stable image 21.02.3](https://downloads.openwrt.org/releases/21.02.3/targets/ath79/generic/openwrt-21.02.3-ath79-generic-ubnt_unifiac-mesh-squashfs-sysupgrade.bin) maybe from the early 22.03.0rc1 or master snapshots having 16MB flash?

Could you please verify using the Original OpenWrt image albeit 22.03 or master build having this same issue or is the image size of the original OpenWrt not huge enough to trigger the event?

pepe2k, 2022-05-13:
@dzzinstant are you able to explain how does the dual-image feature work on these devices (I don't have much practical experience with Ubiquiti)? I'm looking at the partition layout and would like to know what's purpose of `kernel1` (I suppose some kind of a recovery/dual-image?), `bs` and `cfg`.

I might be wrong here (didn't make a proper research) but I think we only check size of the whole final firmware file which has to be smaller than a size of the target container (here: `firmware` mtd partition) and probably number of available erase blocks for creating `rootfs_data` isn't calculated and checked in advance.

blocktrron, 2022-09-08:
@pepe2k The first byte of the bs partition determines which OS is booted (0 for kernel0, 1 for kernel1)

dzzinstant, 2024-09-08:
Apparently, this problem did not reappear for the UniFi AC Mesh.
To solve the problem in general (for all devices), it would be possible to reserve a minimum space needed for erase blocks during the image building process, e.g. in the targets' individual Makefiles `openwrt/target/linux/<..>/image/<..>.mk` or by modifying the `check-size` command in `openwrt/include/image-commands.mk`.

Both options seem somewhat arbitrary and disproportionate to me (because the bug only appears very rarely and can easily be cured), and might also introduce new obscure bugs. So I am rather closing the corresponding bug reports for gluon and openwrt.

closed 2024-09-08 by dzzinstant