openwrt/openwrt #9864: R7800 - 5Ghz breaks after DFS radar detection

NilsRo, 2022-05-10:
After DFS radar detection my 5Ghz wifi does not come back online. In the logs it looks like this. I have to restart wifi to get it working again. Will work around this with a short script but it would be much better if wifi comes back by its own.

Device: Netgear R7800
Firmware: OpenWrt 21.02.2 r16495-bf0c965af0 
Wifi Driver: ath10k-firmware-qca9984-ct-full-htt
hostapd-common | 2020-06-08-5a8b3662-40

[image]
[image]

[openwrt.log](https://github.com/openwrt/openwrt/files/8658999/openwrt.log)

pepe2k, 2022-05-13:
@NilsRo could you please check if this problem occurs in `22.03-rc1` and `master`?

Black6spdZ, 2022-05-15:
why the full-htt driver?

NilsRo, 2022-05-15:
Without my Laptops with Intel WIFI have serious connection issues. full-htt fixed this and on my investigation seems to be the better wifi driver in general (but with more ressource consumption)

NilsRo, 2022-05-21:
> @NilsRo could you please check if this problem occurs in `22.03-rc1` and `master`?

I checked master with similar behaviour. A wifi down and wifi up fixed it which I write a script now...

`May 19 18:10:17 Arbeitszimmer hostapd: wlan0: DFS-RADAR-DETECTED freq=5500 ht_enabled=0 chan_offset=0 chan_width=5 cf1=5570 cf2=0
May 19 18:10:17 Arbeitszimmer hostapd: dfs_downgrade_bandwidth: no DFS channels left, waiting for NOP to finish
May 19 18:10:17 Arbeitszimmer hostapd: wlan0: AP-DISABLED
May 19 18:10:17 Arbeitszimmer hostapd: wlan0: STA f4:f5:e8:16:6b:e2 MLME: MLME-DEAUTHENTICATE.indication(f4:f5:e8:16:6b:e2, 1)
May 19 18:10:17 Arbeitszimmer hostapd: wlan0: AP-STA-DISCONNECTED f4:f5:e8:16:6b:e2
May 19 18:10:17 Arbeitszimmer hostapd: 20/40 MHz: center segment 0 (=114) and center freq 1 (=5510) not in sync
May 19 18:10:17 Arbeitszimmer hostapd: 20/40 MHz: center segment 0 (=114) and center freq 1 (=5510) not in sync
May 19 18:10:17 Arbeitszimmer hostapd: 20/40 MHz: center segment 0 (=114) and center freq 1 (=5510) not in sync
May 19 18:10:17 Arbeitszimmer kernel: [528130.657416] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 1
May 19 18:10:17 Arbeitszimmer kernel: [528130.657480] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 1
May 19 18:10:17 Arbeitszimmer kernel: [528130.663338] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 1
May 19 18:10:17 Arbeitszimmer kernel: [528130.670894] br-GUEST: port 3(wlan0-1) entered disabled state
May 19 18:10:17 Arbeitszimmer netifd: Network device 'wlan0-1' link is down
May 19 18:10:17 Arbeitszimmer kernel: [528130.703207] device wlan0-1 left promiscuous mode
May 19 18:10:17 Arbeitszimmer kernel: [528130.703238] br-GUEST: port 3(wlan0-1) entered disabled state
May 19 18:10:17 Arbeitszimmer hostapd: nl80211: Failed to remove interface wlan0-1 from bridge br-GUEST: No such device
May 19 18:10:17 Arbeitszimmer kernel: [528130.782658] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 2
May 19 18:10:17 Arbeitszimmer kernel: [528130.782727] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 2
May 19 18:10:17 Arbeitszimmer kernel: [528130.788431] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 2
May 19 18:10:17 Arbeitszimmer netifd: Network device 'wlan0-2' link is down
May 19 18:10:17 Arbeitszimmer kernel: [528130.796482] br-iot: port 3(wlan0-2) entered disabled state
May 19 18:10:17 Arbeitszimmer kernel: [528130.823277] device wlan0-2 left promiscuous mode
May 19 18:10:17 Arbeitszimmer kernel: [528130.823311] br-iot: port 3(wlan0-2) entered disabled state
May 19 18:10:17 Arbeitszimmer hostapd: nl80211: Failed to remove interface wlan0-2 from bridge br-iot: No such device
May 19 18:10:17 Arbeitszimmer hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
May 19 18:10:17 Arbeitszimmer kernel: [528130.938817] device wlan0 left promiscuous mode
May 19 18:10:17 Arbeitszimmer kernel: [528130.938933] br-lan: port 2(wlan0) entered disabled state
May 19 18:10:17 Arbeitszimmer kernel: [528131.059133] ath10k_pci 0000:01:00.0: could not get mac80211 beacon, vdev_id: 0 addr: 78:d2:94:a2:05:13
May 19 18:10:18 Arbeitszimmer kernel: [528131.094167] ath10k_pci 0000:01:00.0: mac flush null vif, drop 0 queues 0xffff
May 19 18:10:18 Arbeitszimmer kernel: [528131.095499] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 0
May 19 18:10:18 Arbeitszimmer kernel: [528131.100345] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 0
May 19 18:10:18 Arbeitszimmer kernel: [528131.107199] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 0
May 19 18:10:18 Arbeitszimmer netifd: Network device 'wlan0' link is down
May 19 18:10:18 Arbeitszimmer hostapd: wlan0: interface state ENABLED->DISABLED
May 19 18:10:24 Arbeitszimmer kernel: [528137.131100] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
May 19 18:10:24 Arbeitszimmer kernel: [528137.131130] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
May 19 18:10:24 Arbeitszimmer kernel: [528137.213495] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
May 19 18:10:24 Arbeitszimmer kernel: [528137.214588] ath10k_pci 0000:01:00.0: wmi print 'free: 84936 iram: 13428 sram: 11224'
May 19 18:10:24 Arbeitszimmer kernel: [528137.598606] ath10k_pci 0000:01:00.0: rts threshold -1
May 19 18:10:24 Arbeitszimmer kernel: [528137.603987] ath10k_pci 0000:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
May 19 18:10:24 Arbeitszimmer kernel: [528137.606059] br-lan: port 2(wlan0) entered blocking state
May 19 18:10:24 Arbeitszimmer kernel: [528137.614486] br-lan: port 2(wlan0) entered disabled state
May 19 18:10:24 Arbeitszimmer hostapd: wlan0: interface state DISABLED->COUNTRY_UPDATE
May 19 18:10:24 Arbeitszimmer hostapd: wlan0: interface state COUNTRY_UPDATE->HT_SCAN
May 19 18:10:24 Arbeitszimmer hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
May 19 18:10:24 Arbeitszimmer kernel: [528137.619893] device wlan0 entered promiscuous mode
May 19 18:10:24 Arbeitszimmer hostapd: wlan1: STA f4:f5:e8:16:6b:e2 IEEE 802.11: authentication OK (open system)
May 19 18:10:24 Arbeitszimmer hostapd: wlan1: STA f4:f5:e8:16:6b:e2 MLME: MLME-AUTHENTICATE.indication(f4:f5:e8:16:6b:e2, OPEN_SYSTEM)
May 19 18:10:24 Arbeitszimmer hostapd: wlan1: STA f4:f5:e8:16:6b:e2 MLME: MLME-DELETEKEYS.request(f4:f5:e8:16:6b:e2)
[... cut ...]
May 19 18:14:41 Arbeitszimmer hostapd: wlan1: STA f4:f5:e8:16:6b:e2 IEEE 802.11: authentication OK (open system)
May 19 18:14:41 Arbeitszimmer hostapd: wlan1: STA f4:f5:e8:16:6b:e2 MLME: MLME-AUTHENTICATE.indication(f4:f5:e8:16:6b:e2, OPEN_SYSTEM)
May 19 18:14:41 Arbeitszimmer hostapd: wlan1: STA f4:f5:e8:16:6b:e2 MLME: MLME-DELETEKEYS.request(f4:f5:e8:16:6b:e2)
May 19 18:14:41 Arbeitszimmer hostapd: wlan1: STA f4:f5:e8:16:6b:e2 IEEE 802.11: authenticated
May 19 18:14:41 Arbeitszimmer hostapd: wlan1: STA f4:f5:e8:16:6b:e2 IEEE 802.11: association OK (aid 1)
`

NilsRo, 2022-05-21:
[syslog.log](https://github.com/openwrt/openwrt/files/8747324/syslog.log)

SirDMZ, 2022-07-19:
Howdy, 

https://github.com/openwrt/openwrt/issues/10226 has the following suggestion. Perhaps these issues are related.
Edit your wireless config file and ensure you've got the following. Apparently without channel auto it won't even swap to a not blocked channel.

	option channel 'auto'
	option channels '136, 140, 132, 116, 112, 108, 104, 100' #replace with valid channels  for your country of course!

NilsRo, 2022-07-20:
Solved it (workaround) with this program running every 5 minutes....
https://github.com/NilsRo/Openwrt-radio-check

NilsRo, 2022-07-31:
> Howdy,
>
> #10226 has the following suggestion. Perhaps these issues are related. Edit your wireless config file and ensure you've got the following. Apparently without channel auto it won't even swap [...]

Worked much more stable like with the standard setting from LuCi. It seems that is a missing in the configuration to get it really working...it never hung after setting this but switching channels. But you must be also keep 160Mhz in mind so if you like to use it but only this channels in the list. Have to update my Wifi_Checker to work also with AUTO but actually it seems not very important as it worked fine for 11 days.

NilsRo, 2022-08-01:
Update: it does not work with 160Mhz. But with 80 Mhz it is a nice workaround that OpenWRT will switch between DFS channels. If I put 100 and 36 in the list it will stay on 36 and never switch back. With 100 only it will break...

Attached an updated log from the issue...

```
Mon Aug  1 20:56:10 2022 daemon.notice hostapd: wlan0: DFS-RADAR-DETECTED freq=5500 ht_enabled=0 chan_offset=0 chan_width=5 cf1=5570 cf2=0
Mon Aug  1 20:56:10 2022 daemon.notice hostapd: dfs_downgrade_bandwidth: no DFS channels left, waiting for NOP to finish
Mon Aug  1 20:56:10 2022 daemon.notice hostapd: wlan0: AP-DISABLED
Mon Aug  1 20:56:10 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Mon Aug  1 20:56:10 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0-1 wasn't started
Mon Aug  1 20:56:10 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0-2 wasn't started
Mon Aug  1 20:56:10 2022 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Mon Aug  1 20:56:10 2022 kern.info kernel: [686249.754122] device wlan0 left promiscuous mode
Mon Aug  1 20:56:10 2022 kern.info kernel: [686249.754212] br-lan: port 4(wlan0) entered disabled state
Mon Aug  1 20:56:11 2022 kern.warn kernel: [686249.795588] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 0
Mon Aug  1 20:56:11 2022 daemon.notice hostapd: wlan0: interface state HT_SCAN->DISABLED
Mon Aug  1 20:56:17 2022 kern.warn kernel: [686255.788580] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
Mon Aug  1 20:56:17 2022 kern.warn kernel: [686255.788606] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
Mon Aug  1 20:56:17 2022 kern.info kernel: [686255.870476] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
Mon Aug  1 20:56:17 2022 kern.info kernel: [686255.871316] ath10k_pci 0000:01:00.0: wmi print 'free: 84936 iram: 13428 sram: 11224'
Mon Aug  1 20:56:17 2022 kern.info kernel: [686256.254359] ath10k_pci 0000:01:00.0: rts threshold -1
Mon Aug  1 20:56:17 2022 kern.warn kernel: [686256.259784] ath10k_pci 0000:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
Mon Aug  1 20:56:17 2022 kern.info kernel: [686256.264099] br-lan: port 4(wlan0) entered blocking state
Mon Aug  1 20:56:17 2022 kern.info kernel: [686256.269907] br-lan: port 4(wlan0) entered disabled state
Mon Aug  1 20:56:17 2022 daemon.notice hostapd: wlan0: interface state DISABLED->COUNTRY_UPDATE
Mon Aug  1 20:56:17 2022 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Mon Aug  1 20:56:17 2022 daemon.err hostapd: VLAN: vlan_event_receive: recvfrom failed: No buffer space available
Mon Aug  1 20:56:17 2022 daemon.err hostapd: VLAN: vlan_event_receive: recvfrom failed: No buffer space available
Mon Aug  1 20:56:17 2022 daemon.err hostapd: VLAN: vlan_event_receive: recvfrom failed: No buffer space available
Mon Aug  1 20:56:17 2022 daemon.notice hostapd: netlink: recvfrom failed: No buffer space available
Mon Aug  1 20:56:17 2022 kern.info kernel: [686256.275725] device wlan0 entered promiscuous mode
Mon Aug  1 20:56:17 2022 daemon.notice hostapd: wlan0: DFS-RADAR-DETECTED freq=5500 ht_enabled=0 chan_offset=0 chan_width=5 cf1=5570 cf2=0
Mon Aug  1 20:56:17 2022 daemon.notice hostapd: dfs_downgrade_bandwidth: no DFS channels left, waiting for NOP to finish
Mon Aug  1 20:56:17 2022 daemon.notice hostapd: wlan0: AP-DISABLED
Mon Aug  1 20:56:17 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Mon Aug  1 20:56:17 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0-1 wasn't started
Mon Aug  1 20:56:17 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0-2 wasn't started
Mon Aug  1 20:56:17 2022 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Mon Aug  1 20:56:17 2022 kern.info kernel: [686256.334012] device wlan0 left promiscuous mode
Mon Aug  1 20:56:17 2022 kern.info kernel: [686256.334104] br-lan: port 4(wlan0) entered disabled state
Mon Aug  1 20:56:17 2022 kern.warn kernel: [686256.375359] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 0
Mon Aug  1 20:56:17 2022 daemon.notice hostapd: wlan0: interface state HT_SCAN->DISABLED
Mon Aug  1 20:56:23 2022 kern.warn kernel: [686262.385612] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
Mon Aug  1 20:56:23 2022 kern.warn kernel: [686262.385639] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
Mon Aug  1 20:56:23 2022 kern.info kernel: [686262.467447] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
Mon Aug  1 20:56:23 2022 kern.info kernel: [686262.468267] ath10k_pci 0000:01:00.0: wmi print 'free: 84936 iram: 13428 sram: 11224'
Mon Aug  1 20:56:24 2022 kern.info kernel: [686262.852536] ath10k_pci 0000:01:00.0: rts threshold -1
Mon Aug  1 20:56:24 2022 kern.warn kernel: [686262.858302] ath10k_pci 0000:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
Mon Aug  1 20:56:24 2022 kern.info kernel: [686262.863081] br-lan: port 4(wlan0) entered blocking state
Mon Aug  1 20:56:24 2022 kern.info kernel: [686262.868355] br-lan: port 4(wlan0) entered disabled state
Mon Aug  1 20:56:24 2022 daemon.notice hostapd: wlan0: interface state DISABLED->COUNTRY_UPDATE
[... cut ...]
Mon Aug  1 21:24:50 2022 daemon.notice hostapd: wlan0: DFS-NOP-FINISHED freq=5640 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5640 cf2=0
Mon Aug  1 21:24:50 2022 daemon.notice hostapd: wlan0: interface state DFS->DFS
Mon Aug  1 21:24:50 2022 daemon.notice hostapd: wlan0: DFS-CAC-START freq=5500 chan=100 sec_chan=1, width=0, seg0=114, seg1=0, cac_time=60s
Mon Aug  1 21:24:50 2022 daemon.err hostapd: 20/40 MHz: center segment 0 (=114) and center freq 1 (=5510) not in sync
Mon Aug  1 21:24:50 2022 daemon.err hostapd: Can't set freq params
Mon Aug  1 21:24:50 2022 daemon.err hostapd: DFS start_dfs_cac() failed, -1
```