openwrt/openwrt #9869: MT7621 third PCIe port (pcie2) cannot use

guoxiaoqiao, 2022-05-11:
version: OpenWrt 21.02.3
kernel version: 5.4.188

my board have three PCIe device, just can use first 2 device, third device cannot detected

[    1.298144] rt2880-pinmux pinctrl: pcie is already enabled
[    1.309225] mt7621-pci-phy 1e149000.pcie-phy: PHY for 0xbe149000 (dual port = 1)
[    1.324108] mt7621-pci-phy 1e14a000.pcie-phy: PHY for 0xbe14a000 (dual port = 0)
[    1.338886] mt7621-pci 1e140000.pcie: failed to parse bus ranges property: -22
[    1.453215] mt7621-pci-phy 1e149000.pcie-phy: Xtal is 40MHz
[    1.464325] mt7621-pci-phy 1e14a000.pcie-phy: Xtal is 40MHz
[    1.575360] mt7621-pci 1e140000.pcie: pcie2 no card, disable it (RST & CLK)
[    1.589227] mt7621-pci 1e140000.pcie: PCIE0 enabled
[    1.598935] mt7621-pci 1e140000.pcie: PCIE1 enabled
[    1.608654] mt7621-pci 1e140000.pcie: PCI coherence region base: 0x60000000, mask/settings: 0xf0000002
[    1.627357] mt7621-pci 1e140000.pcie: PCI host bridge to bus 0000:00
[    1.640026] pci_bus 0000:00: root bus resource [io  0x1e160000-0x1e16ffff]
[    1.653717] pci_bus 0000:00: root bus resource [mem 0x60000000-0x6fffffff]
[    1.667408] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.678377] pci 0000:00:00.0: [0e8d:0801] type 01 class 0x060400
[    1.690369] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x7fffffff]
[    1.702853] pci 0000:00:00.0: reg 0x14: [mem 0x00000000-0x0000ffff]
[    1.715405] pci 0000:00:00.0: supports D1
[    1.723388] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
[    1.735185] pci 0000:00:01.0: [0e8d:0801] type 01 class 0x060400
[    1.747183] pci 0000:00:01.0: reg 0x10: [mem 0x00000000-0x7fffffff]
[    1.759668] pci 0000:00:01.0: reg 0x14: [mem 0x00000000-0x0000ffff]
[    1.772211] pci 0000:00:01.0: supports D1
[    1.780193] pci 0000:00:01.0: PME# supported from D0 D1 D3hot
[    1.792934] pci 0000:00:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    1.808896] pci 0000:00:01.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    1.825035] pci 0000:01:00.0: [14c3:7603] type 00 class 0x028000
[    1.837043] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x000fffff]
[    1.849658] pci 0000:01:00.0: PME# supported from D0 D3hot D3cold
[    1.863089] pci 0000:00:00.0: PCI bridge to [bus 01-ff]
[    1.873507] pci 0000:00:00.0:   bridge window [io  0x0000-0x0fff]
[    1.885641] pci 0000:00:00.0:   bridge window [mem 0x00000000-0x000fffff]
[    1.899159] pci 0000:00:00.0:   bridge window [mem 0x00000000-0x000fffff pref]
[    1.913549] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.926925] pci 0000:02:00.0: [14c3:7663] type 00 class 0x000280
[    1.938956] pci 0000:02:00.0: reg 0x10: [mem 0x00000000-0x000fffff 64bit pref]
[    1.953356] pci 0000:02:00.0: reg 0x18: [mem 0x00000000-0x00003fff 64bit pref]
[    1.967752] pci 0000:02:00.0: reg 0x20: [mem 0x00000000-0x00000fff 64bit pref]
[    1.982265] pci 0000:02:00.0: supports D1 D2
[    1.990771] pci 0000:02:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[    2.003980] pci 0000:02:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s x1 link at 0000:00:01.0 (capable of 4.000 Gb/s with 5 GT/s x1 link)
[    2.032852] pci 0000:00:01.0: PCI bridge to [bus 02-ff]
[    2.043267] pci 0000:00:01.0:   bridge window [io  0x0000-0x0fff]
[    2.055401] pci 0000:00:01.0:   bridge window [mem 0x00000000-0x000fffff]
[    2.068931] pci 0000:00:01.0:   bridge window [mem 0x00000000-0x000fffff pref]
[    2.083315] pci_bus 0000:02: busn_res: [bus 02-ff] end is updated to 02
[    2.096546] pci 0000:00:00.0: BAR 0: no space for [mem size 0x80000000]
[    2.109719] pci 0000:00:00.0: BAR 0: failed to assign [mem size 0x80000000]
[    2.123584] pci 0000:00:01.0: BAR 0: no space for [mem size 0x80000000]
[    2.136757] pci 0000:00:01.0: BAR 0: failed to assign [mem size 0x80000000]
[    2.150623] pci 0000:00:00.0: BAR 8: assigned [mem 0x60000000-0x600fffff]
[    2.164147] pci 0000:00:00.0: BAR 9: assigned [mem 0x60100000-0x601fffff pref]
[    2.178531] pci 0000:00:01.0: BAR 8: assigned [mem 0x60200000-0x602fffff]
[    2.192054] pci 0000:00:01.0: BAR 9: assigned [mem 0x60300000-0x604fffff pref]
[    2.206444] pci 0000:00:00.0: BAR 1: assigned [mem 0x60500000-0x6050ffff]
[    2.219972] pci 0000:00:01.0: BAR 1: assigned [mem 0x60510000-0x6051ffff]
[    2.233495] pci 0000:00:00.0: BAR 7: assigned [io  0x1e160000-0x1e160fff]
[    2.247019] pci 0000:00:01.0: BAR 7: assigned [io  0x1e161000-0x1e161fff]
[    2.260552] pci 0000:01:00.0: BAR 0: assigned [mem 0x60000000-0x600fffff]
[    2.274073] pci 0000:00:00.0: PCI bridge to [bus 01]
[    2.283963] pci 0000:00:00.0:   bridge window [io  0x1e160000-0x1e160fff]
[    2.297481] pci 0000:00:00.0:   bridge window [mem 0x60000000-0x600fffff]
[    2.310999] pci 0000:00:00.0:   bridge window [mem 0x60100000-0x601fffff pref]
[    2.325393] pci 0000:02:00.0: BAR 0: assigned [mem 0x60300000-0x603fffff 64bit pref]
[    2.340836] pci 0000:02:00.0: BAR 2: assigned [mem 0x60400000-0x60403fff 64bit pref]
[    2.356272] pci 0000:02:00.0: BAR 4: assigned [mem 0x60404000-0x60404fff 64bit pref]
[    2.371702] pci 0000:00:01.0: PCI bridge to [bus 02]
[    2.381591] pci 0000:00:01.0:   bridge window [io  0x1e161000-0x1e161fff]
[    2.395109] pci 0000:00:01.0:   bridge window [mem 0x60200000-0x602fffff]
[    2.408631] pci 0000:00:01.0:   bridge window [mem 0x60300000-0x604fffff pref]
[    2.423541] mt7530 mdio-bus:1f: Link is Up - 1Gbps/Full - flow control off
[    2.423981] hctosys: unable to open rtc device (rtc0)
[    2.455180] VFS: Mounted root (squashfs filesystem) readonly on device 31:5.

@paraka

paraka, 2022-05-11:
@guoxiaoqiao Which is your board? Can you share with me the specific dts file for it? It looks like you probably need to claim some gpio related to PCIe2 port to get it out of reset and be able to use it. Do you have the schematic of the board? Normally gpios use for PCI are 19, 8 and 7 but this is kind of board dependant. Gpio 19 is claimed as default in 'mt7121.dtsi' file, so I guess might be the reset gpio line for you dual PCIe first two ports 0 and 1. Yo can try to add the following to you overlay pcie node in your board dts file and see if something changes:

reset-gpios = <&gpio 19 GPIO_ACTIVE_LOW>,
				<&gpio 8 GPIO_ACTIVE_LOW>,
				<&gpio 7 GPIO_ACTIVE_LOW>;

guoxiaoqiao, 2022-05-18:
@paraka Thanks for the method you gave, the problem has been solved, it's just that reset-gpios is not set correctly. Also my board is GainStrong Oolite V8.1

`[    1.295048] rt2880-pinmux pinctrl: pcie is already enabled
[    1.306136] mt7621-pci-phy 1e149000.pcie-phy: PHY for 0xbe149000 (dual port = 1)
[    1.321090] mt7621-pci-phy 1e14a000.pcie-phy: PHY for 0xbe14a000 (dual port = 0)
[    1.435984] mt7621-pci-phy 1e149000.pcie-phy: Xtal is 40MHz
[    1.447103] mt7621-pci-phy 1e14a000.pcie-phy: Xtal is 40MHz
[    1.558276] mt7621-pci 1e140000.pcie: PCIE0 enabled
[    1.567992] mt7621-pci 1e140000.pcie: PCIE1 enabled
[    1.577713] mt7621-pci 1e140000.pcie: PCIE2 enabled
[    1.587434] mt7621-pci 1e140000.pcie: PCI coherence region base: 0x60000000, mask/settings: 0xf0000002
[    1.606135] mt7621-pci 1e140000.pcie: PCI host bridge to bus 0000:00
[    1.618807] pci_bus 0000:00: root bus resource [io  0x1e160000-0x1e16ffff]
[    1.632505] pci_bus 0000:00: root bus resource [mem 0x60000000-0x6fffffff]
[    1.646215] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.657182] pci 0000:00:00.0: [0e8d:0801] type 01 class 0x060400
[    1.669182] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x7fffffff]
[    1.681671] pci 0000:00:00.0: reg 0x14: [mem 0x00000000-0x0000ffff]
[    1.694229] pci 0000:00:00.0: supports D1
[    1.702217] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
[    1.714027] pci 0000:00:01.0: [0e8d:0801] type 01 class 0x060400
[    1.726031] pci 0000:00:01.0: reg 0x10: [mem 0x00000000-0x7fffffff]
[    1.738521] pci 0000:00:01.0: reg 0x14: [mem 0x00000000-0x0000ffff]
[    1.751067] pci 0000:00:01.0: supports D1
[    1.759055] pci 0000:00:01.0: PME# supported from D0 D1 D3hot
[    1.770799] pci 0000:00:02.0: [0e8d:0801] type 01 class 0x060400
[    1.782813] pci 0000:00:02.0: reg 0x10: [mem 0x00000000-0x7fffffff]
[    1.795303] pci 0000:00:02.0: reg 0x14: [mem 0x00000000-0x0000ffff]
[    1.807851] pci 0000:00:02.0: supports D1
[    1.815839] pci 0000:00:02.0: PME# supported from D0 D1 D3hot
[    1.828560] pci 0000:00:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    1.844528] pci 0000:00:01.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    1.860475] pci 0000:00:02.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[    1.876617] pci 0000:01:00.0: [14c3:7603] type 00 class 0x028000
[    1.888635] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x000fffff]
[    1.901263] pci 0000:01:00.0: PME# supported from D0 D3hot D3cold
[    1.914731] pci 0000:00:00.0: PCI bridge to [bus 01-ff]
[    1.925161] pci 0000:00:00.0:   bridge window [io  0x0000-0x0fff]
[    1.937302] pci 0000:00:00.0:   bridge window [mem 0x00000000-0x000fffff]
[    1.950827] pci 0000:00:00.0:   bridge window [mem 0x00000000-0x000fffff pref]
[    1.965225] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.978603] pci 0000:02:00.0: [14c3:7663] type 00 class 0x000280
[    1.990634] pci 0000:02:00.0: reg 0x10: [mem 0x00000000-0x000fffff 64bit pref]
[    2.005042] pci 0000:02:00.0: reg 0x18: [mem 0x00000000-0x00003fff 64bit pref]
[    2.019443] pci 0000:02:00.0: reg 0x20: [mem 0x00000000-0x00000fff 64bit pref]
[    2.033950] pci 0000:02:00.0: supports D1 D2
[    2.042464] pci 0000:02:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[    2.055677] pci 0000:02:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s x1 link at 0000:00:01.0 (capable of 4.000 Gb/s with 5 GT/s x1 link)
[    2.084520] pci 0000:00:01.0: PCI bridge to [bus 02-ff]
[    2.094949] pci 0000:00:01.0:   bridge window [io  0x0000-0x0fff]
[    2.107095] pci 0000:00:01.0:   bridge window [mem 0x00000000-0x000fffff]
[    2.120619] pci 0000:00:01.0:   bridge window [mem 0x00000000-0x000fffff pref]
[    2.135015] pci_bus 0000:02: busn_res: [bus 02-ff] end is updated to 02
[    2.148398] pci 0000:03:00.0: [1b21:0611] type 00 class 0x010185
[    2.160412] pci 0000:03:00.0: reg 0x10: [io  0x0000-0x0007]
[    2.171531] pci 0000:03:00.0: reg 0x14: [io  0x0000-0x0003]
[    2.182643] pci 0000:03:00.0: reg 0x18: [io  0x0000-0x0007]
[    2.193751] pci 0000:03:00.0: reg 0x1c: [io  0x0000-0x0003]
[    2.204877] pci 0000:03:00.0: reg 0x20: [io  0x0000-0x000f]
[    2.215989] pci 0000:03:00.0: reg 0x24: [mem 0x00000000-0x000001ff]
[    2.228482] pci 0000:03:00.0: reg 0x30: [mem 0x00000000-0x0000ffff pref]
[    2.241958] pci 0000:03:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s x1 link at 0000:00:02.0 (capable of 4.000 Gb/s with 5 GT/s x1 link)
[    2.270787] pci 0000:00:02.0: PCI bridge to [bus 03-ff]
[    2.281211] pci 0000:00:02.0:   bridge window [io  0x0000-0x0fff]
[    2.293354] pci 0000:00:02.0:   bridge window [mem 0x00000000-0x000fffff]
[    2.306884] pci 0000:00:02.0:   bridge window [mem 0x00000000-0x000fffff pref]
[    2.321277] pci_bus 0000:03: busn_res: [bus 03-ff] end is updated to 03
[    2.334513] pci 0000:00:00.0: BAR 0: no space for [mem size 0x80000000]
[    2.347690] pci 0000:00:00.0: BAR 0: failed to assign [mem size 0x80000000]
[    2.361570] pci 0000:00:01.0: BAR 0: no space for [mem size 0x80000000]
[... 24 lines cut ...]
[    2.701708] pci 0000:00:01.0:   bridge window [io  0x1e161000-0x1e161fff]
[    2.715237] pci 0000:00:01.0:   bridge window [mem 0x60200000-0x602fffff]
[    2.728762] pci 0000:00:01.0:   bridge window [mem 0x60300000-0x604fffff pref]
[    2.743165] pci 0000:03:00.0: BAR 6: assigned [mem 0x60600000-0x6060ffff pref]
[... cut ...]
[    2.875691] pci 0000:00:02.0:   bridge window [mem 0x60600000-0x606fffff pref]
[    2.890718] mt7530 mdio-bus:1f: Link is Up - 1Gbps/Full - flow control off
`

`root@OpenWrt:/# lspci
00:00.0 PCI bridge: Device 0e8d:0801 (rev 01)
00:01.0 PCI bridge: Device 0e8d:0801 (rev 01)
00:02.0 PCI bridge: Device 0e8d:0801 (rev 01)
01:00.0 Network controller: MEDIATEK Corp. Device 7603
02:00.0 Unclassified device [0002]: MEDIATEK Corp. Device 7663
03:00.0 IDE interface: ASMedia Technology Inc. ASM1061 SATA IDE Controller (rev 02)
root@OpenWrt:/# `

closed 2022-05-18 by guoxiaoqiao

paraka, 2022-05-18:
@guoxiaoqiao Good! Thanks for letting me know!

fhaust, 2022-12-20:
I seem to be affected by this too. The ASM1061 controller on my MQMaker Witi Board is not recognized after upgrading to OpenWrt 22.03.2, r19803-9a599fee93

This worked for me before ... I think the previous OpenWRT version was 19.xx.x ... but I can't be sure.

This also seems to be the underlying cause here: https://forum.openwrt.org/t/witi-mt7621-does-not-see-the-asm1061-sata-controller/125424

If I understand the DTS correctly there is no third PCI channel configured there? 

https://github.com/openwrt/openwrt/blob/openwrt-22.03/target/linux/ramips/dts/mt7621_mqmaker_witi.dts#L78-L96

I guess I have to build a new image to change the DTS files, correct?

---

My log:

```
root@OpenWrt:~# dmesg
[    0.000000] Linux version 5.10.146 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r19803-9a599fee93) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Fri Oct 14 22:44:41 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is MQmaker WiTi
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] On node 0 totalpages: 65536
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 15 pages/cpu s30256 r8192 d22992 u61440
[    0.000000] pcpu-alloc: s30256 r8192 d22992 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00008d20
[    0.000000] Readback ErrCtl register=00008d20
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 249300K/262144K available (7026K kernel code, 631K rwdata, 828K rodata, 1292K init, 243K bss, 12844K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000015] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.015822] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.033842] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.106151] pid_max: default: 32768 minimum: 301
[    0.115482] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.129890] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.148751] rcu: Hierarchical SRCU implementation.
[    0.158659] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.174325] smp: Bringing up secondary CPUs ...
[    0.184210] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.184224] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.184239] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.184334] CPU1 revision is: 0001992f (MIPS 1004Kc)
[... 276 lines cut ...]
[   36.902277] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[   36.919416] 8021q: adding VLAN 0 to HW filter on device wan
[   39.557945] mt7530 mdio-bus:1f lan4: Link is Up - 1Gbps/Full - flow control rx/tx
[   39.735838] br-lan: port 4(lan4) entered blocking state
[   39.746458] br-lan: port 4(lan4) entered forwarding state
[   39.763545] br-lan: port 5(wlan0) entered blocking state
[   39.774971] br-lan: port 5(wlan0) entered disabled state
[   39.786814] device wlan0 entered promiscuous mode
[   39.800857] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   40.257975] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
[   40.275365] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
[   41.371378] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   41.384802] br-lan: port 5(wlan0) entered blocking state
[   41.395683] br-lan: port 5(wlan0) entered forwarding state
[   41.859850] pppoe-wan: renamed from ppp0
[   42.313435] br-lan: port 6(wlan1) entered blocking state
[   42.324225] br-lan: port 6(wlan1) entered disabled state
[   42.335947] device wlan1 entered promiscuous mode
[   42.345986] br-lan: port 6(wlan1) entered blocking state
[   42.356740] br-lan: port 6(wlan1) entered forwarding state
[   42.378166] br-lan: port 6(wlan1) entered disabled state
[   44.889395] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   44.902814] br-lan: port 6(wlan1) entered blocking state
[   44.913930] br-lan: port 6(wlan1) entered forwarding state
```

paraka, 2022-12-20:
Hi,

You have to be sure that kernel is compiled with the following options since AFAIK there is no special driver for that SATA controller:

```
CONFIG_ATA
CONFIG_SATA_AHCI
```

Also, it is probably a gpio reset missing in the device tree to get pcie2 at least detected. Try to add the following in your pcie node:

```
pcie {
   status = "okay";

   reset-gpios = <&gpio 19 GPIO_ACTIVE_LOW>,
                        <&gpio 8 GPIO_ACTIVE_LOW>,
                        <&gpio 7 GPIO_ACTIVE_LOW>;
```
Enabling these two and using the device tree with the proposed change should be enough to get your device working.

fhaust, 2022-12-21:
@paraka Hmm no ... that has not done the trick.

---

What I did:

1. Changed the DTS

```
--- a/target/linux/ramips/dts/mt7621_mqmaker_witi.dts
+++ b/target/linux/ramips/dts/mt7621_mqmaker_witi.dts
@@ -12,7 +12,9 @@
 
                reset {
                        label = "reset";
-                       gpios = <&gpio 18 GPIO_ACTIVE_LOW>;
+                       gpios = <&gpio 18 GPIO_ACTIVE_LOW>,
+                               <&gpio 8 GPIO_ACTIVE_LOW>,
+                               <&gpio 7 GPIO_ACTIVE_LOW>;
                        linux,code = <KEY_RESTART>;
                };
        };
```

2. enabled CONFIG_ATA and CONFIG_SATA_AHCI
3. `make defconfig download clean world`
4. flashed new image

(do I have to do something specific to get the changed DTS into the final image?)

---

Logs:

```
root@OpenWrt:~# lspci
00:00.0 PCI bridge: Device 0e8d:0801 (rev 01)
00:01.0 PCI bridge: Device 0e8d:0801 (rev 01)
01:00.0 Network controller: MEDIATEK Corp. MT7662E 802.11ac PCI Express Wireless Network Adapter
02:00.0 Network controller: MEDIATEK Corp. MT7662E 802.11ac PCI Express Wireless Network Adapter
```

Kernel Log:
(there are some error messages related to pci if I understand that correctly)

```
[    0.000000] Linux version 5.10.146 (fhofmann@magret) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r19803-9a599fee93) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Fri Oct 14 22:44:41 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is MQmaker WiTi
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] On node 0 totalpages: 65536
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 15 pages/cpu s30256 r8192 d22992 u61440
[    0.000000] pcpu-alloc: s30256 r8192 d22992 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[... 321 lines cut ...]
[   77.721286] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[   77.736366] 8021q: adding VLAN 0 to HW filter on device wan
[   80.250343] br-lan: port 5(wlan0) entered blocking state
[   80.261113] br-lan: port 5(wlan0) entered disabled state
[   80.272684] device wlan0 entered promiscuous mode
[   80.273115] mt7530 mdio-bus:1f lan4: Link is Up - 1Gbps/Full - flow control rx/tx
[   80.308909] br-lan: port 4(lan4) entered blocking state
[   80.319463] br-lan: port 4(lan4) entered forwarding state
[   80.331938] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   80.913388] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
[   80.928119] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
[   81.847147] pppoe-wan: renamed from ppp0
[   81.932875] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   81.946313] br-lan: port 5(wlan0) entered blocking state
[   81.957076] br-lan: port 5(wlan0) entered forwarding state
[   82.693523] br-lan: port 6(wlan1) entered blocking state
[   82.704235] br-lan: port 6(wlan1) entered disabled state
[   82.715912] device wlan1 entered promiscuous mode
[   82.726128] br-lan: port 6(wlan1) entered blocking state
[   82.737317] br-lan: port 6(wlan1) entered forwarding state
[   82.988696] br-lan: port 6(wlan1) entered disabled state
[   85.294548] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   85.307947] br-lan: port 6(wlan1) entered blocking state
[   85.318682] br-lan: port 6(wlan1) entered forwarding state
```

paraka, 2022-12-21:
@fhaust The enabled stuff for SATA is correct. The problem with your board is that pcie2 is not being detected so maybe a extra sleep is needed to get the port link up like it seems to be for some boards. 

Can you try to adapt this patch:

https://lore.kernel.org/linux-pci/20221209071703.2891714-1-sergio.paracuellos@gmail.com/T/#u

To the mt7621-pcie driver which should be in drivers/staging  for the 5.10 kernel version?

Normally detection problems are because of bad gpio reset (seems to not be the case since we have already tried it) or extra delay needed. You can also try to play with the value of _PERST_DELAY_MS_ if adding the extra msleep in the driver does not the trick for us.

Hope this helps.

fhaust, 2022-12-21:
@paraka Should I revert the gpio changes?

paraka, 2022-12-21:
> @paraka Should I revert the gpio changes?

@fhaust  Not sure yet, maybe you can try both?

fhaust, 2022-12-21:
Ok ... I changed `./build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.146/drivers/staging/mt7621-pci/pci-mt7621.c` to read: 

```
  static int mt7621_pcie_init_port(struct mt7621_pcie_port *port)
  {
  ▷⋅struct mt7621_pcie *pcie = port->pcie;
  ▷⋅struct device *dev = pcie->dev;
  ▷⋅u32 slot = port->slot;
  ▷⋅int err;
  
  ▷⋅err = phy_init(port->phy);
  ▷⋅if (err) {
  ▷⋅▷⋅dev_err(dev, "failed to initialize port%d phy\n", slot);
  ▷⋅▷⋅return err;
  ▷⋅}⋅
  ▷⋅msleep(PERST_DELAY_MS);
  
  ▷⋅err = phy_power_on(port->phy);
  ▷⋅if (err) {
  ▷⋅▷⋅dev_err(dev, "failed to power on port%d phy\n", slot);
  ▷⋅▷⋅phy_exit(port->phy);
  ▷⋅▷⋅return err;
  ▷⋅}
  ▷⋅msleep(PERST_DELAY_MS);
  
  ▷⋅port->enabled = true;
  
  ▷⋅return 0;
  }
```

(not sure where the sleep goes in this version, both should not hurt, right?)

Recompiling now.

fhaust, 2022-12-21:
@paraka Hmm no ... still no luck. (Did I change the wrong file?)

fhaust, 2022-12-21:
... I just realized I had not plugged in any hdd ... that should not hurt, right? lspci should list the SATA controller even if there is no disk plugged in?

paraka, 2022-12-21:
You have made a bad change in the driver. Just insert the msleep in the following line:
https://elixir.bootlin.com/linux/v5.10.146/source/drivers/staging/mt7621-pci/pci-mt7621.c#L525

There is no need to plug any hdd since the problem is a bad pci port detection.

paraka, 2022-12-21:
> @paraka Hmm no ... that has not done the trick.
>
> What I did:
>
> 1. Changed the DTS
>
> ```
> --- a/target/linux/ramips/dts/mt7621_mqmaker_witi.dts
> +++ b/target/linux/ramips/dts/mt7621_mqmaker_witi.dts
> @@ [...]
@fhaust 
Sorry I missread this. This is not the pci related reset gpios but the reset key. You have to change pcie overlay node as follows (in your board is here: https://github.com/openwrt/openwrt/blob/master/target/linux/ramips/dts/mt7621_mqmaker_witi.dts#L74):

```
&pcie {
   status = "okay";

   reset-gpios = <&gpio 19 GPIO_ACTIVE_LOW>,
                        <&gpio 8 GPIO_ACTIVE_LOW>,
                        <&gpio 7 GPIO_ACTIVE_LOW>;
};
```

fhaust, 2022-12-21:
Woohoo! ... one of those changes worked :sweat_smile: 

```
root@OpenWrt:~# lspci
00:00.0 PCI bridge: Device 0e8d:0801 (rev 01)
00:01.0 PCI bridge: Device 0e8d:0801 (rev 01)
00:02.0 PCI bridge: Device 0e8d:0801 (rev 01)
01:00.0 Network controller: MEDIATEK Corp. MT7662E 802.11ac PCI Express Wireless Network Adapter
02:00.0 Network controller: MEDIATEK Corp. MT7662E 802.11ac PCI Express Wireless Network Adapter
03:00.0 SATA controller: ASMedia Technology Inc. ASM1062 Serial ATA Controller (rev 01)
```

(That is changing the DTS and adding the sleep)

@paraka Thanks for the help! Can I get you a beverage somehow?

fhaust, 2022-12-21:
@paraka I think there is no way to direct message you here. Feel free to send me your paypal contact at fho@f12n.de

paraka, 2022-12-21:
> @paraka I think there is no way to direct message you here. Feel free to send me your paypal contact at [fho@f12n.de](mailto:fho@f12n.de)

No worries, I was just kidding. I am happy that all is working now :)

gioreva, 2023-02-19:
> @fhaust https://elixir.bootlin.com/linux/v5.10.146/source/drivers/staging/mt7621-pci/pci-mt7621.c#L525
There is already on line 525
Then need to double the delay ?

fhaust, 2024-08-17:
Two years later and I have the same problem again after upgrading OpenWRT :sweat_smile:

fhaust, 2024-08-18:
> @fhaust Sorry I missread this. This is not the pci related reset gpios but the reset key. You have to change pcie overlay node as follows (in your board is here: [...]

@paraka No idea if you are still around, but are you sure that is gpio 19 and not 18?

paraka, 2024-08-18:
Hi!

> @paraka <https://github.com/paraka> Hmm no ... that has not done the
> trick.
> What I did:
>
>    1. Changed the DTS
>
> --- a/target/linux/ramips/dts/mt7621_mqmaker_witi.dts
> +++ [...]

It depends of your board. Normally 19 is used on most of them so it is
reclamed by default in dtsi. Nothing changed in the driver since ir worked
for you so I don’t know why it happened again. Check gpios in dts and
delays in code to see If it works for you again. It should.

fhaust, 2024-08-18:
Just adding the reset-gpios in the DTS seems to work. Is there some way to upstream this change?

paraka, 2024-08-18:
Read how to properly do a PR in the openwrt git repository to get your
board changes upstrrsmed and avoid already fixed regresions like this.

Thanks.