openwrt/openwrt #9874: TP-Link Archer A10 AC2600 error

wilfredogr, 2022-05-11:
I am trying to install Open WRT in my router AC2600, and it shows an error: 
**_Error code: 4503
The uploaded file was not accepted._**

How can I solve this issue?

Regards

tboege, 2022-05-12:
You need the [from factory ](https://downloads.openwrt.org/releases/21.02.3/targets/ipq806x/generic/openwrt-21.02.3-ipq806x-generic-tplink_c2600-squashfs-factory.bin) firmware to go from stock to Openwrt. 
Afterwards, when you want to go from one version of Openwrt to another, you can use the [sysupgrade ](https://downloads.openwrt.org/releases/21.02.3/targets/ipq806x/generic/openwrt-21.02.3-ipq806x-generic-tplink_c2600-squashfs-sysupgrade.bin)

wilfredogr, 2022-05-12:
Sorry, I think my post description is wrong. I have my router firmware
original and trying to install OpenWRT.

obsy, 2022-05-12:
https://deviwiki.com/wiki/TP-LINK_Archer_A10

Archer A10 is mediatek, not ipq806x. A10 is not supported.

wilfredogr, 2022-05-13:
> @wilfredogr <https://github.com/wilfredogr> TP-Link Archer A10 isn't
> officially supported by OpenWrt.
>
> Cheers,
> Piotr

Ok, thanks.

asterismo, 2023-09-11:
Hi, is there any unofficial firmware for this router?