openwrt/openwrt #9879: ASUS RT-AC58U returning to stock process not working on 22.03.0-rc1

cosminpolifronie, 2022-05-12:
On 22.03.0-rc1 the ASUS restoration tool found the router and uploaded the firmware. After it reboots it boots up OpenWRT instead of the ASUS stock firmware.

Downgrading to 21.02.3 first and then following the process successfully restored the stock ASUS firmware.

https://openwrt.org/toh/asus/rt-ac58u#returning_to_stock

pkgadd, 2022-05-13:
Are you sure that you followed the procedures you quoted?

If you did, there would be no OpenWrt anymore to boot into (as you'd have removed the ubi volume and overwritten the running firmware with trash (the busybox binary)) - only then, the firmware restoration tool can do its job (as it's not smart enough to deal with unexpected ubi volumes).

cosminpolifronie, 2022-05-13:
100% sure. I followed the same instructions on both 22.03.0-rc1 and 21.02.3. Only on the last one they restored to ASUS firmware.

I've found the same issue discussed on the OpenWRT forums in the past.

pepe2k, 2022-05-13:
@cosminpolifronie 

The only way to actually debug this problem I can think of is to attach to the serial console on the device and dump whole output during OpenWrt -> vendor's firmware migration, on both versions (21.02.x and 22.03.x).

I'm not sure if this should be considered as an project issue/bug at all. Documentation (which includes instructions for restoring vendor's firmware) is made by community (it's a Wiki).

> I've found the same issue discussed on the OpenWRT forums in the past.

Would be very useful to provide a link.

Cheers,
Piotr

cosminpolifronie, 2022-05-13:
@pepe2k I had these two topics in mind, but for both of them it seems there was no removal of ubi partition done.
https://forum.openwrt.org/t/revert-to-stock-firmware-on-asus-map-ac2200/127001/3
https://forum.openwrt.org/t/how-to-revert-oopenwrt-installation-back-to-stock-firmware-on-asus-gt-ac5300/122177

> The only way to actually debug this problem I can think of is to attach to the serial console on the device and dump whole output during OpenWrt -> vendor's firmware migration, on both versions (21.02.x and 22.03.x).

I think I can help with that, but I'm away from home for a couple of weeks. I will try to solder a connector to the serial port and then read it with an Arduino as I don't have a serial to USB adaptor at hand. I'll be back in some time with details.

mesposito-pattan, 2022-09-23:
I am also experiencing this issue when attempting to go back to a stock firmware on an ASUS RT-ACRH13. Running the commands as directed in the guide on this page (https://openwrt.org/toh/asus/rt-ac58u) gives the following output via SSH

```
root@OpenWrt:~# ubirmvol /dev/ubi0 -N linux2
root@OpenWrt:~# sysupgrade -F /bin/busybox
Thu Sep 22 19:08:07 EDT 2022 upgrade: Image not in /tmp, copying...
Thu Sep 22 19:08:07 EDT 2022 upgrade: Image metadata not present
Image check failed but --force given - will update anyway!
Thu Sep 22 19:08:08 EDT 2022 upgrade: Saving config files...
Thu Sep 22 19:08:09 EDT 2022 upgrade: Commencing upgrade. Closing all shell sessions.
Command failed: Connection failed
root@OpenWrt:~# Connection to 192.168.77.1 closed by remote host.
Connection to 192.168.77.1 closed.
```
Going back to 21.02.3 solved the issue like the original poster, after running the commands I was able to use the Asus firmware recovery to load the stock firmware back.

JoshuaCH-M7NQL, 2023-06-22:
Figured I'd add this for anyone looking how to return to stock asus firmware, it wasn't 100% clear to me however i'd say as a temp way to return to stock downgrade to 21.02.3, it's still in the openwrt download server & once done just follow the instructions on the return to stock bit as they seem to work with 21.02.3 but not the latest 22.03 while technically a problem it is fixable by downgrading as a temp fix.

thank you for posting this "100% sure. I followed the same instructions on both 22.03.0-rc1 and 21.02.3. Only on the last one they restored to ASUS firmware." wasn't till i looked at it a few times to realize what you meant.

if possible not sure if those in charge of openwrt look over this but maybe add about a downgrade fix to the return to stock section so people don't just assume their stuck with the firmware.

regards,
Josh

RaptorXenon, 2025-02-06:
I wanna say thank you very much. You saved me. I spent nearly three hours figure out what's wrong and why the OpenWrt keep poping up after using ASUS official restore tool. I ensured the code was run correctly and the router is in recovery mode but still keep going back to OpenWrt. Until downgrading to 21.02.3 OpenWRT and do all the steps again, the router eventually get back to the official firmware. Thank you very much.
Yet, I still wish to know what's the problem between these two systems that causes failed to restore back to the offiical firmware in v23 but sucessful from v21 OpenWrt.