openwrt/openwrt #9881: [22.03] NDP relay mode makes it impossible to connect to the IPv6 Internet.

WinLinux1028, 2022-05-13:
I'm using latest 22.03 snapshot on Linksys E8450.
Here is a list of what I know about this issue.
1. When the PC is connected to the router, the IPv6 address can be obtained correctly.
2. When I try IPv6 connection on the PC, all packets are dropped.
```
$ ping -6 google.com
PING google.com(nrt12s51-in-x0e.1e100.net (2404:6800:4004:801::200e)) 56 data bytes
^C
--- google.com ping statistics ---
8 packets transmitted, 0 received, 100% packet loss, time 7083ms
```
3. When I try IPv6 connection on the router, I can communicate without any problem.
```
# ping -6 google.com
PING google.com (2404:6800:4004:813::200e): 56 data bytes
64 bytes from 2404:6800:4004:813::200e: seq=0 ttl=116 time=20.414 ms
64 bytes from 2404:6800:4004:813::200e: seq=1 ttl=57 time=15.904 ms
64 bytes from 2404:6800:4004:813::200e: seq=2 ttl=116 time=16.378 ms
64 bytes from 2404:6800:4004:813::200e: seq=3 ttl=116 time=16.273 ms
64 bytes from 2404:6800:4004:813::200e: seq=4 ttl=57 time=16.029 ms
^C
--- google.com ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 15.904/16.999/20.414 ms
``` 
4. After communication from the PC to the IPv6 address assigned to the router itself or to the IPv6 address of the modem, the PC can connect to the IPv6 Internet.

WinLinux1028, 2022-05-13:
The contents of `/etc/config/network`, `/etc/config/dhcp`, and '/etc/config/firewall' are written below.

/etc/config/network:
```
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option packet_steering '1'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.0.1'
        option netmask '255.255.255.0'
        option ip6assign '64'

config interface 'wan'
        option device 'wan'
        option proto 'static'
        option ipaddr '192.168.255.2'
        option netmask '255.255.255.0'
        option gateway '192.168.255.1'
        list dns '1.1.1.1'
        list dns '8.8.8.8'

config interface 'wan6'
        option device 'wan'
        option proto 'dhcpv6'
```

/etc/config/dhcp:
```
config dnsmasq
        option domainneeded '1'
        option boguspriv '1'
        option filterwin2k '0'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option nonegcache '0'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
        option nonwildcard '1'
        option localservice '1'
        option ednspacket_max '1232'
        list server '127.0.0.1#5054'
        list server '127.0.0.1#5053'

config dhcp 'lan'
        option interface 'lan'
        option force '1'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option dhcpv4 'server'
        list dhcp_option '6,1.1.1.1,8.8.8.8'
        option ra 'relay'
        option dhcpv6 'relay'
        option ndp 'relay'

config dhcp 'wan6'
        option interface 'wan6'
        option ra 'relay'
        option dhcpv6 'relay'
        option ndp 'relay'
        option master '1'

config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/tmp/hosts/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'
        option loglevel '4'
```

/etc/config/firewall:
```
config defaults
        option flow_offloading '1'
        option flow_offloading_hw '1'
        option synflood_protect '1'
        option drop_invalid '1'
        option input 'DROP'
        option output 'DROP'
        option forward 'DROP'

config zone
        option name 'lan'
        list network 'lan'
        option input 'ACCEPT'
        option output 'ACCEPT'
        option forward 'ACCEPT'

config zone
        option name 'wan'
        list network 'wan'
        list network 'wan6'
        option input 'DROP'
        option forward 'DROP'
        option output 'ACCEPT'
        option mtu_fix '1'
        option masq '1'

config forwarding
        option family 'any'
        option src 'lan'
        option dest 'wan'

config forwarding
        option family 'any'
        option src 'wan'
        option dest 'lan'

config rule
        option name 'Allow-DHCP-Renew'
        option src 'wan'
        option proto 'udp'
        option dest_port '68'
        option target 'ACCEPT'
        option family 'ipv4'

config rule
        option name 'Allow-Ping'
        option src 'wan'
        option proto 'icmp'
        option icmp_type 'echo-request'
        option family 'ipv4'
        option target 'ACCEPT'

config rule
        option name 'Allow-IGMP'
        option src 'wan'
        option proto 'igmp'
        option family 'ipv4'
        option target 'ACCEPT'

config rule
        option name 'Allow-DHCPv6'
        option src 'wan'
        option proto 'udp'
        option src_ip 'fc00::/6'
        option dest_ip 'fc00::/6'
        option dest_port '546'
        option family 'ipv6'
        option target 'ACCEPT'

config rule
        option name 'Allow-MLD'
        option src 'wan'
        option proto 'icmp'
        option src_ip 'fe80::/10'
        list icmp_type '130/0'
        list icmp_type '131/0'
        list icmp_type '132/0'
        list icmp_type '143/0'
        option family 'ipv6'
        option target 'ACCEPT'

config rule
        option name 'Allow-ICMPv6-Input'
        option src 'wan'
        option proto 'icmp'
        list icmp_type 'echo-request'
        list icmp_type 'echo-reply'
        list icmp_type 'destination-unreachable'
        list icmp_type 'packet-too-big'
        list icmp_type 'time-exceeded'
        list icmp_type 'bad-header'
        list icmp_type 'unknown-header-type'
        list icmp_type 'router-solicitation'
        list icmp_type 'neighbour-solicitation'
        list icmp_type 'router-advertisement'
        list icmp_type 'neighbour-advertisement'
        option limit '1000/sec'
        option family 'ipv6'
        option target 'ACCEPT'

config rule
        option name 'Allow-ICMPv6-Forward'
        option src 'wan'
        option dest '*'
        option proto 'icmp'
        list icmp_type 'echo-request'
        list icmp_type 'echo-reply'
        list icmp_type 'destination-unreachable'
        list icmp_type 'packet-too-big'
        list icmp_type 'time-exceeded'
        list icmp_type 'bad-header'
        list icmp_type 'unknown-header-type'
        option limit '1000/sec'
        option family 'ipv6'
        option target 'ACCEPT'

config rule
        option name 'Allow-IPSec-ESP'
        option src 'wan'
        option dest 'lan'
        option proto 'esp'
        option target 'ACCEPT'

config rule
        option name 'Allow-ISAKMP'
        option src 'wan'
        option dest 'lan'
        option dest_port '500'
        option proto 'udp'
        option target 'ACCEPT'
```

WinLinux1028, 2022-05-13:
The router's route list looked like this.
```
# ip -6 route
default from (global prefix)::/64 via fe80::da0f:99ff:fed9:6ed1 dev wan  metric 384 
(global prefix):1111:1111:1111:1111 dev wan  metric 1024      // Upstream modem
(global prefix):1db8:94fb:ba69:33c5 dev br-lan  metric 1024   // My PC
(global prefix)::/64 dev wan  metric 256 
unreachable (global prefix)::/64 dev lo  metric 2147483647 
fe80::/64 dev br-lan  metric 256 
fe80::/64 dev eth0  metric 256 
fe80::/64 dev wan  metric 256 
fe80::/64 dev wlan0  metric 256 
fe80::/64 dev wlan1  metric 256 
anycast (global prefix):: dev wan  metric 0 
anycast fe80:: dev eth0  metric 0 
anycast fe80:: dev br-lan  metric 0 
anycast fe80:: dev wan  metric 0 
anycast fe80:: dev wlan0  metric 0 
anycast fe80:: dev wlan1  metric 0 
multicast ff00::/8 dev br-lan  metric 256 
multicast ff00::/8 dev eth0  metric 256 
multicast ff00::/8 dev wan  metric 256 
multicast ff00::/8 dev wlan0  metric 256 
multicast ff00::/8 dev wlan1  metric **256**
```

howl, 2022-05-14:
Few days ago I did a "because I just want to mess up a little" setup. I make my old router Archer C7 v2 to connect to my main router in the WAN zone. I setup it just as you have, but, I could only be able to get neighbours by setting up also
        option ra 'relay'
        option dhcpv6 'relay'
        option ndp 'relay'
in the wan interface, not only in the wan6.

Anyways, despite I then get the neighbours discovered in the Archer C7, ipv6 connectivity was not possible in the connected devices to that router, only ipv4, and the router itself could do ipv4 and ipv6 just as you say.

If you want check if settings ndp also in the wan interface could work for you. Check before Status -> Routes and see if neighbours appear, if not do the wan thing, restart the network interfaces and wait a sometime to see if some neighbours start to appear (I assume there are more devices conected to the router present in the wan zone).

MentalPower, 2022-08-25:
Anything we (users) can do to run this to ground? Package bisects, different configs, etc?

oxwivi, 2022-11-20:
I seem to be facing the same issue. I simply followed the [IPv6 relay](https://openwrt.org/docs/guide-user/network/ipv6/configuration#ipv6_relay) example in the wiki, to the same behavior described in OP: router can do IPv6, clients get IPv6 addresses but no connection.

However, I do have something to add: I'm facing this issue with my new Belkin RT3200, but I've been using IPv6 on the same network, with the same settings running in 22.03 on the Archer C7 that was replaced. The only difference I can think of is, the C7 grandfathered the IPv6 settings from the previously installed OW versions instead of being freshly flashed like OW in the RT3200. (I manually redid all the settings on the RT3200 instead of taking a backup of C7 and applying it to the new device). The C7 still retains the settings so I'll post it here later.

oxwivi, 2022-11-20:
I checked, and really, there's no difference to post about:

Archer C7:
```
config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option dhcpv4 'server'
	option ra 'relay'
	option dhcpv6 'relay'
	option ndp 'relay'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'
	option master '1'
	option ra 'relay'
	option dhcpv6 'relay'
	option ndp 'relay'
```
Belkin RT3200:
```
config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option ndp 'relay'
	option ra 'relay'
	option dhcpv6 'relay'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'
	option master '1'
	option ra 'relay'
	option dhcpv6 'relay'
	option ndp 'relay'
```

Which is all the more confusing to me. Are there any other relevant settings that might be worth looking at? I've downloaded backup configurations of both to easily compare.

oxwivi, 2022-11-29:
I forgot to mention, both devices described are running OW 22.03.2.

Today I discovered that IPv6 works on some devices. It's working on an old HP laptop I have where I tried to verify IPv6 on a whim. I made sure to try on other devices, my newer ASUS laptop and couple 4 year old Android phones but IPv6 didn't work on them.

aredridel, 2023-04-23:
Wild. I'm also seeing this — some devices can connect, some can't. I see the packets reaching and maybe even fully traversing the openwrt node, but not working past that.

weyou, 2023-07-25:
Confirm that this problem still exists on OpenWrt 22.03.3 (Linksys WRT1200AC)

sianzhan, 2023-07-28:
Confirm that this problem still exists on OpenWrt 22.03.5 (Xiaomi Mi Router 4C)

cmd2001, 2023-08-03:
Confirm that this problem still exists on OpenWrt 22.03.5 (D-Link DIR-882)

One more thing is that I found IPv6 works after the client ping the **WAN IP**(of course IPv6) of the router.
I used `ip -6 neigh show proxy | grep {IPv6_of_the_Client}` to see that happened, then I found:
before ping:
```
xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:f8db dev wan  used 450/1343/446 probes 6 FAILED
```
after ping:
```
xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:f8db dev wan  used 450/1343/446 probes 6 FAILED
xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:xxxx:f8db dev br-lan lladdr xx:xx:xx:xx:xx:xx ref 1 used 1/63/3 probes 0 DELAY
```
A new NDP item with the client's IP is created at the `br-lan` after the ping.
I do not know much about IPv6, however, I think it is weird that the NDP item for the client is not created at `br-lan` side at the beginning, and I wonder if there is a easy workaround such as creating NDP item when the router receive a IPv6 packet from a client from `br-lan` for the frist time.

smilekevin, 2023-08-28:
what I found is openwrt is not trying to discover the neighbor on lan(br-lan). it only tries it on wan6(eth0). 

(I masked my actual ipv6 prefix)

```
[root@OpenWrt:12:09 AM ~] #  ip -6 neigh |grep c5e3
240e:xxxx:xxxx:xxxx:4c20:d790:fa85:c5e3 dev eth0 FAILED 
```

and trying to ping this global ip on lan it failed. 
```
[root@OpenWrt:11:58 PM ~] # ping 240e:xxxx:xxxx:xxxx:4c20:d790:fa85:c5e3 -I br-lan
PING 240e:xxxx:xxxx:xxxx:4c20:d790:fa85:c5e3 (240e:xxxx:xxxx:xxxx:4c20:d790:fa85:c5e3): 56 data bytes
ping: sendto: Network unreachable
```

and then I found in the routing table, there is no route on lan for make the router ever be able to connect to the devices if using devices global ip. 

so I did this. 
```
route -A inet6 add 240e:xxxx:xxxx:xxxx::/64 dev br-lan
```
after that it worked that proxy is added for any new connected device.

bluewavenet, 2023-09-01:
@WinLinux1028 @howl @MentalPower @oxwivi @aredridel @Headcrabed @FLIER001 @weyou @sianzhan @cmd2001 @smilekevin 
```

        option dhcpv6 'relay'
        option ndp 'relay'
```

Neither of these options are supported at all.
The dnsmasq init.d reads in option dhcp6 but does nothing with it.
Option ndp is not anywhere to be seen in the init.d or in the dnsmasq man page.....

Am I misunderstanding something?
The user guide certainly needs updating...
https://openwrt.org/docs/guide-user/base-system/dhcp

bluewavenet, 2023-09-03:
@KA2107 
On the lan side, Dnsmasq is the default for both. Also if enabled odhcpd is controlled by the same init.d file.
On the wan side, if dhcp id used for ipv6 then it is controlled by the upstream dhcp server, whatever that is.

ZhouZiY, 2023-09-05:
I have the same problem😫.

bluewavenet, 2023-09-06:
@KA2107 
Ah! Yes you are correct. It seems odhcpc picks up the ipv6 configs (like the ra related ones) even if it is set to `maindhcp='0'` in the config. It looks like there is no way to use dnsmasq for these options - at least not in OpenWrt.

howl, 2023-09-06:
@bluewavenet I think is possible but not by only setting it in configuration, I think is needed to replace the package with a more complete one with ipv6 support that is not installed by default because the use of odhcpd.

cmd2001, 2023-09-08:
A Possible Solution is: Delete IPv6 ULA-Prefix from Global network options.
I tried manually adding route as [smilekevin](https://github.com/smilekevin) mentioned in https://github.com/openwrt/openwrt/issues/9881#issuecomment-1696024576, then I found `ip -6 neigh | grep {IPv6_of_the_Client}` shows the NDP item for new client, but the status of such item is always FAILED, and the IPv6 of a client could be pinged from other clients but especially NOT from the router.
Eventually, I found [2021 Latest OpenWrt Router IPv6 Configuration](https://www.right.com.cn/forum/thread-4085840-1-1.html) from www.right.com.cn, a Chinese forum where hardwares/softwares of routers are discussed.
The Owner of that thread mentioned to delete IPv6 ULA-Prefix from global network options, after doing this, all my devices successfully accessed IPv6 Internet.

smilekevin, 2023-09-10:
> A Possible Solution is: Delete IPv6 ULA-Prefix from Global network options. I tried manually adding route as [smilekevin](https://github.com/smilekevin) mentioned in [#9881 (comment)](https://github.com/openwrt/openwrt/issues/9881#issuecomment-1696024576), then I found `ip -6 neigh | grep {IPv6_of_the_Client}` shows the NDP item for new client, but the status of such item is always FAILED, and the IPv6 of a client could be pinged from other clients but especially NOT from the router. Eventually, I found [2021 Latest OpenWrt Router IPv6 Configuration](https://www.right.com.cn/forum/thread-4085840-1-1.html) from [www.right.com.cn](http://www.right.com.cn), a Chinese forum where hardwares/softwares of routers are discussed. The Owner of that thread mentioned to delete IPv6 ULA-Prefix from global network options, after doing this, all my devices successfully accessed IPv6 Internet.

it also worked for me without adding a route entry.

vikingloudmouth, 2023-09-18:
I have the same issue as [this comment](https://github.com/openwrt/openwrt/issues/9881#issuecomment-1664010483) in that I need to ping the wan ipv6 address of the router to get my linux or windows computer working. My android phone works without any issue though and immediately gets working ipv6 internet without needing to do anything

oxwivi, 2023-12-04:
As stated in my previous comments, on my previous OpenWrt router Archer C7 v5, IPv6 used to work with `ra`, dhcpv6` and `ndp` all set to `relay`. Neither of the workarounds mentioned here worked me either. However, I came across a solution: a post on the forum mentioned Windows and Android not supporting DHCPv6 requires disabling it and configuring SLAAC instead, as shown here:  [IPv6: DHCPv6 vs SLAAC](https://github.com/fakemanhk/openwrt-jp-ipoe/discussions/2)

It worked for me, I have IPv6 on all of the clients on my network. Keep in mind, with the previous `relay` configuration I used to sometimes get IPv6 working on Android and Windows machines, and sometimes not. And never on my main Linux machine. But SLAAC worked for some reason and doesn't appear to have stopped or hiccuped since.

**NB** For context, ISP assigns `/64` prefix over PPPoE. That was the case with my previous router, and that is the case now. Make of it what you will.

fakemanhk, 2023-12-05:
> Keep in mind, with the previous DHCPv6 `relay` configuration I used to sometimes get IPv6 working on Android and Windows machines, and sometimes not.

For Android/ChromeOS, DHCPv6 NEVER WORKS because Google decides not to support it (you can search), when you mention "sometimes working" probably because some of your configuration might have passed down the stateless to downstream.

Technically you don't have to disable DHCPv6 for Android to work (it will ignore the not usable one), but for sake of simplicity, in my guide I chose to disable DHCPv6 entirely so that I don't have to manage both (imagine you are looking at DHCPv6 lease however only a few entries there)

oxwivi, 2023-12-05:
> For Android/ChromeOS, DHCPv6 NEVER WORKS because Google decides not to support it (you can search), when you mention "sometimes working" probably because some of your configuration might have passed down the stateless to downstream.

Yes, I did see your referenced links about Google's response. But with my previous router, I'm pretty sure all of my machines had working IPv6, with all `relay` and without explicitly enabling SLAAC. But then it was also a year ago, perhaps I'm remembering wrong.

Anyhow, this neither on-topic nor fruitful to argue about what I may or may not remember; I'll refrain from commenting further.

takeru, 2024-01-14:
In my environment, even though some clients can obtain a public v6 address, they are unable to communicate via v6. The same happens when I ping outside the WAN, it becomes possible to communicate via IPv6.

When I ping, the client gets added to the neighbor list. Entries for the client’s address are also added to the routing table.

It seems that the management of entries is done by this:
https://openwrt.org/docs/techref/odhcpd

https://github.com/openwrt/odhcpd/tree/master

It might be good to investigate what kind of communication the clients that don’t have an entry added are triggering.

takeru, 2024-01-14:
According to the information on https://openwrt.org/docs/techref/odhcpd, the updated version of https://datatracker.ietf.org/doc/html/rfc6204, which is https://tex2e.github.io/rfc-translater/html/rfc7084.html, states that 

> ULA addressing is useful where the IPv6 CE router has multiple LAN interfaces with hosts that need to communicate with each other. If the IPv6 CE router has only a single LAN interface (IPv6 link), then link-local addressing can be used instead.

Following this, as mentioned in https://github.com/openwrt/openwrt/issues/9881#issuecomment-1711820408, I found that removing the IPv6 ULA-Prefix allowed clients to communicate without needing to ping. However, it seems to take tens of seconds for communication to be established. Instead of 'This prefix is randomly generated at first install,' having a default empty field might be a better option.

bayasdev, 2024-03-27:
> so I did this.
> 
> ```
> route -A inet6 add 240e:xxxx:xxxx:xxxx::/64 dev br-lan
> ```
> 
> after that it worked that proxy is added for any new connected device.

This worked for me after removing ULA as well as setting WAN6 all to master relay and LAN all to relay mode

138161, 2024-04-13:
opkg install owipcalc
/etc/hotplug.d/iface/80-reset-route6
#!/bin/sh
wan_dev="wan6"
[ "$HOTPLUG_TYPE" = "iface" ] || exit 0
[ "$INTERFACE" = "$wan_dev" ] || exit 0
RTMETRIC=127
. /lib/functions/network.sh
network_get_physdev lan_dev lan || exit 0
ifup_cb() {
        local _lan_dev="$1"
        local _metric="$2"
        local wan_subnet
        network_get_subnet6 wan_subnet "$wan_dev" || return
        _wan_network=$(owipcalc "${wan_subnet}" network)
        ip -6 route replace "$_wan_network" dev "$_lan_dev" metric "$_metric"
}

ifdown_cb() {
        local _lan_dev="$1"
        local _metric="$2"

        ip -6 route flush dev "$_lan_dev" metric "$_metric"
}

case "$ACTION" in
        ifup)
                ifup_cb "$lan_dev" "$RTMETRIC"
                ;;
        ifdown)
                ifdown_cb "$lan_dev" "$RTMETRIC"
                ;;
        ifupdate)
                ifdown_cb "$lan_dev" "$RTMETRIC"
                sleep 1
                ifup_cb "$lan_dev" "$RTMETRIC"
                ;;
        *)
                ;;
esac

exit 0
chmod +X /etc/hotplug.d/iface/80-reset-route6

ip6tables -I INPUT -p ipv6-nonxt -m length --length 40 -j ACCEPT
ip6tables -I FORWARD -p ipv6-nonxt -m length --length 40 -j ACCEPT
ip6tables -I zone_lan_input -p udp   --sport 546 --dport 547  -j ACCEPT

ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 1  -m limit --limit 100/s --limit-burst 10  -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 2  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 3  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 4  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 128  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 129  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 130  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 131  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 132  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 133  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 134  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 135  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 136  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 137  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 141  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 142  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 143  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 148  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 149  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 151  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 152  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I input_rule  -i eth0 -p ipv6-icmp --icmpv6-type 153  -m limit --limit 100/s --limit-burst 10 -j ACCEPT

ip6tables -I forwarding_rule -i br-lan -p ipv6-icmp --icmpv6-type 1  -m limit --limit 100/s --limit-burst 10  -j ACCEPT
ip6tables -I forwarding_rule -i br-lan -p ipv6-icmp --icmpv6-type 2  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I forwarding_rule -i br-lan -p ipv6-icmp --icmpv6-type 3  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I forwarding_rule -i br-lan -p ipv6-icmp --icmpv6-type 4  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I forwarding_rule -i br-lan -p ipv6-icmp --icmpv6-type 128  -m limit --limit 100/s --limit-burst 10 -j ACCEPT
ip6tables -I forwarding_rule -i br-lan -p ipv6-icmp --icmpv6-type 129  -m limit --limit 100/s --limit-burst 10 -j ACCEPT

网络--接口--全局网络选项，IPv6 ULA 前缀：空
网络--接口--wan6，IPv6 分配长度：已禁用
网络--接口--lan，IPv6 分配长度：已禁用

/etc/config/dhcp 如下
config dhcp 'lan'
        option interface 'lan' #端口名称
        option start '100'    #端口号起始--结束
        option limit '150'
        option dhcpv4 'server'
        option leasetime '12h'  # infinite 租期，v4v6均有效
        option ra 'relay'
        option ndp 'relay'        #中继运行模式
        option dhcpv6 'relay'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'

config dhcp 'wan6'
        option interface 'wan6'
        option master '1'
        option ra 'relay'
        option ndp 'relay'
        option dhcpv6 'relay'

AkiraVoid, 2025-03-02:
Still happens in 24.10, neither deleting ULA prefix nor pinging WAN GUA of the router works this time. See [my post on forum](https://forum.openwrt.org/t/cannot-access-internet-by-using-ipv6-with-openwrt-as-a-2nd-level-router-and-ipv6-relay-mode/226380) for my config.

gergelypolonkai, 2025-03-30:
Also doesn’t work on 24.10 using a brand new OpenWRT One. Since i need IPv6 for my work, my new router is just collecting dust in the corner.

The default settings matched the ones on the [troubleshooting](https://openwrt.org/docs/guide-user/network/ipv6/troubleshooting) page except the prefix length delegated to the clients (my ISP delegates a /56; OpenWRT got a /60 from my ISP router and it wanted to delegate the whole /60 by default).

Disabling ULA assignment didn’t help. Changing the delegated prefix from /60 to /64 didn’t help, either. I can ping and generally access the WAN interface of the router, but nothing beyond that (including the ISP router’s addresses, both LAN and WAN, WAN being on a completely different subnet).

`/etc/config/network`:
```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'
config globals 'globals'
config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth1'
config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '64'
config interface 'wan'
	option device 'eth0'
	option proto 'dhcp'
config interface 'wan6'
	option device 'eth0'
	option proto 'dhcpv6'
```

`/etc/config/dhcp`:
```
config dnsmasq
	option domainneeded '1'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option cachesize '1000'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option localservice '1'
	option ednspacket_max '1232'
config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option dhcpv4 'server'
	option dhcpv6 'server'
	option ra 'server'
	list ra_flags 'managed-config'
	list ra_flags 'other-config'
	option ndp 'relay'
config dhcp 'wan'
	option interface 'wan'
	option ignore '1'
config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'
config host
	option name 'machina'
	list mac '24:4b:fe:54:dd:53'
	option ip '192.168.1.2'
	option leasetime 'infinite'
	option duid '00043ca1f3a3783b1bc9584e2e06c2643597'
```

gergelypolonkai, 2025-03-30:
Running tcpdump on the pinged host and the router i get strange results. The ICMPv6 echo request and reply can be seen on the target host. On the router, though:

```
12:21:22.251573 [WRT WAN MAC] > [ISP router MAC], ethertype IPv6 (0x86dd), length 118: [client addr] > [target addr]: ICMP6, echo request, id 10, seq 2, length 64
12:21:22.251815 [ISP router MAC] > [WRT WAN MAC], ethertype IPv6 (0x86dd), length 118: [ISP router LAN addr] > [client addr]: ICMP6, destination unreachable, unknown unreach code (5)
```

hydrz, 2025-08-15:
> Still happens in 24.10, neither deleting ULA prefix nor pinging WAN GUA of the router works this time. See [my post on forum](https://forum.openwrt.org/t/cannot-access-internet-by-using-ipv6-with-openwrt-as-a-2nd-level-router-and-ipv6-relay-mode/226380) for my config.

I also encountered the same problem; it occurred when using 24.10 + Docker. It was resolved by the following command.
```
ip6tables -I DOCKER-USER 1 -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
ip6tables -I DOCKER-USER 2 -j ACCEPT
```