openwrt/openwrt #9887: Netgear XR500 / R7800 SLOW AC WiFi

Black6spdZ, 2022-05-14:
no matter the settings max AC throughput I can get is ~43MB/s "both 80 and 160Mhz, multiple channels etc".. with latest dd-wrt I get DOUBLE the throughput! "88MB/s@160Mhz" Exact same with two different laptops "one with AX200 and other with 9260" Both 21.03-rc1 and latest snapshot.  Something is SERIOUSLY wrong here

Black6spdZ, 2022-05-15:
packet steering is DISABLED by default and NEEDS to be enabled to get full speed.. this should be fixed in default config

pkgadd, 2022-05-16:
packet steering helps some devices, but significantly regresses performance on others. It needs to be tested for the individual use case and cannot be enabled by default.

Black6spdZ, 2022-05-20:
ok so it was not the packet steering that was the issue, latest SNAPSHOT even with it enabled is exhibiting the ~50MB/s max wifi throughput again. Using 160Mhz on Ch 100.. how do you know if it is a DFS/radar issue?

Ansuel, 2022-06-04:
@pkgadd considering we have some scripts that fix the frequency scale to a value... wonder if a good idea would be adding a switch-case and start enabling packet steering on device that would benefits from it.

pkgadd, 2022-06-04:
Personally I've only seen negligible speed differences with packet steering on g10/ ipq8064 and nbg6817/ ipq8065, but if those devices (at least those where it does no harm) can be positively identified, why not (well, there's something said for consistency, but…).