openwrt/openwrt #9889: OpenSSH 9.0 and scp

PolynomialDivision, 2022-05-15:
As written in the arch wiki:
> Note: Since OpenSSH 9.0 the scp utility uses the SFTP protocol by default. The -O option must be used to use the legacy SCP protocol.
> https://wiki.archlinux.org/title/SCP_and_SFTP#Secure_copy_protocol_(SCP)

A normal scp will result in:
```
ash: /usr/libexec/sftp-server: not found
scp: Connection closed
```

pkgadd, 2022-05-15:
Considering that you're filing a bug, what change would you envision?
- Changing the default for OpenSSH can't be it, as this would be the wrong place to file that bug.
- Adding sftp (dæmon) support to OpenWrt can't be it either, as that would grow the image size by just shy of one Megabyte, which isn't going to happen (for that reason) either.

mrkiko, 2022-05-15:
Naive question - why 1 MB ?
I usually can enable sftp simply installing openssh-sftp-server but it does not seem to occupy 1 MB.

pkgadd, 2022-05-15:
openssh-sftp-server_9.0p1-1_arm_cortex-a15_neon-vfpv4.ipk | 54.7 KB
- `Depends: libc, libopenssl1.1, zlib`
  - libopenssl1.1_1.1.1n-2_arm_cortex-a15_neon-vfpv4.ipk | 1067.8 KB
    - `Depends: libc`
  - zlib_1.2.12-2_arm_cortex-a15_neon-vfpv4.ipk | 38.0 KB
    - `Depends: libc`

```
    54.7 KB
+ 1067.8 KB
+   38.0 KB
===========
  1160.5 KB
```

bye, bye 8 MB flash devices.

PolynomialDivision, 2022-05-16:
@pkgadd I am not sure what to do. I wanted to make the issue "googleable". It took me some time until I found out what's wrong.

pkgadd, 2022-05-16:
The problem just is, it's not a bug in OpenWrt at all (arguably it is a bug in your OpenSSH v9.0 client, to silently move from scp to sftp) - and therefore not really actionable. Yes, enabling sftpd support in OpenWrt would technically be possible, but +1.1 MB (maybe +500-600 KB if everything else would switch from wolfssl to openssl as well) just isn't an option, at all.

txt-file, 2022-05-22:
Image sizes of an `make defconfig` (ath79-generic) build:
```
5'702'449 Byte without openssh-sftp-server
5'767'985 Byte with patched openssh-sftp-server (without openssl)
6'554'417 Byte with current openssh-sftp-server (with openssl)
```

Current openssh-sftp-server adds 851'968 Byte compared to image without sftp-server. Patched openssh-sftp-server without openssl adds 65'536 Byte compared to image without sftp-server. As this is exactly 2^16 I expact some padding is involved.

To get rid of openssh I changed https://git.openwrt.org/?p=feed/packages.git;a=blob;f=net/openssh/Makefile;h=15fef1bc85bf50b3fa98a9b6823ee77c20c16081;hb=HEAD#l34 from `DEPENDS:=+libopenssl +zlib` to `DEPENDS:= +zlib` and https://git.openwrt.org/?p=feed/packages.git;a=blob;f=net/openssh/Makefile;h=15fef1bc85bf50b3fa98a9b6823ee77c20c16081;hb=HEAD#l175 from `--with$(if $(CONFIG_OPENSSL_ENGINE),,out)-ssl-engine` to `--without-openssl`.

Sadly I found no option to disable zlib.

This is not runtime tested. I did not try to find more space savings.

So I think a `openssh-sftp-server-noOpenssl` package might solve this problem.

mrkiko, 2022-10-11:
>          o  Depends: libc

Oh, sorry.I overlooked the openssl. And btw I would like to keep 8mb devices around as long as possible, so ... you're right.
Thanks!

jow-, 2022-10-11:
The `-O` flag is needed for the foreseeable future, there won't be an SFTP server by default.

closed 2022-10-11 by jow-

luckman212, 2023-02-19:
Does anyone know of a way (other than a shell wrapper) to make `scp` use the `-O` flag by default? Something we can set in `~/.ssh/config` perhaps?