openwrt/openwrt #9894: [22.03] Not booting with LS1021A-IOT

GMunari, 2022-05-16:
Dear Sirs,
I tried 22.03 version image with LS1021A-IOT board and it does not work.

It seems to be an issue related to squash format which is not recognized by the board.

In fact, the 19.04 version which is still in ext4 format works perfectly.

ArtelMike, 2022-10-21:
Our embedded board is based on the LS1021A-IOT and I have been upgrading it to use OpenWrt. Version 19.07 which works just fine. When building an image from 21 or 22 it will not generate an ext4 image, even when I ask it to and the squashfs image won't boot. It seems u-boot is trying to boot ext4 even though it is squashfs.

Any help or pointers as to what might need to be changed to generate an ext4 image would be helpful. NXP seems to have abandoned development as of 2018. I would much rather spend time upgrading to a supported version and not get stuck on 19 which is now EOL.

ynezz, 2022-10-22:
@yangbolu1991 @CHKDSK88 any idea?

ArtelMike, 2022-10-22:
I looked at the console and found the following:
1st problem is the u-boot environment bootargs are out of date. It is:

fdtaddr=0x8f000000
loadaddr=0x81000000
fdt_high=0x8fffffff
initrd_high=0xffffffff
sd_boot=ext4load mmc 0:1 $loadaddr fitImage;bootm $loadaddr
bootargs=root=/dev/mmcblk0p2 rw rootwait rootfstype=squashfs,f2fs noinitrd earlycon=uart8250,mmio,0x21c0500 console=ttyS0,115200
bootcmd=echo starting openwrt ...;run sd_boot
bootdelay=3
fsl_bootcmd_mcinitcmd_set=y

NOTE that ext4load is no longer valid. It should be ext2load.

2nd problem. When running the boot manually using ext2load, it seems to load the image OK but bootm says the image is invalid and stops.

The boot image, fitImage, is on the partition.

Regards,
Mike

ArtelMike, 2022-10-22:
I forgot to mention that I am pretty well versed in quilt for this BSP and can create and test the patches, then submit them to you for review and commit. All I Need is some guidance as to what needs to be changed.

Mike

CHKDSK88, 2022-10-23:
Any chance for bootlog for bootlog?

ArtelMike, 2022-10-24:
Here is a console log. I added a printenv and a manual boot try at the end.
If there is any other info you want, please let me know.

[putty.log](https://github.com/openwrt/openwrt/files/9851254/putty.log)

CHKDSK88, 2022-10-24:
@ArtelMike 

Please try build image with this patch:

https://github.com/CHKDSK88/openwrt-1/commit/a834761bb830275bddf18f406f0411ea43a03683

ArtelMike, 2022-10-24:
Sorry, no-go. It doesn't get into the u-boot second stage.

[putty.log](https://github.com/openwrt/openwrt/files/9853613/putty.log)

Mike

CHKDSK88, 2022-10-24:
@ArtelMike 

Yeah, Image was too big. Please try one more time:
https://github.com/CHKDSK88/openwrt-1/commit/c8647b7bd3e32bae0f27b594fb934e0934cfc6f6

ArtelMike, 2022-10-24:
It is never that easy...
New failure. ** No partition table - mmc 0 **

[putty.log](https://github.com/openwrt/openwrt/files/9855587/putty.log)

Thank you for being so responsive,
Mike

CHKDSK88, 2022-10-25:
@ArtelMike 

Please try one more time:
https://github.com/CHKDSK88/openwrt-1/commit/00e7972bf06f3f8ddadd2d040f497861f6f8dab3

ArtelMike, 2022-10-25:
Not quite there yet. I changed the boot environment settings in package/boot/uboot-layerscape/files/fsl_ls1021a-iot-sdboot-uEnv.txt to match fsl_ls1088a-rdb-sdboot-uEnv.txt, which looks correct.

[fsl_ls1021a-iot-sdboot-uEnv.txt](https://github.com/openwrt/openwrt/files/9859865/fsl_ls1021a-iot-sdboot-uEnv.txt)

But now it fails with: Unknown command 'fsl_mc' - try 'help'

[putty.log](https://github.com/openwrt/openwrt/files/9859872/putty.log)

I am sure that the boot environment needs to be changed. I just don't know the exact syntax needed or if another config option is required.

Your thoughts?

Mike

CHKDSK88, 2022-10-26:
Ok. But what was happen without  env swap?

ArtelMike, 2022-10-26:
** No partition table - mmc 0 ** when doing ext4load.

Mike

ArtelMike, 2022-10-26:
Also, I can mount and display the partitions on my Linux PC.

CHKDSK88, 2022-10-26:
I'm AFK. I go back to it tommrow evening.

CHKDSK88, 2022-10-27:
@ArtelMike 

https://github.com/CHKDSK88/openwrt-1/commit/c045ac90d152cc2740fa21f942f6ebb82b254162

Now it should read partition table.

ArtelMike, 2022-10-27:
YES! This now works with one environment change. The sd_boot line needs a change to the variable $loadaddr adding brackets around it like ${loadaddr} for both the ext4load and bootm commands. I'll edit the env file tomorrow morning and build a new image.

Now the question I have is it better to increase the size of the uboot sector or decrease the size of uboot as you have done?

Thank you for the quick resolution.

Best regards,
Mike

CHKDSK88, 2022-10-28:
@ArtelMike 

Lets try enlarge u-boot:
https://github.com/CHKDSK88/openwrt-1/commit/09e7633f43156836de5ce0c563784db18afd5db1

ArtelMike, 2022-10-28:
Works great, along with this change to the environment, it produces bootable images.

[fsl_ls1021a-iot-sdboot-uEnv.txt](https://github.com/ArtelMike/openwrt-1/commit/291ca044c5b121c4985bd9c7a5a8e34732167285#diff-e734958d69caef8fa134fc07350633403e15bccda0633a2d90bf4dce67e9541e)

Thank you again,
Mike

CHKDSK88, 2022-10-28:
@ArtelMike 

Thanks for feedback. I will prepare PR at evening.

Is `${loadaddr}` really required? On LS1046A-RDB works `$loadaddr`...

What error is without it?

ArtelMike, 2022-10-28:
Without the brackets it gives a syntax error and the "Usage" message text for both ext4load and bootm.
I would not be surprised if a lot of them are wrong! One that looks like it was corrected is fsl_ls1046a-frwy-sdboot-uEnv.txt.

CHKDSK88, 2022-10-28:
Strange. For example my LS1046A-RDB with newest OpenWrt U-boot:

```
NOTICE:  8 GB DDR4, 64-bit, CL=15, ECC on, CS0+CS1
NOTICE:  BL2: v2.4(release):reboot-20996-g7dbafa95b51e
NOTICE:  BL2: Built : 07:54:18, Oct 20 2022
NOTICE:  BL2: Booting BL31
NOTICE:  BL31: v2.4(release):reboot-20996-g7dbafa95b51e
NOTICE:  BL31: Built : 07:54:18, Oct 20 2022
NOTICE:  Welcome to ls1046ardb BL31 Phase

U-Boot 2021.04-OpenWrt-r19508+1488-0d2d52df69e9 (Oct 20 2022 - 07:54:18 +0000)

SoC:  LS1046AE Rev1.0 (0x87070010)
Clock Configuration:
       CPU0(A72):1800 MHz  CPU1(A72):1800 MHz  CPU2(A72):1800 MHz
       CPU3(A72):1800 MHz
       Bus:      600  MHz  DDR:      2100 MT/s  FMAN:     700  MHz
Reset Configuration Word (RCW):
       00000000: 0c150012 0e000000 00000000 00000000
       00000010: 11335559 40005012 60040000 c1000000
       00000020: 00000000 00000000 00000000 00238800
       00000030: 20124000 00003101 00000096 00000001
Model: LS1046A RDB Board
Board: LS1046ARDB, boot from SD
CPLD:  V2.3
PCBA:  V2.0
SERDES Reference Clocks:
SD1_CLK1 = 156.25MHZ, SD1_CLK2 = 100.00MHZ
DRAM:  7.9 GiB (DDR4, 64-bit, CL=15, ECC on)
       DDR Chip-Select Interleaving Mode: CS0+CS1
Using SERDES1 Protocol: 4403 (0x1133)
Using SERDES2 Protocol: 21849 (0x5559)
PCIe1: pcie@3400000 Root Complex: no link
PCIe2: pcie@3500000 Root Complex: no link
PCIe3: pcie@3600000 Root Complex: no link
NAND:  512 MiB
MMC:   FSL_SDHC: 0
Loading Environment from MMC... OK
EEPROM: NXID v1
In:    serial
Out:   serial
Err:   serial
SEC0:  RNG instantiated
Net:
MMC read: dev # 0, block # 18432, count 128 ...
Fman1: Uploading microcode version 106.4.18
eth0: fm1-mac3, eth1: fm1-mac4, eth2: fm1-mac5, eth3: fm1-mac6, eth4: fm1-mac9, eth5: fm1-mac10
Hit any key to stop autoboot:  0
=> printenv
bootargs=root=/dev/mmcblk0p2 rw rootwait rootfstype=squashfs,f2fs noinitrd earlycon=uart8250,mmio,0x21c0500 console=ttyS0,115200
bootcmd=echo starting openwrt ...;run sd_boot
bootdelay=3
eth1addr=00:04:9F:06:53:7B
eth2addr=00:04:9F:06:53:76
eth3addr=00:04:9F:06:53:77
eth4addr=00:04:9F:06:53:79
eth5addr=00:04:9F:06:53:78
ethaddr=00:04:9F:06:53:7A
fdt_high=0x8fffffff
fdtaddr=0x8f000000
fdtcontroladdr=fbc1a4f0
fman_ucode=fbc2e7d0
fsl_bootcmd_mcinitcmd_set=y
hwconfig=fsl_ddr:bank_intlv=auto
initrd_high=0xffffffffffffffff
loadaddr=0x81000000
sd_boot=ext4load mmc 0:1 $loadaddr fitImage;bootm $loadaddr
stderr=serial
stdin=serial
stdout=serial

Environment size: 645/8188 bytes
=> boot
starting openwrt ...
7328780 bytes read in 314 ms (22.3 MiB/s)
## Loading kernel from FIT Image at 81000000 ...
   Using 'config-1' configuration
   Trying 'kernel-1' kernel subimage
     Description:  ARM64 OpenWrt Linux-5.15.74
     Type:         Kernel Image
     Compression:  gzip compressed
     Data Start:   0x810000e8
     Data Size:    7300195 Bytes = 7 MiB
     Architecture: AArch64
     OS:           Linux
     Load Address: 0x80000000
     Entry Point:  0x80000000
     Hash algo:    crc32
     Hash value:   32309130
     Hash algo:    sha1
     Hash value:   2c8939017a68f44a28eafb3e9227c0279708a985
   Verifying Hash Integrity ... crc32+ sha1+ OK
## Loading fdt from FIT Image at 81000000 ...
   Using 'config-1' configuration
   Trying 'fdt-1' fdt subimage
     Description:  ARM64 OpenWrt fsl_ls1046a-rdb-sdboot device tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x816f6694
     Data Size:    26665 Bytes = 26 KiB
     Architecture: AArch64
     Hash algo:    crc32
     Hash value:   93f97137
     Hash algo:    sha1
     Hash value:   08867076abdbb0766f09336ddd62d0bbcd2144a1
   Verifying Hash Integrity ... crc32+ sha1+ OK
   Booting using the fdt blob at 0x816f6694
   Uncompressing Kernel Image
   Loading Device Tree to 000000008ffe6000, end 000000008ffff828 ... OK
WARNING failed to get smmu node: FDT_ERR_NOTFOUND
WARNING failed to get smmu node: FDT_ERR_NOTFOUND

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd082]
[    0.000000] Linux version 5.15.74 (krolik@WBEC678) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r19508+1481-0d2d52df69e9) 11.3.0, GNU ld (GNU Binutils) 2.37) #0 SMP PREEMPT Thu Oct 20 07:54:18 2022
[    0.000000] Machine model: LS1046A RDB Board
```

CHKDSK88, 2022-10-28:
Ok, I got it. Brackets was required for this board.

Here are I deleted it: https://github.com/openwrt/openwrt/commit/a254279a6c3048feb83909649c41ee4a7d37ec93

I don't know why for this particular board brackets are required.

Please make last test and I send PR:

https://github.com/CHKDSK88/openwrt-1/commit/162d10bae7bd77e9d384aa06644710d2ab9f8e60

ArtelMike, 2022-10-28:
Looks good to me.
Again, thank you for the prompt resolution.
Mike

CHKDSK88, 2022-10-28:
@ArtelMike

I created PR. After acceptation, I will prepare backport PRs.

commit d75ed372 (2022-11-05): uboot-layerscape: adjust LS1012A-IOT config and env

In a254279a6c30 LS1012A-IOT kernel image was switched to FIT.

But u-boot config is lack of FIT and ext4 support.

This patch enables it.

It also fix envs, because for some reason this board need to use "loadaddr"
variable in brackets.

Fixes: #9894
Fixes: a254279a6c30 ("layerscape: Change to combined rootfs on sd images")
commit e313cd8d (2022-11-05): uboot-layerscape: adjust LS1012A-IOT config and env

In a254279a6c30 LS1012A-IOT kernel image was switched to FIT.

But u-boot config is lack of FIT and ext4 support.

This patch enables it.

It also fix envs, because for some reason this board need to use "loadaddr"
variable in brackets.

Fixes: #9894
Fixes: a254279a6c30 ("layerscape: Change to combined rootfs on sd images")

(cherry picked from commit d75ed3726d994fd050730e9ab5923d6232913054)
commit 8f459812 (2022-11-12): uboot-layerscape: adjust LS1012A-IOT config and env

In a254279a6c30 LS1012A-IOT kernel image was switched to FIT.

But u-boot config is lack of FIT and ext4 support.

This patch enables it.

It also fix envs, because for some reason this board need to use "loadaddr"
variable in brackets.

Fixes: #9894
Fixes: a254279a6c30 ("layerscape: Change to combined rootfs on sd images")

(cherry picked from commit d75ed3726d994fd050730e9ab5923d6232913054)
closed 2022-11-05 by openwrt-bot