openwrt/openwrt #9901: Tag 22.03.0-rc2

aparcar, 2022-05-17:
Some more pushes happened to the `openwrt-22.03` branch, let's have another tag.

I'll try to write the meeting notes within the next two days and tag everything.

jow-, 2022-05-17:
At the very least we should bump fw4 before tagging, Git HEAd contains various fixes for issues reported in rc1 but I didn't bump the package yet since I was waiting for more feedback.

I also need to cherry-pick a number of LuCI fixes yet.

edrikk, 2022-05-18:
Just for awareness; I posted at the below PR, but roughly around this kernel bump 22.03 (and master) have started to softbrick the Xiaomi Redmi AC2100 (the Netgear R6700v2 of the same architecture is ok however).

https://github.com/openwrt/openwrt/pull/9789#issuecomment-1128362236

pepe2k, 2022-05-18:
@edrikk please file an issue so this problem can be properly tracked.

edrikk, 2022-05-18:
Thank you.  Done.  I:

- Have just created [this issue](https://github.com/openwrt/openwrt/issues/9905) for the Xiaomi AC2100 softbricking post 22.03rc1
- Had previously also created [this issue](https://github.com/openwrt/luci/issues/5678) under the Luci branch for upnp information not showing with the main status page or within the upnp app page.

aparcar, 2022-05-21:
@jow- did you do the backporting?

aparcar, 2022-05-21:
Tagged for now, the backports are included. The mentioned issues should become part of rc3

closed 2022-05-21 by aparcar