openwrt/openwrt #9905: [22.03- pre-rc2] softbricks Xiaomi Redmi AC2100

edrikk, 2022-05-18:
Hi,

I put this comment [within this PR](https://github.com/openwrt/openwrt/pull/9789#issuecomment-1128362236) as although potentially not directly due to this Kernel bump, at this point I have observed that flashing the Xiamomi Redmi AC2100 results in a non-booting device, requiring recovery (using Xiaomi's recovery tool, following by process to get back to older openwrt version).

I can confirm that a few commits post versions post 22.03-rc1 (which was .111 kernel) worked as well.

Of note is that the Netgear R6700, which is based on the same chipset and compile structure works with all versions of kernel including the above noted .114, .115, as well as current .116.   

Unfortunately, this router does not have easily connectable UART, so I couldn't/can't pull the bootlog.   Posting this for awareness and in case anyone else with this router (and with soldering skills for the UART) can provide further information.

timocapa, 2022-05-18:
Not a help for your issue, but

> requiring recovery (using Xiaomi's recovery tool)

R3G's `breed` bootloader works on the AC2100.

castiel652, 2022-05-21:
ran in to this issue as well

castiel652, 2022-05-28:
@edrikk can you try revert https://github.com/openwrt/openwrt/commit/ec45e1ff68d198aded784ad7712484c1474ede60 and see if it boots?

rsalvaterra, 2022-05-29:
@castiel652, great call! Will someone please revert 0bc794a66845738eef7eeb7e13877ffb8aec17f7 from master? It definitely breaks the Redmi AC2100.

edrikk, 2022-05-30:
Confirmed.
Reverting this patch solves the issue.

edrikk, 2022-06-15:
Closing as this has been resolved through:

[Master Commit](https://github.com/openwrt/openwrt/commit/b047ca1aa0e25f671cf7c927a8db0c2b413202a0)
[22.03 Commit](https://github.com/openwrt/openwrt/commit/a83dc6b06dc424605040f0b0ee8152a8ef164a19)

closed 2022-06-15 by edrikk