openwrt/openwrt #9913: Netgear Nighthawk X4S R7800: rx ring became corrupted: -5

enunes, 2022-05-20:
Name the tree/revision/version: OpenWrt 21.02.3 r16554-1d4dea6d4f / LuCI openwrt-21.02 branch git-22.083.69138-0a0ce2a

Name the affected device: Netgear Nighthawk X4S R7800

Once every few days, wifi stops working. I don't have an exact time frame but I have it twice in the past month or so. It seems to happen at a random time with nothing different in particular that triggers it. This device has multiple SSIDs including 5 GHz and 2.4 GHz. The one that affects me the most is 2.4 GHz which I have for embedded/IoT devices so some automation stops working.

This is reported in dmesg:
```
[18961.240376] ath10k_pci 0001:01:00.0: htt tx: fixing invalid VHT TX rate code 0xff
[23036.658429] ath10k_pci 0000:01:00.0: Invalid VHT mcs 15 peer stats
[54216.793970] ath10k_pci 0001:01:00.0: wmi: fixing invalid VHT TX rate code 0xff
[2274617.218030] ath10k_pci 0001:01:00.0: rx ring became corrupted: -5
```
The first 3 messages appear to be harmless, the 4th one shows up when the problem is triggered.
I see that there were already earlier occurrences of this problem as in https://forum.openwrt.org/t/build-for-netgear-r7800/316/167 but couldn't find any report for recent releases.

I don't have an easy reproducer for it.

The only relevant information I can think of is that I have used the same device with the same configuration, same setup for a long time with 19.07.x and recall never seeing that before. So that might be a regression in some way.

pepe2k, 2022-05-23:
@enunes could you please check if this problem appears also in `22.02-rc2` (https://downloads.openwrt.org/releases/22.03.0-rc2/targets/ipq806x/generic/) and/or `master` (https://downloads.openwrt.org/snapshots/targets/ipq806x/generic/)?

Cheers,
Piotr

quarkysg, 2022-05-23:
@enunes Would you be interested to try OpenWrt 21.02.1?  I would be interested to know if this release is working for you.  There was a change in the WiFi airtime scheduler between 21.02.1 and 21.02.2,  Of course there were many other changes between both revisions, but for my R7800, I started seeing issues with WiFi around Nov 2021 (I do my own custom builds) and that is when the new scheduler was introduced to the 21.02 tree.  My suspision is that the ath10k driver does not play nice with the new airtime scheduler.

enunes, 2022-05-26:
Hi all, thanks for the feedback.
I could try another firmware on it, but since this network contain services that are on the internet and used by upstream projects, I should give them a notice, disable it, set up a maintenance time. So it is complicated to try a different firmware every other day.
The problem did happen with an earlier 21.02.x before I updated it, but I'm not sure if that was 21.02.1 or 21.02.2.
I can schedule that maintenance and switch the firmware, but then considering it may take several weeks to get some useful conclusion, which version do you think it is more useful to try first now; double check 21.02.1 or 22.02-rc?

quarkysg, 2022-05-26:
@enunes my bet would be on either 21.02.1 or on the lastest master snapshot built, as there are folks in the forums stating that the recent update to the qca9984 firmware seems to resolve the WiFi drop issues.  It could be that the new firmware may have fix any incompatibility between mac80211 and ath10k.

I may be wrong, but so far, I do not see any changes in 22.02 relating to ath10k that may help resolve this issue.

enunes, 2022-05-27:
Ok, I'll flash 21.02.1 at the next maintenance and monitor it for issues. It may take a while for it to become useful evidence but I'll report back when I have more information. Thank you.

enunes, 2022-07-19:
> What about testing out the RC5?

It is one option, but I would like to avoid using this as a test network. It is very inconvenient if it hits the bug at a random time again and especially if that wouldn't give much relevant information to fix it.

I would be willing to give it a try if there is some clue that the issue may be fixed there.
If someone is more familiar with the development history for this target, is there some particular fix or rework from 21.02 to 22.02-rc that may have fixed this?

aism, 2023-08-09:
I have the same issue on the same box, also with 2.4GHz.

**Symptoms**:
Once this issue is hit the SSID starts being advertised very intermittently, and once I try join it from any of my clients it fails to join with nothing in the OpenWRT's system log about that client's attempt.
From LuCi the radio in question is up and the SSID's being advertised.
Trying to restart the SSID makes the box unresponsive (LuCi's landing is accessible and displays `[LuCI - Lua Configuration Interface]`, but doesn't redirect to UI anymore; SSH cert-based auth stops working with `Permission denied (publickey)`).
However, 5GHz SSID keeps working.
Had to hard-reset the box to get back the mgmt. This also brings back 2.4GHz SSID.

**Versions**:
The problem was on `v22.03.5`. Just upgraded to `23.05.0-rc2` and hit it again almost immediately (uptime is less than few hours).

enunes, 2025-01-07:
Thanks for checking more on this. I have had that same device for years now on 19.07.x as a "dumb AP" and never saw the issue again. If you have a reliable reproducer from a clean config, I guess one approach could be to bisect back to some change even from 19.07.x -> 21.02.x (and then kernel bisect?) for a hint of what initially started triggering it. Unfortunately I really don't have the time capacity to do that.