openwrt/openwrt #9920: firewall is broken with flow offloading ON

xosevp, 2022-05-21:
running latest snapshots r19689-19ef3b54f4 on TP-Link Archer C7 v5

```
root@ap2:/etc/config# /etc/init.d/firewall start
Section @rule[0] (Allow-DHCP-Renew) is disabled, ignoring section
Section @rule[1] (Allow-DHCPv6) is disabled, ignoring section
Section @rule[2] (Allow-Ping) is disabled, ignoring section
Section @rule[3] (Allow-ICMPv6-Input) is disabled, ignoring section
Section @rule[4] (Allow-ICMPv6-Forward) is disabled, ignoring section
Section @rule[5] (Support-UDP-Traceroute) is disabled, ignoring section
Section @rule[6] (Allow-ISAKMP) is disabled, ignoring section
Section @rule[7] (Allow-IGMP) is disabled, ignoring section
Section @rule[8] (Allow-MLD) is disabled, ignoring section
Section @rule[9] (Allow-IPSec-ESP) is disabled, ignoring section
/proc/self/fd/0:9:12-13: Error: Could not process rule: Not supported

/proc/self/fd/0:58:29-44: Error: Could not process rule: No such file or directory
```
I have to comment this two in /etc/config/firewall, to avoid this bug:

```
#       option flow_offloading '1'
#       option flow_offloading_hw '1'
```

Thanks.

xosevp, 2022-05-21:
@jow-
related to 2df17604a4f891447beb66988e3d83e23ab3a3b0

dave14305, 2022-05-21:
There are conversations happening upstream about these behaviors.

https://lore.kernel.org/netfilter-devel/20220510202739.67068-1-nbd@nbd.name/

xosevp, 2022-06-02:
with latest snapshots r19726-b21b98627d (Thu Jun 2 22:45:17 2022) on TP-Link Archer C7 v5

Now firewall works, but it says:
`Hardware flow offloading unavailable, falling back to software offloading`

TP-Link Archer C7 v5 used to have "Hardware flow offloading" in previous versions.

And I get ~27 MB/s downloading, when before was ~70MB/s

Thanks.

taylorkline, 2024-03-26:
@xosevp Was this resolved for you?