openwrt/openwrt #9921: [Feature Request] wpad-full for high rom devices

howl, 2022-05-21:
This is a feature request commit for what I have seem in github automation commit from @robimarko for ipq807x. Devices that have so many space could have with no problems the full version bringing the out of the box ability to support for example mesh networks. This could be interesting as this way is possible to use the "official" images for devices that are connected only by wireless method and using not supported modes to be updated without making them to connect in an alternative way to the ap providing wan access.

Also mesh mode is displayed always no matter if wpad-basic is installed, but this goes to luci issue tracker.

robimarko, 2022-05-21:
I don't think it's an issue, one can simply use opkg to install the feature full wpad on the fly or even use imagebuilder to make images with it.

The only reason why full wpad is included in my WIP for ipq807x is that there you cant use opkg to just install the full wpad

howl, 2022-05-21:
@robimarko I tagged it as feature request, I know is not an issue. In official devices it could cover the use case I mention, out of the box working update for wireless connected devices. Your commit for ipq807x just bring me the idea to ask if it could be possible. There are more platforms that exhibits so many storage space.

Just wanted to be take into account, myself appreciate a lot the do it small as possible ideas, with tl-wr1043nd is not possible now a days to have a reliable haproxy with ssl setup but it is with 19.07 that is so impressive. In this feature request I ask for high storage devices.

robimarko, 2022-05-21:
I get your point, however that would add more code and require further device categorization.

My point is that you can use imagebuilder, there is even a online one to include the packages you want or something like ASU to solve this

jow-, 2022-05-30:
Similar feature requests have been rejected in the past, this one will be similar. We aim to minimize the amount of different permutations as much as possible.

Since it is unlikely that this issue will be acted upon, I'll close it.

closed 2022-05-30 by jow-

howl, 2022-05-30:
@jow- with permutations you mean using different wpad flavours for different targets?