openwrt/openwrt #9923: fw4: Doesn't take into account time restrictions

howl, 2022-05-22:
Add a simple REJECT rule for wan zone. Set start time to for example 16:00:00 and stop 15:00:00. That should make work internet only between 15:00 and 16:00, but it works all day. Doesn't vary anything setting utc or not.

Have tested days of week and they work, haven't tested days of month.

dave14305, 2022-05-22:
Can you show your rule from `/etc/config/firewall` and the output of `fw4 reload`? Any errors?

howl, 2022-05-22:
With this one the device doesn't have internet whole day between Monday and Friday and it have internet Saturday and Sunday, so this works as it should:
```
config rule
	option name 'Rule without time'
	list proto 'all'
	option src '*'
	list src_mac 'AA:BB:CC:DD:EE:FF'
	option dest 'wan'
	option target 'REJECT'
	option weekdays 'Mon Tue Wed Thu Fri'
```

With the following rule the device has internet all the days all the time, it should only have between 20:00 and 21:00 from Monday to Friday and whole day Saturday and Sunday (doesn't matter if we set the time is UTC instead local):
```
config rule
	option name 'Rule with time'
	list proto 'all'
	option src '*'
	list src_mac 'AA:BB:CC:DD:EE:FF'
	option dest 'wan'
	option target 'REJECT'
	option weekdays 'Mon Tue Wed Thu Fri'
	option start_time '21:00:00'
	option stop_time '20:00:00'
```

A firewall reload doesn't show anything but warning about some rules that I have disabled.

With fw3 this last rule works as expected.

dave14305, 2022-05-22:
Can you post `nft list ruleset | grep time`

howl, 2022-05-22:
That only gives me two rules Allow-ICMPv6-Input and Allow-ICMPv6-Forward because the time-exceeded icmp types.

howl, 2022-05-22:
The time based rules appears like this:
ether saddr aa:bb:cc:dd:ee:ff meta hour "21:00"-"20:00" meta day { "Monday", "Tuesday", "Wednesday", "Thursday", "Friday" } counter packets 0 bytes 0 jump reject_to_wan comment "!fw4: Rule with time"

dave14305, 2022-05-22:
This scenario seems to work better when specifying a range for `meta hour` instead of two separate expressions.
```
ether saddr aa:bb:cc:dd:ee:ff meta hour "21:00:00"-"20:00:00" meta day { "Sunday", "Monday", "Tuesd
ay", "Wednesday", "Thursday", "Friday", "Saturday" } counter jump reject_to_wan comment "!fw4: Rule with time"
```
instead of:
```
ether saddr aa:bb:cc:dd:ee:ff meta hour >= "21:00:00" meta hour <= "20:00:00" meta day { "Sunday", "Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday" } counter jump reject_to_wan comment "!fw4: Rule with time"
```
results in a working rule:
```
ether saddr aa:bb:cc:dd:ee:ff meta hour != "1193042:28:16"-"1193043:28:16" meta day { "Sunday", "Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday" } counter packets 39 bytes 2554 jump reject_to_wan comment "!fw4: Rule with time"
```
I don't know what's going on with the display of the time, but it works. I just loaded the latest 22.03.0-rc2 before testing.

Maybe @jow- can re-work the way both a `start_time` and `end_time` are handled in a rule, so that it puts out a `meta time` range instead of 2 separate expressions. When the start time in the range is later than the end time, nftables will invert the logic to be `meta time != "20:00:00"-"21:00:00"`

howl, 2022-05-23:
@dave14305 just to be sure I understand it, if having one meta hour is the way to go, you mean that actually this is done right, but, the only missing thing is to transpose start stop and negate when the start hour is higher than stop? I could test that when the time is defined with start lower that stop it works as expected as you said.

I could invert the logic and make after another rule to REJECT always but it's not critical for me so could wait for @jow- to resolve. Thanks for your investigation @dave14305.

howl, 2022-05-31:
Works partially, something like:
        option start_time '16:11'
        option stop_time '16:10'
result in:
meta hour != "16:10"-"16:11"
and works.

But, something like this:
        option start_time '16:11:35'
        option stop_time '16:10:42'
result in:
meta hour "16:00:35"-"16:00:42"
and not works.

Seems that minutes when seconds are specified are not get into account, and that makes in this particular example to think start is before stop, but the problem is in fact the minutes missing.

jow-, 2022-05-31:
@howl - I can't reproduce what you describe.

```
root@er-x:~# uci show firewall.@rule[-1]
firewall.cfg1792bd=rule
firewall.cfg1792bd.dest='*'
firewall.cfg1792bd.proto='all'
firewall.cfg1792bd.dest_ip='6.6.6.6'
firewall.cfg1792bd.start_time='16:11:35'
firewall.cfg1792bd.stop_time='16:10:42'
firewall.cfg1792bd.target='accept'
root@er-x:~# fw4 print | grep 6.6.6.6
		ip daddr 6.6.6.6 meta hour "16:11:35"-"16:10:42" counter accept comment "!fw4: @rule[13]"
root@er-x:~# fw4 restart
root@er-x:~# nft list ruleset | grep 6.6.6.6
		ip daddr 6.6.6.6 meta hour != "16:10:42"-"16:11:35" counter packets 0 bytes 0 accept comment "!fw4: @rule[13]"
root@er-x:~# 
```

jow-, 2022-06-01:
Ah yes, I know why. Will look into it...

commit 7f998088 (2022-06-01): ucode: update to latest Git HEAD

d996047 syntax: adjust number literal parsing and string to number conversion
9efbe18 lib: refactor `uc_int()`

Fixes: #9923
commit dad15330 (2022-06-01): ucode: update to latest Git HEAD

d996047 syntax: adjust number literal parsing and string to number conversion
9efbe18 lib: refactor `uc_int()`
da3f089 lib: rework uc_index() implementation
559029e ci: make jobs faster during pull request testing

Fixes: #9923

(backported from commits 251e70c88725fb33960ca60202dd55a84ae6b4d7 and
 7f998088f01baf0da24a9fd20de5d5389023361a)
closed 2022-06-01 by jow-