openwrt/openwrt #9924: [22.03.0-rc1] softbricks Ubnt Picostation M2 (swx-m2b)

txt-file, 2022-05-22:
I just tried https://downloads.openwrt.org/releases/22.03.0-rc1/targets/ath79/generic/openwrt-22.03.0-rc1-ath79-generic-ubnt_picostation-m-squashfs-factory.bin (flashed via tftp) and got a PicoStation that
* didn't give me an IPv4 address
* SSH connection errored with
```
user@host:~/Downloads$ ssh root@192.168.1.1
kex_exchange_identification: read: Connection reset by peer
Connection reset by 192.168.1.1 port 22
user@host:~/Downloads$ dbclient root@192.168.1.1

dbclient: Connection to root@192.168.1.1:22 exited: Remote closed the connection
user@host:~/Downloads$
```
* luci errored that main is corrupt (sorry, but didn't save the error message)

I expect that most other 32 MiB RAM devices are broken with 22.03. So might be better to disable building images for them.

sbkg0002, 2022-06-12:
Where you able to fix your issue ?

txt-file, 2022-08-08:
Did not try again.