openwrt/openwrt #9925: Netgear R7800: sysupgrade does not restore all previously installed packages, eg. openvpn and ocserv

mmokrejs, 2022-05-22:
Hi,
  with every sysupgrade ( https://downloads.openwrt.org/releases/21.02.3/targets/ipq806x/generic/openwrt-21.02.3-ipq806x-generic-netgear_r7800-squashfs-sysupgrade.bin ) I loose packages installed and especially OpenVPN and OpenConnect packages are not installed after the update automatically. I upgraded from 21.02 now but happened also when migrating from 17.x.

  I think also some kernel modules are not re-introduced. I ticked also the third option about restoring installed_packages.txt although I do not understand why isn't that a default.

  Also, the below screenshot seems scary. The error messages do not make sense to me. Is this a new feature so taht the certificates to be used for verification were not yet on my system and hence why the upgrade complained that the new image cannot be verified? Then do not issue a false alarm. But there are some packahes in the reository with certificates so should I install the certificates beforehand? Which package exactly?

OpenWrt 21.02.3 r16554-1d4dea6d4f / LuCI openwrt-21.02 branch git-22.119.37126-a993714

[image]

castiel652, 2022-05-23:
That's normal behavior because official images only contain bare minimum

StSaens, 2022-05-23:
https://sysupgrade.openwrt.org/
That's what you're looking for.

But, at the moment it does not (afaik) support upgrading from 21.x -> 22.x

closed 2022-05-23 by pepe2k

mmokrejs, 2022-10-07:
So I installed luci-app-attendedsysupgrade and tried it. Aside from the fact that the webpage kept running the javascript instead of redirecting me to the newly flashed router it probably went fine. I just lost patience after 7 minutes and opened another TAB in my web browser to visit my router's IP address. It showed the uptime 7 mins and the new release version being installed. So, the reload of the page does not really work properly, at least in current Firefox.

[image]

mmokrejs, 2022-10-09:
Doh! Of course it did not preserve files housed in /overlay/upper/etc/ referenced from /etc/rc.local. Therefore, my luks key was lost while the startup scripts was preserved so the service created a new, empty shared filesystem export. I also had to manually re-install at least `openssh-sftp-client openssh-sftp-server sshfs vsftpd-tls` which did not get re-installed after the "attended-upgrade". After copying there back the missing LUKS key I could manually shutdown the wrong syncthing instance and start it up by the preserved /etc/rc.local script. Luckily I had the crypto key elsewhere. Doesn't seem `luci-app-attendedsysupgrade` did what I expected it to do: it has lost some packages installed previously and has lost also files referenced from /etc/rc.local, at least.