openwrt/openwrt #9927: firewall4 typo

notgood, 2022-05-22:
Latest 22.03 tree, firewall4 fails with a cryptic message:
`/proc/self/fd/0:475:59-64: Error: syntax error, unexpected string`

Had to dump "fw4 print" to file and execute it with nft to see the real error.
```
meta nfproto ipv4 iifname { "br-lan" } ip dsaddr @my_subnets counter meta mark set 0x1 comment "!fw4: Mark forwarded packets destined to my subnets"
		                          ^^^^^^
```
Typo is a typo, but fw4 could probably be more verbose when failing, print the real nft error to user and logs.

notgood, 2022-05-22:
Relevant rule from /etc/config/firewall
```
config rule
	option name 'Mark forwarded packets destined to my subnets'
	option family 'ipv4'
	option proto 'all'
	option src 'lan'
	option dest '*'
	option ipset 'my_subnets'
	option set_mark '0x1'
	option target 'MARK'

```

dave14305, 2022-05-22:
Seems like it comes from:
https://git.openwrt.org/?p=project/firewall4.git;a=blob;f=root/usr/share/ucode/fw4.uc;h=cfef69c14c98c8c7f579c851a22407d60bc1358d;hb=HEAD#l341
Where the 's' after ${dir} might need to be removed. Try to edit `/usr/share/ucode/fw4.uc` and see if it fixes it.
```
	for (let i, t in set.types) {
		let dir = ((match.dir?.[i] || set.directions[i] || 'src') == 'src' ? 's' : 'd');

		switch (t) {
		case 'ipv4_addr':
			fields[i] = `ip ${dir}saddr`;
			break;

		case 'ipv6_addr':
			fields[i] = `ip6 ${dir}saddr`;
			break;
```

notgood, 2022-05-22:
> Where the 's' after ${dir} might need to be removed.

Thank you, that fixes the problem!

pepe2k, 2022-05-23:
@jow- ping

jow-, 2022-05-31:
Please test the firewall bump commit at https://git.openwrt.org/49e6daa8

notgood, 2022-06-01:
Latest commit fixes this issue. Thank you!