openwrt/openwrt #9928: Openwrt 22.03.0-rc2 bug repositories.conf

predators46, 2022-05-22:
@hauke 

`cat repositories.conf   
                               
## Place your custom repositories here, they must match the architecture and version.  
# src/gz %n https://downloads.openwrt.org/releases/22.03-SNAPSHOT                      
# src custom file:///usr/src/openwrt/bin/armvirt/packages                                                                                                                     
## Remote package repositories                                                         
src/gz openwrt_core https://downloads.openwrt.org/releases/22.03-SNAPSHOT/targets/armvirt/64/packages                                                                         
src/gz openwrt_base https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/base                                                                    
src/gz openwrt_luci https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/luci                                                                    
src/gz openwrt_packages https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/packages                                                            
src/gz openwrt_routing https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/routing                                                              
src/gz openwrt_telephony https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/telephony                                                                                                                                                 
## This is the local package repository, do not remove!                                
src imagebuilder file:packages                                                                                                                                                
option check_signature`

predators46, 2022-05-22:
Ping @aparcar

predators46, 2022-05-22:
`cat /home/admin/22-rc2/build_dir/target-aarch64_cortex-a53_musl/root-armvirt/etc/opkg/distfeeds.conf                                                 src/gz openwrt_core https://downloads.openwrt.org/releases/22.03-SNAPSHOT/targets/armvirt/64/packages                                                                         src/gz openwrt_base https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/base                                                                    src/gz openwrt_kmods https://downloads.openwrt.org/releases/22.03-SNAPSHOT/targets/armvirt/64/kmods/5.10.116-1-b9d12596f4a7e29debd9491beddbee91                               src/gz openwrt_luci https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/luci                                                                    src/gz openwrt_packages https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/packages                                                            src/gz openwrt_routing https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/routing                                                              src/gz openwrt_telephony https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/telephony`

dave14305, 2022-05-24:
`maketag.sh` did not update `package/base-files/image-config.in` for some reason.

predators46, 2022-05-27:
Fix 22.03.0-rc3. Thanks

closed 2022-05-27 by predators46