openwrt/openwrt #9935: firewall4: 22.03.0-rc2 fails to start fw4 on devices without hw-offload capability

Shine-, 2022-05-23:
After upgrading to 22.03.0-rc2 (in-place from rc1), the device couldn't route traffic anymore. Troubleshooting showed that fw4 fails to load when `flow_offloading_hw='1'` is set. Device is not hw offload capable. Device worked fine with all previous versions, including 22.03.0-rc1, with identical config.

**Steps to reproduce:**

Use a device that isn't hw offloading capable and enter the following commands:

```
uci set firewall.@defaults[0].flow_offloading='1'
uci set firewall.@defaults[0].flow_offloading_hw='1'
uci commit
/etc/init.d/firewall restart
```

**Expected behavior (up to 22.03.0-rc1):**

fw4 starts with the following output:
`Hardware flow offloading unavailable, falling back to software offloading`
Device starts routing traffic.

**Actual behavior (22.03.0-rc2):**

fw4 throws the following error messages and does not start:
```
/proc/self/fd/0:9:12-13: Error: Could not process rule: Not supported

/proc/self/fd/0:53:29-44: Error: Could not process rule: No such file or directory
```
Device does not route traffic.

Moreover, even when resetting the conflicting setting:
```
uci set firewall.@defaults[0].flow_offloading_hw='0'
uci commit
/etc/init.d/firewall restart
```
fw4 still does not start:
`The fw4 firewall does not appear to be loaded, try fw4 flush to delete all rules.`

Following that advice, fw4 still doesn't start:
```
fw4 flush
/etc/init.d/firewall restart
```
`The fw4 firewall does not appear to be loaded, try fw4 flush to delete all rules.`

**Expected fw4 restart behavior:**

Using `/etc/init.d/firewall restart` should **unconditionally restart** fw4. It should not fail unless there's still a configuration issue.

Shine-, 2022-05-23:
Uneducated guess:
In `fw4.uc`, function `resolve_offload_devices`, when the `devices` array remains empty (as is the case for me), the code path that unsets `flow_offloading_hw` is never reached, causing nftables to fail because it tries to use nonexistent hw offload.

So, might the following be the fix? Or am I wrong here?

```
@@ -468,10 +468,12 @@
 				if (nft_try_hw_offload(devices))
 					return devices;
 
+				devices = [];
+			}
+			if (!length(devices)) {
+				
 				this.warn('Hardware flow offloading unavailable, falling back to software offloading');
 				this.state.defaults.flow_offloading_hw = false;
-
-				devices = [];
 			}
 		}
```

With this change, fw4 starts fine again for me, at least.

Shine-, 2022-05-31:
Working perfectly, thank you, you rock! :-)

closed 2022-05-31 by Shine-

jow-, 2022-06-01:
Fixed with a7ddef6ef122a935d6bdf38d7ead686b12738007