openwrt/openwrt #9945: mt7622: packet passing ethernet to wifi broken

PolynomialDivision, 2022-05-26:
Setup:
```
Client <-WiFi (wlan5-ff)-> AP (Belkin MT7622) <-Ethernet Vlan 40-> Core-Router (Belkin MT7622) (switch0.40 br-dhcp)
```
What works:
- Clients' packets arrive at Core-Router
- Core-Router response
- Response can be seen on switch0.40 on AP

What does not work:
- packets are never passed to wlan5-ff (client network)

How to reproduce:
- Try to reach a client from Lan (VLAN) to WiFi

I suspect:
A patch series is adding offloading from **ethernet to WLAN** on MT7622 SoC (https://lwn.net/Articles/862864/). As explicitly stated, **WLAN->Ethernet offload** is not supported by MT7622.
So probably, the offload fails, and that is why we see the packets arriving at the core-router because there is no offloading from WLAN->Ethernet.

Bisecting shows that OpenWrt 22.03-rc1 is working, but 22.03-SNAPSHOT is not. The log shows a probably related commit (https://github.com/openwrt/openwrt/commit/77e123340f0b5490905e27ddc92f0dff8ed017a5).

Patches:
```
ethernet: mtk_eth_soc: add support for coherent DMA

It improves performance by eliminating the need for a cache flush on rx and tx
In preparation for supporting WED (Wireless Ethernet Dispatch), also add a
function for disabling coherent DMA at runtime.
```

```
arm64: dts: mediatek: mt7622: add support for coherent DMA

It improves performance by eliminating the need for a cache flush on rx and tx
```

```
net: ethernet: mtk_eth_soc: implement flow offloading to WED devices

This allows hardware flow offloading from Ethernet to WLAN on MT7622 SoC
```

```
arm64: dts: mediatek: mt7622: introduce nodes for Wireless Ethernet Dispatch

Introduce wed0 and wed1 nodes in order to enable offloading forwarding
between ethernet and wireless devices on the mt7622 chipset.
```

```
net: ethernet: mtk_eth_soc: add ipv6 flow offload support

Add the missing IPv6 flow offloading support for routing only.
Hardware flow offloading is done by the packet processing engine (PPE)
of the Ethernet MAC and as it doesn't support mangling of IPv6 packets,
IPv6 NAT cannot be supported.
```

```
net: ethernet: mtk_eth_soc: support TC_SETUP_BLOCK for PPE offload

This allows offload entries to be created from user space
```

```
net: ethernet: mtk_eth_soc: remove bridge flow offload type entry support

According to MediaTek, this feature is not supported in current hardware
```

ping @nbd168

PolynomialDivision, 2022-05-26:
I tried to revert everything, but it did not seem to help:
https://github.com/PolynomialDivision/openwrt/commit/b985047edd88ec28e89fe7681500ca65489779df

Maybe I did a mistake when doing the revert.

spolack, 2022-05-27:
I can confirm that rc1 is working and rc2 is already broken.

Setup:
* E8450 acting as an Access Point.
* switch0.41 and wlan interface are member of a bridge
* bridge is reachable from vlan and bridge
* frames from vlan to wlan are seen on switch0.41, bridge, but not on wlan interface. - Other direction seems to work fine.

dfateyev, 2022-05-29:
> I neither see any git tags nor openwrt version specific branches (for 22.03 and 21.02) in the [mt76](https://github.com/openwrt/mt76) repo. How is it decided which commit to use for a specific openwrt branch?

To my understanding, there is just periodic repo sync like fb4edcef712432e0e42f0c28be8829b6d2fb44e9.

PolynomialDivision, 2022-06-05:
Finally, I can reproduce the bug reliable. Put a 802.11s mesh wifi and client access on the 5 ghz channel. Your packets won't be forwarded. (@nbd168)
```
# Radio: 11a_standard
config wifi-device 'radio1'
	option type 'mac80211'
	option hwmode '11a'
	option path '1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option htmode 'VHT20'
	option channel '36'
	option country 'DE'
	option legacy_rates '0'

config wifi-iface 'radio1_if0'
	option device 'radio1'
	option network 'dhcp'
	option ifname 'wlan5-ff'
	option mode 'ap'
	option encryption 'none'
	option ssid 'asdf'
	option isolate '1'

config wifi-iface 'radio1_if1'
	option device 'radio1'
	option network 'dhcp'
	option ifname 'wlan5-ffowe'
	option mode 'ap'
	option encryption 'owe'
	option ssid 'asdf Encrypted'
	option ieee80211w '1'
	option isolate '1'

config wifi-iface 'radio1_if2'
	option device 'radio1'
	option ifname 'wlan5-mesh'
	option mode 'mesh'
	option mesh_id 'Mesh-Freifunk-Berlin'
	option mesh_fwding '0'
	option mcast_rate '12000'
	option network 'mesh_core_5ghz'
```

f00b4r0, 2022-06-06:
I see the E8450 is using the MT7915E radio.

VLAN offloading has been fixed for this radio, but the fix isn't yet in 22.03 AIUI:
https://github.com/openwrt/mt76/commit/eecbb49920732b9b364f15c1ef0066342544b22b

I suspect this is related to this problem.

rotanid, 2022-08-10:
@PolynomialDivision following the discussion on the openwrt forum ( https://forum.openwrt.org/t/issue-introduced-in-22-03-0-rc4-still-there-in-rc5-an-rc6/133770 ), have you tried reverting https://github.com/openwrt/openwrt/commit/33df033b73365487c5bb5a58b77aed04d4ca6ac1 ?

maertsen, 2022-08-11:
Do you still see this behaviour @PolynomialDivision? My topology is similar to the one you described, yet I have yet so see the problematic behaviour after moving from -rc1 to -rc6 earlier this week.

pmagid, 2023-02-02:
I am still seeing this on my E8450....

pmagid, 2023-03-27:
^^^  Is it just me or others still struggling with this bug?

pmagid, 2023-03-28:
As of Snapshot 2023-03-27 my problem appears to be resolved..... Magically...