openwrt/openwrt #9946: Fast Ethernet devices connected to a Gigabit Ethernet switch are limiting upload speeds of Gigabit Ethernet devices to 100 Mbit/s

pesa1234, 2022-05-27:
WRT1900ACSv2 with OpenWrt SNAPSHOT or 22.03rc1 and later (kernel 5.10) gets ~940/940 Mbit/s only if there's no slower devices connected to its switch. When a Fast Ethernet device is connected, upload speeds drop to 100 Mbit/s for all devices, even for those that can handle Gigabit Ethernet. Download speeds are not affected.

xwang1498, 2022-05-27:
Also seeing this behavior on a Linksys WRT3200ACM, running `SNAPSHOT r19704-efff48529b`.

pesa1234, 2022-05-27:
> Also seeing this behavior on a Linksys WRT3200ACM, running `SNAPSHOT r19704-efff48529b`.

I'm quite sure on all series...

sppmasterspp, 2022-05-28:
Can any of you see my findings and tests I performed here. I think
https://github.com/openwrt/openwrt/issues/9202
Please try to run iperf3 between a Gigabit device and Fast Ethernet device (Gbit device uploading to 100Mbps device) and at the same time run WAN speedtest.
In my case I see huge performance degradation and huge ping (that maybe present on your hardware too) and that problem is worst than just limiting the upload speed to 100Mbps.
I've posted the exact steps to reproduce/trigger the bug.
https://forum.openwrt.org/t/netgear-r7800-exploration-ipq8065-qca9984/285/3187?u=sppmaster
If you can, try those tests and comment if you see further WAN/LAN performance degradation.

f4nff, 2022-05-28:
https://github.com/openwrt/openwrt/issues/9796

You said this problem is very similar, but r4s is limited to about 3Mbps, only the uplink speed is limited, and the LAN is not limited, which limits the uplink speed of access to the Internet.
But `iperf3 -c www.google.com -P 2` is fine again.

f4nff, 2022-05-28:
[image]

<html>
<body>

OpenWrt 22.03.0-rc3 r19378-9f415792e1 / LuCI openwrt-22.03 branch git-22.140.66268-ef99568
--

<br class="Apple-interchange-newline">
</body>
</html>

f4nff, 2022-05-28:
r4s：
I have repeatedly confirmed that it is the openwrt firmware ([https://downloads.openwrt.org/releases/22.03.0-rc3/targets/rockchip/armv8/openwrt-22.03.0-rc3-rockchip-armv8-friendlyarm_nanopi-r4s-ext4 -sysupgrade.img.gz](https://downloads.openwrt.org/releases/22.03.0-rc3/targets/rockchip/armv8/openwrt-22.03.0-rc3-rockchip-armv8-friendlyarm_nanopi-r4s-ext4 -sysupgrade.img.gz)), I tried to flash other firmware, but nothing happened.

The following link firmware does not have such a bug.
https://drive.google.com/file/d/1drrJ86LNJZKmcA7117ioUlPabBpD-vCi/view?usp=sharing

pesa1234, 2022-05-28:
I was speaking of Linksys WRT series from 1200...

This bug was not present on kernel 5.4.x but is present on 5.10.x

Ansuel, 2022-06-04:
Mhh interesting that this is similar to https://github.com/openwrt/openwrt/issues/9202
Thing is they are different switch on different target o.O

pesa1234, 2022-06-05:
Yes similar, but not present on kernel 5.4, on that device with kernel 5.4 was ok ?

pesa1234, 2022-09-06:
Dear all,
finally I got time to test and seems to be solved by remove flow control on lan port with less than 1gbit connection...

example:`
ethtool -A lan2 autoneg off rx off`

closed 2022-09-07 by pesa1234

Alia5, 2022-12-15:
> Dear all, finally I got time to test and seems to be solved by remove flow control on lan port with less than 1gbit connection...
> 
> example:` ethtool -A lan2 autoneg off rx off`

Thank you so much for this workaorund!