openwrt/openwrt #9953: Always true condition in `rtl83xx/dsa.c`

qarmin, 2022-05-28:
https://github.com/openwrt/openwrt/blob/064e7e57b483e6879de0facef4f1fce86ec4ad47/target/linux/realtek/files-5.10/drivers/net/dsa/rtl83xx/dsa.c#L549

Any number is bigger than 52 or smaller than 55

jow-, 2022-06-01:
@bkobl @dangowrt - any idea? Seems to have been introduced with 51c8f766124487f6e84076c272645e5f29d32f5e

bkobl, 2022-06-02:
Yes, this should be && not ||. But I am not sure it is worth the effort to fix it. There are no RTL931x devices in master, so this is of no consequence and for the next feature updates foreseen with more support for the RTL931x the code already fixes this, by distinguishing the MAC-SerDes and SerDes-Media link states, which is a much better solution:
https://github.com/bkobl/openwrt/blob/f16b0b026a79d773598eb3aec9dda0e93900d648/target/linux/realtek/files-5.10/drivers/net/dsa/rtl83xx/dsa.c#L482

dangowrt, 2022-06-02:
@bkobl Will you open a PR once the code in your repo is ready to be considered for merging to master?

bkobl, 2022-06-02:
Yes, absolutely. I want to have full support for the XS1930 so that others like you can do the RTL931x testing and at that point I want to create  the PR.

Djfe, 2023-04-26:
Just fishing through old issues, what's the status here. Can this be closed? :)

MaZderMind, 2023-09-24:
This code still exist and is probably still broken. I can open a PR for it but I don't have a `rtl93xx` device.

commit 9fb5082e (2023-10-03): rtl93xx: fix condition intended to only select internal serdes ports

This condition was introduced in commit 51c8f7661244 ("realtek: Improve
MAC config handling for all SoCs") to correctly report the speed of the
internal serdes ports as 10G, but instead makes all ports read 10G
because the or-operator should have been an and-operator.

Fixes: #9953
Fixes: 51c8f7661244 ("realtek: Improve MAC config handling for all SoCs")

[ wrap comment to 72 column and improve commit ref ]
PR #13532 (2023-10-03): rtl93xx: fix condition intended to only select internal serdes ports
This condition was introduced in 51c8f76 to correctly report the speed of the internal serdes ports as 10G, but instead makes all ports read 10G because the or-operator should have been an and-operator.

This fixes #9953

Please note that I do not have a rtl93xx device to test this on, but the code in question was obviously wrong and the change is quite miniscule.
commit 3fff6255 (2023-10-03): rtl93xx: fix condition intended to only select internal serdes ports

This condition was introduced in commit 51c8f7661244 ("realtek: Improve
MAC config handling for all SoCs") to correctly report the speed of the
internal serdes ports as 10G, but instead makes all ports read 10G
because the or-operator should have been an and-operator.

Fixes: #9953
Fixes: 51c8f7661244 ("realtek: Improve MAC config handling for all SoCs")

[ wrap comment to 72 column and improve commit ref ]

(cherry picked from commit 9fb5082e258ac4672dc69636e5eb79f426defac8)
closed 2023-10-03 by openwrt-bot