openwrt/openwrt #9954: [22.03] RC3 Not booting with Microtik hAP ac lite - RB952Ui-5ac2nD - ath79

0nelight, 2022-05-28:
Hi when trying to boot the initramfs 22.03-RC3 with the standard-method it **doesn't boot.**

Image is **22.03.0-rc3**:
https://downloads.openwrt.org/releases/22.03.0-rc3/targets/ath79/mikrotik/openwrt-22.03.0-rc3-ath79-mikrotik-mikrotik_routerboard-952ui-5ac2nd-initramfs-kernel.bin

Method:
https://openwrt.org/toh/mikrotik/rb952ui-5ac2nd_hap_ac_lite

Same method with OpenWRT 19 works fine!

This **commit** should be there in 22.3 RC3 but it doesn't seem to work/boot with this device.
[ath79: add support for MikroTik RouterBOARD hAP ac lite](https://github.com/openwrt/openwrt/commit/2bd33e8626bd04fd7115ee1a42aaf03aae2fffb8)

walterav1984, 2022-05-31:
What is the latest Mikrotik R.OS / bootloader version you have ran / upgraded to, before going to OpenWrt?

0nelight, 2022-05-31:
Maybe related: [ath79: mikrotik: fix initram image booting](https://github.com/openwrt/openwrt/pull/9911)

0nelight, 2022-05-31:
Maybe related: [Link](https://github.com/openwrt/openwrt/pull/9825)

0nelight, 2022-06-03:
Now tried with another "Mikrotik hAP ac lite" with updated software to:
routeros-mipsbe: 6.48.6
system: 6.48.6
current-firmware: 6.48.6

also doesn't work with the 22.03.0-rc3 - but does with 19.

f00b4r0, 2022-06-06:
> Maybe related: [ath79: mikrotik: fix initram image booting](https://github.com/openwrt/openwrt/pull/9911)

This should help, yes.

If you are able to build your own initramfs, you'll want to trim it down, the issue is that the current initramfs is too big to be booted by the device's bootloader.

For those trying to install 22.03, it should be possible to boot from 19.07 initramfs and then sysupgrade -F the 22.03 image.

sgabe, 2022-06-13:
I can confirm that booting from 19.07 initramfs and then sysupgrading the 22.03 image works on MikroTik 962UiGS-5HacT2HnT (hAP ac).

0nelight, 2023-03-14:
Booting [openwrt-22.03.3-initramfs](https://downloads.openwrt.org/releases/22.03.3/targets/ath79/mikrotik/openwrt-22.03.3-ath79-mikrotik-mikrotik_routerboard-952ui-5ac2nd-initramfs-kernel.bin) via BOOTP and then sysupgrading to 
[openwrt-22.03.3-sysupgrade](https://downloads.openwrt.org/releases/22.03.3/targets/ath79/mikrotik/openwrt-22.03.3-ath79-mikrotik-mikrotik_routerboard-952ui-5ac2nd-squashfs-sysupgrade.bin) works now.

jamilabrishami, 2025-12-27:
Very useful topic guys, thank you.
it appears that some version won't work well with some devices.
i'm using rb952 ac lite and i've been trying to  install openwrt for 4 days and it's really pi..ed me off.
the version i was trying is 24.10.4 and i couldn't upload the sysupgrade file after the first init install.

it didn't get the right ip, it resets for itself sometimes and rivert to RouterOS &...

so i found this topic and i tried an older version as you guys said and it worked! 
the current version is 23.03.3 and it's working.

Thank you guys.

commit 68add5f3 (2022-11-27): ath79: mikrotik: use OpenWrt loader for initram image

Return to using the OpenWrt kernel loader to decompress and load kernel
initram image.

Continue to use the vmlinuz kernel for squashfs.

Mikrotik's bootloader RouterBOOT on some ath79 devices is
failing to boot the current initram, due to the size of the initram image.

On the ath79 wAP-ac:
a 5.7MiB initram image would fail to boot
After this change:
a 6.6MiB initram image successfully loads

This partially reverts commit e91344776b9ba7c864be88d915c9c0df0eb790dd.

An alternative of using RouterBOOT's capability of loading an initrd ELF
se [... cut]