openwrt/openwrt #9956: WRT3200ACM 5 GHz DFS not working (and workaround)

Techmeology, 2022-05-28:
I have a WRT3200ACM (bought in the UK), and have found that the DFS channels in the 5 GHz band do not work out of the box. There is a workaround, which I'll talk about below.

## Basic information:
 - OpenWRT version: 22.03.0-rc3
 - Router: WRT3200ACM (bought in the UK)

## Steps to reproduce

1. Install OpenWRT version 22.03.0-rc3 (or probably also any other version) on the router.
2. Set up a 5 GHz access point on channel 64 (or presumably any other DFS channel).

## Expected behaviour

Wifi clients should be able to connect to the created access point.

## Actual behaviour

The access point fails to initialize. The following is printed in the syslog:

```
Sat May 28 07:30:04 2022 [kern.info](http://kern.info/) kernel: [ 3068.774183] device wlan0 entered promiscuous mode
Sat May 28 07:30:04 2022 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sat May 28 07:30:05 2022 daemon.notice netifd: Wireless device 'radio0' is now up
Sat May 28 07:30:09 2022 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Sat May 28 07:30:09 2022 kern.debug kernel: [ 3073.781491] ieee80211 phy0: change: 0x100
Sat May 28 07:30:09 2022 kern.debug kernel: [ 3073.793910] ieee80211 phy0: change: 0x60
Sat May 28 07:30:09 2022 kern.debug kernel: [ 3074.008811] ieee80211 phy0: change: 0x40
Sat May 28 07:30:10 2022 daemon.notice hostapd: wlan0: interface state HT_SCAN->DFS
Sat May 28 07:30:10 2022 daemon.notice hostapd: wlan0: DFS-CAC-START freq=5320 chan=64 sec_chan=-1, width=1, seg0=58, seg1=0, cac_time=60s
Sat May 28 07:30:10 2022 daemon.err hostapd: DFS start_dfs_cac() failed, -1
Sat May 28 07:30:10 2022 daemon.err hostapd: Interface initialization failed
Sat May 28 07:30:10 2022 daemon.notice hostapd: wlan0: interface state DFS->DISABLED
Sat May 28 07:30:10 2022 daemon.notice hostapd: wlan0: AP-DISABLED
```

## Workaround

Run `opkg remove kmod-mwifiex-sdio` (thanks to a friend for helping me find this). This fixed the problem, and also (though I haven't tested extensively) seemed to make the wifi more stable/reliable in general.

This is a [known workaround](https://github.com/kaloz/mwlwifi/issues/173#issuecomment-307984048). I'm not sure what this package is intended to do or of the implications of removing it. But I wonder whether it would be sensible to remove this package from the image for this router (and perhaps other affected routers), or blacklist the kernel module so it's not loaded unless explicitly requested. This would make the DFS channels work out of the box.

Per [this comment](https://github.com/kaloz/mwlwifi/issues/173#issuecomment-308209114), I wonder whether that also makes the `mwifiex-sdio-firmware` package redundant (I haven't investigated whether any alternative kernel module uses that).

Thoughts? Thanks!

howl, 2022-05-28:
You can also set the country hardcoded in the other interfaces for mwifiex with a kernel module parameter, and then you have one more wifi interface to be used at the same time like the others also with DFS.

SimonIremonger, 2022-05-28:
@howl Please be clear -- does country-code setting in that fashion, also work-around the issue, or is that just a related change worth knowing-about and not part of this issue report?    p.s. please note the 3rd wifi interface has low power / small antennae which can be problematic, in any case.

howl, 2022-05-28:
@SimonIremonger is a way to workaround the DFS issue while leave all the radios. When I had WRT3200ACM in use I could have more reliably that third radio that the mwlwifi ones. You just need to set in /etc/modules.d/mwifiex (create if not exists, or check if there is another file there that load mwifiex) with just the line "mwifiex reg_alpha2=FR" without quotes. FR is for the European WRT3200ACM, you should check the country code hardcoded, now that you have removed mwifiex you can issue "iw reg get" to see the country code. Matching the country code DFS will work, because the issue is an interface trying to set up the country code while mwlwifi mess it up if another different to its hardcorded one is tried to set, you end up with no region with no DFS rules.

This issue has been taking long discussions but the real issue is marvell not doing the things right. There was patches to make the module not take into account the hardcoded regions and you could for example in Europe set the country you really are instead of FR for all.

If you have started to use this router my suggestion is sold it and get another one, you are going to face a lot of more issues like no mesh, no WPA3, no WPA2 encrypted frames... mwlwifi module is an incomplete really pain in the ass piece of...

Techmeology, 2022-05-28:
My thinking in creating this issue was to get Wifi working on it as well as possible out of the box, as that helps other people with this router. For me, the problem is solved, though it is useful to know that the problem could also be solved by setting that kernel parameter, as then I could use the third radio too.

Thinking about it further, it would probably also work to just blacklist the kernel module by default rather than not provide it at all. Given your suggestion, I wonder if it would be possible to solve the issue while keeping the third radio by default by reloading the kernel module having determined what region needs to be set in the kernel parameter. That way, it could still be made to work out of the box.

Is there a reason why OpenWRT doesn't use the patches you mention?

howl, 2022-05-28:
#2397 nobody wanted to review it, that patch makes mwlwifi to get the user defined reg domain. Nowadays probably is needed to adapt a little the patch, but, at least for me doesn't deserve the time, is also possible to hang the router with intensive wifi load.

Great router, for it's time, very bad wifi module. There are better options for less price actually.

Shine-, 2022-05-29:
When I had a WRT32X (iirc, same hardware as WRT3200ACM), I used the attached patch for 19.07 and 21.02 (also fixes driver start on unsupported TX power setting). The 21.02 patch should still work for 22.03, since there hasn't been any change since. Enjoy.

[mwlwifi-patch.zip](https://github.com/openwrt/openwrt/files/8793587/mwlwifi-patch.zip)

howl, 2022-05-29:
@Shine- is the same patch I pull requested, @ValCher1961 is the author. The point is that is not included in the mwlwifi "official" repository, the kaloz's one, but also that repository is mostly dead before born.

Shine-, 2022-05-29:
Actually, I'm the author of the two patches I attached. But since these changes are trivial, I'm quite sure that I wasn't the first to figure out that hack.

The Marvel WiFi business has been sold to NXP, and I'm quite sure that NXP doesn't have the slightest bit of interest in maintaining drivers (for free) for years-old AC Wave 1 WiFi chips (ie. legacy products). So, unless the community takes over responsibilty, don't expect **any** development for the mwlwifi driver to happen anymore, at all.

4rs3n, 2022-08-18:
Hi,
Could anyone help me with this?
I ran into the same problem but i was not able to get any radio to work, without drops or luci failing, eventhough i did the opkg remove kmod-mwifiex-sdio (but atm im on 21.02) (im pretty new to openwrt)

66enligne, 2022-08-30:
Hi
Running 21.02, i removed kmod-mwifiex-sdio (unable to remove mwifiex-sdio-firmware because of too many dependances, then reset /etc/config/wireless and reboot. Looks to be working. THX to all for the suggestion
Question to Shine:
I downloaded your patch zip package. While I know how to compile my own firmware I don't know how to use your patch package.
Can you please give me some guidance?
THX

spacebares, 2023-11-08:
reporting, still broken 23.05.0

dermoth, 2024-01-02:
FWIW the proper way to blacklist these kmods would be adding the following lines in `/etc/modules.conf`:

```conf
blacklist mwifiex
blacklist mwifiex_sdio
```

If I'm not mistaken there isn't even any benefit in removing the kmods since it's part of the base image, so it's actually *only* marking the files as deleted in overlayfs without freeing up anything.

kevin-marshall, 2024-01-02:
> FWIW the proper way to blacklist these kmods would be adding the following lines in `/etc/modules.conf`:
> 
> ```
> blacklist mwifiex
> blacklist mwifiex_sdio
> ```
> 
> If I'm not mistaken there isn't even any benefit in removing the kmods since it's part of the base image, so it's actually _only_ marking the files as deleted in overlayfs without freeing up anything.

I just tried this and I don't this this is a complete solution. Here is what I saw after I rebooted.
[image]

dermoth, 2024-01-02:
And removing the `kmod-mwifiex-sdio` changes this??

All this package contains is the kmod files and a config to load them automatically, and by blacklisting it you inhibit module load.

The 3rd radio probably gets autodetected and added at book if the module is loaded, ytyou can remove it with:

```sh
uci del wireless.radio2
```

You should remove the wifinet first (that's something you created yourself, openwrt defaults to no config for that radio).

But with no driver, that radio can hardly cause any harm.

dermoth, 2024-01-04:
FWIW I never had the issue mentioned above in 21.02, but I did have intermittent issues with DFS. I can't say I tested much with this module on 23.05 (I disabled the `mwifiex_sdio` module after less than a day after upgrading) but I keep external logs (with `radio*.log_level=1`) and even after plenty of reboots as I was setting it up and testing (fresh config) I didn't see anything like the issue mentioned above. 

Even with `mwifiex_sdio` blacklisted I see the same DFS issue on 23.05 that I had before on 21.02: Most of the times when a radar is detected it fails to switch the affected channel and `hostapd` starts spitting out `Failed to set beacon parameters` errors every couple seconds, until I restart the WIFI or reboot the router. Someradar channel switches worked flawlessly in 21.02 (away and back to the affected channel), I can't say for 23.05 though as I switched back to non-DFS channels after the first radar event (which failed).

I was hoping the 23.05 upgrade fixes this... I couldn't test the 22.03 release due to a known bug with the switch & VLANs (I even tried with the 22.03 wireless firmware alone, the driver refused to load with it)... But I digress, this is clealy a different DFS issue.

I'm not sure if there are different hardware revisions, I never even opened the shell to check the board, but it's a pretty old one for sure, I bought it in June 2017 in Canada. I tried using DFS to get 160Mhz, and the only 160Mhz channel available in Canada is 36-64 (5180-5320 Mhz with the top half in DFS zone).

kpcyrd, 2024-10-06:
I think I experience the same bug with `OpenWrt 23.05.4, r24012-d8dd03c46f`, I noticed the output of `ip a` also hangs, but completes eventually:

```
root@OpenWrt:~# strace -f ip a
execve("/sbin/ip", ["ip", "a"], 0xbe971da8 /* 14 vars */) = 0
set_tls(0xb6fed4dc)                     = 0
set_tid_address(0xb6fec7d4)             = 3450
brk(NULL)                               = 0x1975000
brk(0x1977000)                          = 0x1977000
mmap2(0x1975000, 4096, PROT_NONE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x1975000
open("/etc/ld-musl-armhf.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=45058, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\0\0\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 49152, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xb6f6a000
mmap2(0xb6f74000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xa000) = 0xb6f74000
close(3)                                = 0
mprotect(0xb6f74000, 4096, PROT_READ)   = 0
mprotect(0x77000, 4096, PROT_READ)      = 0
prctl(PR_GET_NAME, "ip")                = 0
socket(AF_NETLINK, SOCK_RAW, NETLINK_ROUTE) = 3
bind(3, {sa_family=AF_NETLINK, nl_pid=0, nl_groups=00000000}, 12) = 0
getsockname(3, {sa_family=AF_NETLINK, nl_pid=3450, nl_groups=00000000}, [12]) = 0
clock_gettime64(CLOCK_REALTIME, {tv_sec=1728257415, tv_nsec=653527224}) = 0
write(3, "\24\0\0\0\22\0\1\3\210\35\3g\0\0\0\0\0\257\200\276", 20

) = 20
[... this write call hangs really long, afterwards resumes normally]
```

uninstalling the `kmod-mwifiex-sdio` package and adding `blacklist mwifiex` and `blacklist mwifiex_sdio` to `/etc/modules.conf` didn't help. `lsmod | grep mwi` didn't output anything, but I noticed I still have entries related to `radio2` in `/etc/config/wireless`, I've removed them manually and testing now.

iacchi, 2024-12-30:
> EDIT: it didn't seem to fix the issue, anything else I can try?

as @howl wrote at the beginning of this conversation:

You just need to set in /etc/modules.d/mwifiex (create if not exists, or check if there is another file there that load mwifiex) with just the line "mwifiex reg_alpha2=FR" without quotes.

I just did this and it worked, on 23.05.5, without uninstalling/disabling anything.

(I know my answer comes late, but for future reference)

muzzah, 2025-01-28:
@iacchi I have `mwifiex-sdio` already in that directory. Is it necessary to still create `mwifiex` or can I input that line into the former file?

iacchi, 2025-01-28:
> [@iacchi](https://github.com/iacchi) I have `mwifiex-sdio` already in that directory. Is it necessary to still create `mwifiex` or can I input that line into the former file?

I have both in my installation, they're different things.

JavaMiata, 2025-03-23:
> You just need to set in /etc/modules.d/mwifiex (create if not exists, or check if there is another file there that load mwifiex) with just the line "mwifiex reg_alpha2=FR" without quotes.

I assume if I'm in the USA I should set "mwifiex reg_alpha2=US"?  or is the French region setting special somehow?

ascendbeing, 2025-10-31:
@oopen you can specify "-kmod-mwifiex-sdio -wpad-basic-mbedtls" in firmware selector. might save you time. put it as the final word and anything before it gets cancelled out by deselection.