openwrt/openwrt #9958: firewall: move custom input_rule, forwarding_rule and output_rule chains

ne20002, 2022-05-29:
I wonder why the custom _input_rule_, _forwarding_rule_ and _output_rule_ chains are placed as first in the firewall's INPU, OUTPUT and FORWARD chains.
For me it looks like due to this placing, those chains' filter are also applied to all pakets allready belonging to related,established state.
 
[image]

This seems to be unneccessary, as pakets of already related/established connection shouldn't be needed to be filtered again. 

Is there a way to place the custom chains **after** cstate REALTED,ESTABLISHED rule?

jow-, 2022-05-30:
It is intentional to allow creating user rules that do match already established flows (e.g. time based ones, those should also affect already established flows). There is no option to change that.

closed 2022-05-30 by jow-

ne20002, 2022-05-31:
Hi @jow- 
I do understand this intention but I believe to have at least an option to choose wether to put the custom chain before or after the related/established chains would be handy.

E.g.: I just added BCP83 which I believe is a usefull filter. But I don't see why this filtering needs to be applied before related/established. It is just consuming cpu when every and each packet is processed in the filter.
Same applies for IPTV ... it's just a huge amount of packtes which do not need to run through the filter.

It's maybe not an issue as soon as offloading is working again ... (btw, how would offloading would work with time based filters?)