openwrt/openwrt #9959: dnsmasq: TCP DNS queries give no reply

lowjoel, 2022-05-29:
On a Linksys E8450, running OpenWrt 22.03-rc1: when trying to resolve a domain with many DNS records across an IPsec VPN, nslookup (`nslookup-bind`) initially issues a UDP request, sees that there are more records, and switches to TCP:

```
<redacted>
Server:         <redacted>
Address:        <redacted>#53

Name:   <redacted>
Address: <redacted>
Name:   <redacted>
Address: <redacted>
Name:   <redacted>
Address: <redacted>
Name:   <redacted>
Address: <redacted>
Name:   <redacted>
Address: <redacted>
;; Truncated, retrying in TCP mode.
;; communications error to <redacted>#53: end of file

;; communications error to <redacted>#53: end of file

>
```

Enabling dnsmasq query logs show that the UDP packet is received, but not the repeat TCP ones. `tcpdump` shows that a TCP RST packet was received. So, I switched to running dnsmasq in foreground directly using nslookup and forcing tcp:

On dnsmasq session:
```
# /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c --no-daemon --log-debug
dnsmasq: started, version 2.86 cachesize 150
dnsmasq: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
dnsmasq: UBus support enabled: connected to system bus
<snip>
dnsmasq: read /etc/hosts - 4 addresses
dnsmasq: read /tmp/hosts/dhcp.cfg01411c - 10 addresses
dnsmasq: read /tmp/hosts/dhcp.cfg02411c - 10 addresses
dnsmasq: read /tmp/hosts/dhcp.cfg03411c - 10 addresses
dnsmasq: read /tmp/hosts/dhcp.cfg04411c - 10 addresses
dnsmasq: read /tmp/hosts/dhcp.cfg05411c - 10 addresses
dnsmasq: read /tmp/hosts/odhcpd - 2 addresses
dnsmasq-dhcp: read /etc/ethers - 0 addresses
dnsmasq: 1 <redacted>/41881 query[A] google.com from <redacted>
dnsmasq: 1 <redacted>/41881 forwarded google.com to <redacted>
dnsmasq: 1 <redacted>/41881 reply google.com is 74.125.24.102
dnsmasq: 1 <redacted>/41881 reply google.com is 74.125.24.100
dnsmasq: 1 <redacted>/41881 reply google.com is 74.125.24.113
dnsmasq: 1 <redacted>/41881 reply google.com is 74.125.24.138
dnsmasq: 1 <redacted>/41881 reply google.com is 74.125.24.101
dnsmasq: 1 <redacted>/41881 reply google.com is 74.125.24.139
Segmentation fault
#  echo $?
139
```

On nslookup session:
```
# /tmp/usr/libexec/nslookup-bind
> lserver <redacted>
Default server: <redacted>
Address: <redacted>
> set vc
> google.com
Server: <redacted>
Address: <redacted>

Non-authoritative answer:
Name:   google.com
Address: 74.125.24.102
Name:   google.com
Address: 74.125.24.100
Name:   google.com
Address: 74.125.24.113
Name:   google.com
Address: 74.125.24.138
Name:   google.com
Address: 74.125.24.101
Name:   google.com
Address: 74.125.24.139
;; Connection to <redacted>#53(<redacted>) for google.com failed: timed out.
;; communications error to <redacted>#53: connection reset
```

Queries in the reverse direction (i.e. issuing queries to dnsmasq on OpenWrt 21.02 on my older WRT1900ACS) seem to be OK. I did try attaching gdb while recreating this scenario and wasn't successful at getting any debug pointers; I couldn't even extract a backtrace out of it.

If there's something else I can help to track this down, do let me know. Thanks!

lowjoel, 2022-05-30:
I tried building dnsmasq with debug symbols and did the same test:

```
# /tmp/usr/bin/gdb -p 21631
GNU gdb (GDB) 11.2
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-openwrt-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Attaching to process 21631
Reading symbols from /tmp/dnsmasq-debug...

warning: Unable to determine the number of hardware watchpoints available.

warning: Unable to determine the number of hardware breakpoints available.
Reading symbols from /lib/libubox.so.20211120...
(No debugging symbols found in /lib/libubox.so.20211120)
Reading symbols from /lib/libubus.so.20220228...
(No debugging symbols found in /lib/libubus.so.20220228)
Reading symbols from /lib/libgcc_s.so.1...
(No debugging symbols found in /lib/libgcc_s.so.1)
Reading symbols from /lib/ld-musl-aarch64.so.1...
(No debugging symbols found in /lib/ld-musl-aarch64.so.1)

warning: Unable to find dynamic linker breakpoint function.
GDB will be unable to debug shared library initializers
and track explicitly loaded dynamic code.
0x0000007fa1309dcc in ?? ()
(gdb) cont
Continuing.

Program received signal SIGSEGV, Segmentation fault.
0x0000007fa12d5638 in ?? ()
(gdb) bt
#0  0x0000007fa12d5638 in ?? ()
#1  0x0000000000000016 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```

I'm going to next try to put breakpoints at various parts of the code and see where how far the code takes me. If anyone else has suggestions do let me know.

lowjoel, 2022-05-30:
Apparently not using `--no-daemon` I can't reproduce the crash, but I'm still not getting responses to queries via TCP.

lowjoel, 2022-06-24:
After further debugging, it seems that this is a curious interaction between `--bind-dynamic` and TCP: if the request comes from an address that is unrecognised, the request is dropped. Using `--bind-interfaces` solves this problem.

Why this doesn't affect UDP - I do not know, but since the original problem can be fixed, I'll close this as invalid.

closed 2022-06-24 by lowjoel

mnordhoff, 2022-06-24:
That sounds like a DNSMasq bug, then?